Hamlib/Hamlib #1618: Regression Issue - rigerror

foldynl, 2024-10-02:
Commit 428ca4a196d6b50032495e120671ed92d4b1fce4 changed the behavior of the rigerror function between v 4.4. and 4.5. The original (4.4) format of string was: Rig Error String + Debug message. Format of 4.5 is Debug message + Rig Error String. 

The problematic lines are all  `strcat` - example below
https://github.com/Hamlib/Hamlib/commit/428ca4a196d6b50032495e120671ed92d4b1fce4#diff-79c406beb22c5bf0c5eedc5f0343a87c6eb434ae6de6171b493e44a18ff43e2cR332

mdblack98, 2024-10-02:
Can you show the difference?  I don't see it.And why does it matter?

428ca4a#diff-79c406beb22c5bf0c5eedc5f0343a87c6eb434ae6de6171b493e44a18ff43e2cR332

foldynl, 2024-10-02:
I think it's clear from the code that the two changes generate different results; sprintf(error Text, debug error) vs strcat (debugerror, error Text). This must generate different results.

But if you want proof, here it is:
```
(Hamlib-4.4)$ ./testrig 2

testrig: Hello, I am your main() !
rig_open: error = IO error
network.c(301):network_open return(-6)
iofunc.c(177):port_open return(-6)
rig.c(800):rig_open return(-6)
```

```
(Hamlib-4.5)$ ./testrig 2
testrig: Hello, I am your main() !
rig_open: error = rig_check_rig_caps: p1=0x77c4dfb60160, p2=0x77c4dfb666e0, rig_model=0x77c4dfb60160, macro_name=0x77c4dfb666e0
initrigs4_dummy: _init called
rig_init(557): 0x626ecc211b6c rs->comm_state==0?=0
rig_init: rx_range_list1 is empty, using rx_range_list2
rig_init: rig does not have tx_range!!
netrigctl_init version 20220722.0
 1:rig.c(807):rig_open entered
rig_settings_get_path: path=/home/foldynl/.config/hamlib_settings
rig_settings_load_all: settings_file (/home/foldynl/.config/hamlib_settings): No such file or directory
rig_open: cwd=/home/foldynl/Development/Osobni/HAMRADIO/Hamlib/tests
rig_open: /home/foldynl/Development/Osobni/HAMRADIO/Hamlib/tests/hamlib_settings does not exist
rig_open: async_data_enable=0, async_data_supported=0
rig_open: using network address 127.0.0.1:4532
network_open: hoststr=127.0.0.1, portstr=4532
connect to 127.0.0.1:4532 failed, (trying next interface): Network error 111: Connection refused
network_open: failed to connect to 127.0.0.1:4532
rig_open: rs->comm_state==0?=0
 1:rig.c(1013):rig_open returning(-6) IO error

IO error
```

The issue is that in case of 4.5 (and newer) `IO Error` (Error text) is displayed at the end of string. But `IO Error` (Error text) is present at the beginning of the string.  It means that Error message structure in 4.5 (and newer) is Debug Trace + Error Text but 4.4 (and older) is Error Text + Debug Trace.

>>> And why does it matter?

I don't think I need to explain to you that this is a public function and some applications use it. If you change the result structure, you can break many applications. Hamlib is not only the current branch. Ubuntu 22.04 contains Hamlib 4.3. Ubuntu 24.04 contains 4.5. If you run the same application on 22.04 and 24.04, the result is different. In my opinion it is a regression issue, isn't it?

foldynl, 2024-10-02:
just for completeness current master. Another Text error, but the structure is again Debug Message + Error Text

```
 (master)$ ./testrig 2
testrig: Hello, I am your main() !
rig_open: error = rig_init: 2038 time test passed
rig_check_rig_caps: p1=0x5f7aca5de2e0, p2=0x5f7aca5e4868, rig_model=0x5f7aca5de2e0, macro_name=0x5f7aca5e4868
initrigs4_dummy: _init called
rig_init: rig_model=Hamlib NET rigctl 20240418.0
rig_init: rx_range_list1 is empty, using rx_range_list2
rig_init: rig does not have tx_range!!
netrigctl_init version 20240418.0
rig_token_lookup called for rig_pathname
rig_confparam_lookup called for rig_pathname
rig_set_conf called
rig.c(939):rig_open entered
rig_settings_get_path: path=/home/foldynl/.config/hamlib_settings
rig_open: cwd=/home/foldynl/Development/Osobni/HAMRADIO/Hamlib/tests
rig_open: /home/foldynl/Development/Osobni/HAMRADIO/Hamlib/tests/hamlib_settings No such file or directory
rig_open: async_data_enable=0, async_data_supported=0
network_open: TCP connect
rig_open: rs->comm_state==0?=0
rig.c(1168):rig_open returning2(-1) Invalid parameter

Invalid parameter
```

All text above are results of code

```
   retcode = rig_open(my_rig);

    if (retcode != RIG_OK)
    {
        printf("rig_open: error = %s\n", rigerror(retcode));
        exit(2);
    }

```

foldynl, 2024-10-03:
I just hope that your approach to regression issues will not lead to the fact that function close will perform the open operation.

To be honest, I have no problem in the fact that things will change, but the changes must be described to be able to adopt them. it is also missing. It can mean that every hamlib upgrade can generate unexpected surprises for applications. And there is no such WSJT that is using hamlib. 

I wish you good luck on this journey.

foldynl, 2024-10-03:
You can ignore regressions if you have a system that doesn't have an API used by other applications. However, if you are implementing a library where different versions are distributed across various distributions, this unfortunately cannot be overlooked. That's my opinion. If you want to improve the debug output, perfect, every positive improvement is welcome, but it has to be accompanied by a note somewhere stating that this API has changed or by using a different function name. For example, the Qt library constantly changes something, but each release is accompanied by a list of all the public API changes, so library users know what to expect. Hamlib isn't released that often, so this should be possible to create. This way, each application can adapt before users start reporting issues.

My comment regarding open/close was just rhetorical. It's not relevant to this discussion. Look, I don't want to create an unfriendly environment here. Open-source projects should cooperate. But you also have to think about the projects that use your product and what impact it might have on them.

mdblack98, 2024-10-03:
There is no API for the debug output.  I've tried to standardize some things but it is not intended to be parsed by anything.  And I don't know of anybody trying to parse it.

Mike W9MDB

mdblack98, 2024-10-03:
> The rigerror function is listed in the Hamlib API description. In version 4.4, it generated the output differently, but it worked the way I described above. If there were a request to print only debug information or something more specific, then the new function's name should be different, similar to how rigerror2 was introduced.
>
> Yes, applying this new function throughout the source code instead of using rigerror would lead to significant modifications. However, if you're familiar with how sed works, it's just one command to make that change. I don’t want to tell you what to do or how to do it, as I’m not as knowledgeable about the Hamlib code as you are, but I wanted to provide feedback from the perspective of someone who uses this library.

The -vvvvv output is not part of the API.

Are you running the testrig.c in the tests directory or your own testrig.c?

Can you show the code?

foldynl, 2024-10-03:
there may have been a misunderstanding here. I use directly library calls in my application. I did not reported an issue with -vvvv. If I want to report a Hamlib's error in my application, I use rigerror and take the first line (it worked perfect until version 4.5, because there was no rigerror2 function). 4.5 changed the result of the function so it is needed to use the new function rigerror2 or to take the last line from rigerror string.

Just to demonstrate the different outputs, I used the Hamlib's test/testrig.c utility where the rigerror is used.

mdblack98, 2024-10-03:
OK...I see your point now.  Unfortunately not a fix that I can do now.  The purpose of this was to give me more context than just a single line which told me very little about the error that occurred.  And yes...should have been mentioned in the release notes.
At least you did find out the other function which returns a single line.

closed 2024-10-03 by mdblack98