Hamlib/Hamlib #1619: Icom IC-707 CI-V Scan Bug:  Cannot scan VFO

MiM-TMassey, 2024-10-03:
Version:  rigctld Hamlib 4.6~git 2024-09-29
OS:  Windows 10 x64
Radio:  Icom IC-707 via XGGComms USB-Digimode-4 on COM4

### Very brief tl;dr:
On the Icom IC-707, Hamlib cannot start a scan in VFOA/VFOB.  It will scan in MEM VFO, but not in VFOA/VFOB.  Using raw CI-V command does scan in VFO as expected, but not Hamlib.

Two small notes:  It seems "g ?" should return "STOP" as a supported function;  the code that emits "icom_scan: unsupported scan 0x20" is missing a newline, which makes the debug output confusing.

### Details:
I'm working my way through the IC-707 manual and the CI-V reference manual (3rd edition) and comparing them to the results from Hamlib.  I've gotten to Scan functions.

The IC-707 manual describes scans in both VFO mode (from VFOA or VFOB), and in MEM mode (from MEM VFO).  The CI-V reference manual describes only one command 0x0e with only two subcommands:  0x00 to stop a scan, and 0x01 to start a scan.

Using the CI-V commands manually seems to work as the IC-707 manual describes.  When the radio is in one of the actual VFO's (VFOA/VFOB), sending "... 0x0e 0x01 ..." starts the scan:  "SCAN" appears on the radio display and the current frequency begins to increase.  When the radio is in the MEM VFO, sending the same command starts the scan as well:  "SCAN" appears on the radio display and the current memory channel begins to increase.  In either case, sending "... 0x0e 0x00 ..." stops the scan:  "SCAN" disappears on the radio display and the frequency or memory channel stops increasing.  This behavior matches what the radio manual describes.

Howver, this does not work with Hamlib.  First, there seems to be a bit of a disconnect between the CI-V interface and the Hamlib interface.  Hamlib seems to have a single command:  "g".  This command takes two parameters:  "Scan Fct" (I assume function), and "Scan Channel".  A "g ?" gives me the string "VFO MEM", so it seems there are two functions for my radio:  VFO and MEM.  This seems to agree with the radio manual.  I don't really have a "Scan Channel".  It seems that while the number is required (the parser won't continue without it), it doesn't matter what number I use for this radio.  I've been using "0" because it seems to cause the least issues for me.

So, now to use the functions.  Starting with MEM, the results seem to work in a reasonable way.  If I give Hamlib a "g MEM 0", the radio will put itself into the MEM VFO (no matter what VFO I started from), and then begin the scan.  If I then use the "STOP" Scan Fct (i.e. "g STOP 0"), the scan stops.  (An aside:  It would be useful if a "g ?" reported "STOP" as well as VFO and MEM...  It *is* a valid function!)  So far, so good.

However, I do not have the same results with the VFO Scan Fct.  If I give Hamlib a "g VFO 0", I get a "RPRT -1" and nothing happens.  It does not matter what VFO I am in or any other changes I make, I always get that same error.  This is a different error than if I were to alter the ScanFct.  For example, if I give Hamlib a "g VFOA 0", I get a "RPRT -11".  So Hamlib is definitely trying to do something with my "g VFO 0" command, but something is wrong.

I increased to debug output and re-tried the commands.  I've included the output below.  First is the output from executing a "g MEM 0", waiting a few seconds and then executing a "g STOP 0".  These work successfully.  The second output is from executing a "g VFO 0".

The relevant part of the failed output seems to be this:  "icom_scan: unsupported scan 0x20".  (NOTE:  there seems to be an error in the code that generated the output of that line:  it's missing a newline.  You will see what I'm talking about if you look at the failed logs below.  It made it a bit difficult to figure out what the actual error ouptput was.)

An examination of the CI-V documentation I have (including radios up to the early 2000's) does *not* have either an 0x20 command nor a 0x20 subcommand for the 0x0e command.  So I'm not sure what this command is for, or why Hamlib is trying this command.  Possibly something that works for modern radios?

In any case, I'm not sure where to go from here.  MEM scans work, VFO scans do not.  Both work with manual CI-V commands.  I would greatly appreciate any help you might be able to give me.

Once again, thank you very much for all of your work on Hamlib.  It's an incredible resource for those of us who love to combine computers and radios.  I'm amazed how well it works, even with a seemingly-unpopular radio from more than 30 years ago.  And I saw that my previous issue is already fixed!  I'll test that right now!  Thank you very much.

---

===== BEGIN DEBUG OUTPUT OF "g MEM 0" / "g STOP 0"
```
2024-10-03T18:05:37.292271-0500: mutex_rigctld: client lock engaged
2024-10-03T18:05:37.292271-0500: rigctl_parse.c(4009):rigctl_scan entered
2024-10-03T18:05:37.307895-0500: rig_parse_scan called
2024-10-03T18:05:37.307895-0500: **2:rig.c(7161):rig_scan entered
2024-10-03T18:05:37.307895-0500: ***3:rig.c(7126):rig_has_scan entered
2024-10-03T18:05:37.307895-0500: ***3:rig.c(7129):rig_has_scan returning(1)
2024-10-03T18:05:37.307895-0500: ***3:icom.c(8668):icom_scan entered
2024-10-03T18:05:37.307895-0500: ****4:rig.c(3287):rig_set_vfo entered
2024-10-03T18:05:37.307895-0500: rig_set_vfo called vfo=MEM
2024-10-03T18:05:37.307895-0500: icom_set_vfo called vfo=MEM
2024-10-03T18:05:37.307895-0500: *****5:frame.c(507):icom_transaction entered
2024-10-03T18:05:37.323519-0500: icom_transaction: cmd=0x08, subcmd=0xffffffff, payload_len=0
2024-10-03T18:05:37.323519-0500: ******6:frame.c(145):icom_one_transaction entered
2024-10-03T18:05:37.445942-0500: ******6:frame.c(485):icom_one_transaction returning(0)
2024-10-03T18:05:37.445942-0500: *****5:frame.c(538):icom_transaction returning(0)
2024-10-03T18:05:37.445942-0500: icom.c(3189):icom_set_vfo returning2(0)
2024-10-03T18:05:37.445942-0500: icom_get_freq called for MEM, curr_vfo=MEM
2024-10-03T18:05:37.445942-0500: icom_get_freq: using vfo=MEM
2024-10-03T18:05:37.445942-0500: icom.c(9548):set_vfo_curr returning2(0)
2024-10-03T18:05:37.445942-0500: *****5:frame.c(507):icom_transaction entered
2024-10-03T18:05:37.445942-0500: icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
2024-10-03T18:05:37.445942-0500: ******6:frame.c(145):icom_one_transaction entered
2024-10-03T18:05:37.608717-0500: ******6:frame.c(485):icom_one_transaction returning(0)
2024-10-03T18:05:37.608717-0500: *****5:frame.c(538):icom_transaction returning(0)
2024-10-03T18:05:37.624344-0500: *****5:frame.c(507):icom_transaction entered
2024-10-03T18:05:37.624344-0500: icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
2024-10-03T18:05:37.624344-0500: ******6:frame.c(145):icom_one_transaction entered
2024-10-03T18:05:37.778017-0500: ******6:frame.c(485):icom_one_transaction returning(0)
2024-10-03T18:05:37.778017-0500: *****5:frame.c(538):icom_transaction returning(0)
2024-10-03T18:05:37.778017-0500: icom.c(9548):set_vfo_curr returning2(0)
2024-10-03T18:05:37.778017-0500: from_bcd called
2024-10-03T18:05:37.778017-0500: icom_get_freq exit vfo=MEM, curr_vfo=MEM, freq=1.41e+07
2024-10-03T18:05:37.778017-0500: icom.c(1921):icom_get_freq returning2(0)
2024-10-03T18:05:37.778017-0500: ****4:rig_set_vfo: elapsed=470ms
[... 52 lines cut ...]

===== BEGIN DEBUG OUTPUT OF "g VFO 0"
```
2024-10-03T18:06:10.083323-0500: mutex_rigctld: client lock engaged
2024-10-03T18:06:10.098926-0500: rigctl_parse.c(4009):rigctl_scan entered
2024-10-03T18:06:10.098926-0500: rig_parse_scan called
2024-10-03T18:06:10.098926-0500: **2:rig.c(7161):rig_scan entered
2024-10-03T18:06:10.098926-0500: ***3:rig.c(7126):rig_has_scan entered
2024-10-03T18:06:10.098926-0500: ***3:rig.c(7129):rig_has_scan returning(32)
2024-10-03T18:06:10.098926-0500: ***3:icom.c(8668):icom_scan entered
2024-10-03T18:06:10.098926-0500: icom_scan: unsupported scan 0x202024-10-03T18:06:10.098926-0500: ***3:icom.c(8742):icom_scan returning(-1) Invalid parameter

2024-10-03T18:06:10.098926-0500: **2:rig.c(7175):rig_scan returning(-1) Invalid parameter

2024-10-03T18:06:10.100003-0500: rigctl_parse.c(4021):rigctl_scan returning2(-1) Invalid parameter

2024-10-03T18:06:10.100003-0500: mutex_rigctld: client lock disengaged
2024-10-03T18:06:10.100003-0500: handle_socket: rigctl_parse retcode=-1
2024-10-03T18:06:10.100003-0500: mutex_rigctld: client lock engaged
2024-10-03T18:06:10.100003-0500: mutex_rigctld: client lock disengaged
2024-10-03T18:06:10.100003-0500: mutex_rigctld: client lock engaged
2024-10-03T18:06:10.100003-0500: mutex_rigctld: client lock disengaged

```
=====  END  DEBUG OUTPUT OF "g VFO 0"

MiM-TMassey, 2024-10-06:
Upgraded to:  hamlib-w32-4.6~git-20241006-53a3d8635.

I looked at the patch:  it seems you added a couple of additional scan types.  So I did a "g ?".  "STOP" is now listed, thank you!  "SLCT" is now also listed.  Digging into the doxygen doc, it seems that "SLCT" is 'scan elected memory channels'.  I'm not sure that this is necessary?  I cannot determine a difference between "MEM" and "SLCT" on this radio -- unless there's some way to select channels in Hamlib?  I don't see anything like that in the rigctld manpage.  I don't believe there is a way to do that on the radio itself, and there's only a single "Start scan" CI-V command.  Anyway, doing a "g SLCT 0" command seems to work just like "g MEM 0".

Performing a "g VFO 0" had exactly the same result.  This includes the same failure and the same missing newline ( icom_scan: unsupported scan 0x202024-10-06T14:33:06.610172-0500: ***3:icom.c(8742):icom_scan returning(-1) Invalid parameter )  This is likely to be expected, seeing as no underlying code was changed.

Detailed log output below.  Please let me know if there is something else you'd like me to do.  Thank you for your help!

```
2024-10-06T14:33:06.594569-0500: mutex_rigctld: client lock engaged
2024-10-06T14:33:06.594569-0500: rigctl_parse.c(4009):rigctl_scan entered
2024-10-06T14:33:06.594569-0500: rig_parse_scan called
2024-10-06T14:33:06.594569-0500: **2:rig.c(7161):rig_scan entered
2024-10-06T14:33:06.594569-0500: ***3:rig.c(7126):rig_has_scan entered
2024-10-06T14:33:06.594569-0500: ***3:rig.c(7129):rig_has_scan returning(32)
2024-10-06T14:33:06.610172-0500: ***3:icom.c(8668):icom_scan entered
2024-10-06T14:33:06.610172-0500: icom_scan: unsupported scan 0x202024-10-06T14:33:06.610172-0500: ***3:icom.c(8742):icom_scan returning(-1) Invalid parameter

2024-10-06T14:33:06.610172-0500: **2:rig.c(7175):rig_scan returning(-1) Invalid parameter

2024-10-06T14:33:06.610172-0500: rigctl_parse.c(4021):rigctl_scan returning2(-1) Invalid parameter

2024-10-06T14:33:06.610172-0500: mutex_rigctld: client lock disengaged
2024-10-06T14:33:06.625806-0500: handle_socket: rigctl_parse retcode=-1
2024-10-06T14:33:06.625806-0500: mutex_rigctld: client lock engaged
2024-10-06T14:33:06.625806-0500: mutex_rigctld: client lock disengaged
2024-10-06T14:33:06.625806-0500: mutex_rigctld: client lock engaged
2024-10-06T14:33:06.625806-0500: mutex_rigctld: client lock disengaged
```

MiM-TMassey, 2024-10-09:
Upgraded to: rigctl Hamlib 4.6~git 2024-10-07T17:19:03Z SHA=882d22 32-bit

It seems that "SCLT" has been removed from the list of scan types:

```
g ?

MEM VFO STOP
RPRT 0
```

Seems clean and logical.  Thank you!

Also, great news:  VFO scanning now works somewhat.  If I perform a "g VFO 0", I get a RPRT 0 and the radio begins to scan.  Excellent!

One issue I find, though, is that the scan is forced to be on VFOA.  It doesn't matter which VFO I'm on, the radio is forcibly moved to VFOA before the scan starts.  This is true even if I put both a VFO change and a scan together on the same line ("V VFOB g VFO 0").

I'm not sure that this makes sense.  I understand that the radio won't report which VFO it's on, so Hamlib can't be 100% confident which type of scan it's doing unless it changes to a specific VFO.  But it's certainly unexpected to have it switch away from VFOB when you ask for a VFO scan.

I think I also understand the issue with the mismatch between the interface Hamlib provides and what ICOM CI-V is providing.  The Hamlib scan request specifies a scan on a specific type of VFO, either VFO or MEM, and CI-V only has a single command:  start a scan on the current VFO, no matter the type.

There might be a straightforward way to address this, however.  Instead of Hamlib using the CI-V command to explicitly change to a specific VFO (Command:  0x07 Subcommand:  0x00), why not use the CI-V command that returns to the last used VFO (Command 0x07 **no** subcommand)?  From my testing, that works correctly whether the radio is in MEM mode or currently in a VFO.  For MEM mode, it returns to the last-used VFO.  For VFOA/VFOB, it does nothing.

I tested with raw CI-V commands:  the radio will indeed perform a VFO scan while in VFO B.  And when paired with the switch-to-last-used-VFO command seems to give consistent and intuitive results.

In any case, thank you very much for all of your help!  At least the basic function is at least somewhat possible now, if limited to VFOA.

MiM-TMassey, 2024-10-09:
I just went digging a bit into icom.c to figure out how the scan command is being sent.  Turns out I was looking at a slightly older version.  I found the commit that fixed scan operation for the IC-707:  https://github.com/Hamlib/Hamlib/commit/baf2897e76b367325a1a2085df7bf4b0459fd862  Glad I decided to try the latest just in case!  :)

It seems that the VFOA switch is indeed hard-coded into the icom.c icom_scan() function now (line 8681):  `retval = rig_set_vfo(rig, scan==RIG_SCAN_MEM?RIG_VFO_MEM:RIG_VFO_A);`  It seems what we need is something like a "RIG_VFO_LAST" using the 0x07 CI-V command without a subcommand, instead of forcing "RIG_VFO_A".

A little more exploring and I found where the VFO change subcommand parameters are coded:  icom_defs.h starting at line 151.  Unfortunately, there doesn't seem to be an option for no subcommand, at least not there.

Looking around the icom.c/h code, I don't really see any concept of being able to specify "last VFO":  it always seems to be hard-coded as VFOA or VFOB (usually A).  But it seems all we need is a way to specify a RIG_VFO_LAST that will execute the 0x07 command without subcommand and use that instead of RIG_VFO_A in that line in icom.c.

I know, easy for me to say:  I'm not doing the work!  :)

If adding the idea of "last VFO" would be too difficult, would it be possible to make the ICOM scan commands a two-part command?  If the "scan MEM" command were a 0x08 (switch to MEM) command followed by the 0x0e 0x01 (start scan) command, and the "scan VFO" command were a 0x07 (switch to last VFO) / 0x0e 0x01 (start scan).  Then icom_scan() wouldn't have to explicitly change VFO at all -- let the radio do the work! :)

I'm pretty certain you need **none** of this information -- I'm sure you know Hamlib and CI-V **way** better than I.  But I wanted to record what I found at least for myself, so I thought I'd add it here to keep it all in one place.

Again, thank you **very** much for all of your efforts!  The IC-707 is really so much more useful with Hamlib, and I'm grateful for the cleanup effort on such an old and little-used radio.

mdblack98, 2024-10-09:
The behavior now should be that you'll get the current vfo to scan instead of VFOA.So#1 If rig is in MEM mode "G VFO 0" will switch to the last VFO#2 If rig is in VFO mode "G VFO 0" will stay on the current VFO#3 If rig is in VFO mode "G MEM 0" will switch to MEM mode
If you run rigctl with the --vfo option you can directly select VFOA/VFOBG VFOA VFO 0G VFOB VFO 0

MiM-TMassey, 2024-10-10:
Upgraded to: rigctld Hamlib 4.6~git 2024-10-09T13:53:37Z SHA=e09642 32-bit

VFO scanning works **mostly** correctly.  If a user does not touch the radio, everything works correctly:  `g VFO 0` from VFOA/VFOB starts a scan in the current VFO.  From MEM it moves back to the previous VFO and starts the scan.  `g MEM 0` works from anywhere.

**HOWEVER**:  If the user switches the VFO on the radio, this does not work.  For example, if I use Hamlib to `V VFOA`, then press the radio's button to go to VFOB, then switch to MEM (either on the radio or in Hamlib), doing a `g VFO 0` will switch to VFOA instead of VFOB.  

I get it:  Hamlib doesn't see the button-push, and can't update it's currVFO.  But this isn't a problem if Hamlib were to use the 0x07 command **without** a subcommand to switch to VFO instead of specifying a specific VFO.

I would appreciate your thoughts.  Thank you!

mdblack98, 2024-10-30:
Can you send me debug data showing this sequence.  I tried duplicating on an IC-7300 and it works correctly.
V VFOB
V MEM
g VFO 0 -- and VFOB scans correctly

MiM-TMassey, 2024-12-23:
I missed this reply!  Sure thing: I’ll get working on that right now.

MiM-TMassey, 2024-12-23:
Upgraded to:  Hamlib 4.7~git 2024-12-21T21:19:25Z SHA=2b2305 32-bit

I'm no longer able to reproduce this as I was before.  I'm going to go with this being fixed.  I'm not completely convinced -- I *think* I saw this re-occur at one point while I was testing -- but like I said, I can't cleanly reproduce it.

Besides, at this point, I'm pretty annoyed with the entire way this radio handles C-IV with user changes to the radio.  I'm basically fed up with the severe lack of information you can get from the radio.  I'm not sure what Icom was thinking:  did they want the computer to continuously monitor the stream of packets coming from the radio and update its internal state from that?  How was that supposed to work with non-multitasking 1980's computer technology?!?  Whatever.

And the combination of memory and scan really aren't something I'm interested in at this time.  Maybe if someone cares more about this feature than I do they might be able to dig through this.  Otherwise, I'm not sure it's important to anyone else at the moment.  So I think this can be closed.

For my purposes, I'm going to consider the idea of dealing with user changes on this radio to be a non-starter.  If the user pushes buttons, they get what they get!  :)  If you keep your hands off the radio, it works plenty well enough.  You have computer control:  you're going to have to **use** the computer to control... :)

OK, enough ranting.  Again, thank you for your efforts.  I apologize for missing this request, but it seems to be at least mostly resolved.  Looking forward to 4.7!

closed 2024-12-23 by MiM-TMassey