Hamlib/Hamlib #1622: TS-480SAT Command "l RFPOWER" problem

rz4ag, 2024-10-20:
Hello!

Just found problem with command "l RFPOWER" on TS-480SAT
First time it display correct value. But if you send it again then rigctl stops with "Segmentation fault" message. 
Every 2nd call of "l RFPOWER" kills rigctl. 
Other get_level comands work after first "l RFPOWER" without problems. 

Alex.

mdblack98, 2024-10-20:
What version of Hamlib?

rigctl --version

rz4ag, 2024-10-20:
Version 4.6~git

mdblack98, 2024-10-20:
I need some debug info

So run like this:
rigctl -m 2028 -r com4 -s 115200 -vvvvv -Z 2>log.txt
And send me the log.txt file -- adjust com port and baud rate as appropriate

rz4ag, 2024-10-20:
File is attached
[log.txt](https://github.com/user-attachments/files/17451894/log.txt)

mdblack98, 2024-10-20:
That was fixed Oct 1st.

You need a newer version.  Either do a "git pull" or download it.

https://n0nb.users.sourceforge.net/

Mike W9MDB

rz4ag, 2024-10-21:
Just downloaded and installed hamlib-4.6~git-20241020-c02cf9726.tar.gz
Still have the same problem.

mdblack98, 2024-10-21:
You can build a debug version and track down the offending line....I don't see the problem right now.

./configure --enable-static CFLAGS="-g -O0 -fPIC -D_FORTIFY_SOURCE=2"
make install
gdb rigctl
gdb> run -m 2028 -r com4 -s 115200 -vvvvv -Z 

If you can copy/paste the last 100 lines or so of the debug output that should show the error well enough to fix it.

Mike W9MDB

rz4ag, 2024-10-21:
I belive it's enought:

Reading symbols from rigctl...
(gdb) run -m 2028 -r /dev/ttyUSB0 -s 9600
Starting program: /usr/local/bin/rigctl -m 2028 -r /dev/ttyUSB0 -s 9600
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0xf7407400 (LWP 5617)]
[New Thread 0xf6aff400 (LWP 5618)]

Rig command: l RFPOWER
Level Value: 1

Rig command: l RFPOWER

Thread 1 "rigctl" received signal SIGSEGV, Segmentation fault.
0xf7741da8 in kenwood_get_power_minmax (rig=rig@entry=0x44668, 
    power_now=0xf7c35520 <ts480_caps+20480>, power_min=power_min@entry=0x0, 
    power_max=0xf7c30520 <ts480_caps>, power_max@entry=0x0, restore=0)
    at kenwood.c:3037
3037	        rs->power_min = *power_min = 5;

rz4ag, 2024-10-21:
hmmm... Very strange. I made
make uninstall 
then removed old version sources and recompiled new one.
I'll try again

mdblack98, 2024-10-21:
make uninstall will not remove any packages you installed.
You have to do that through your package management system.

rz4ag, 2024-10-21:
Finally I got new version.
rigctl Hamlib 4.6~git 2024-10-18T14:30:19Z SHA=c02cf9 32-bit

/home/rz4ag/local/bin/rigctl -m 2028 -r /dev/ttyUSB0 -s 9600 

Rig command: l RFPOWER
Level Value: 20                     <=== 100 watts on trx

Rig command: l RFPOWER
Level Value: 10                     <=== 50 watts on trx

Rig command: l RFPOWER
Level Value: 1                     <=== 5 watts on trx

Rig command: L RFPOWER 0.5                      <=== sets 5 watts on trx

Rig command: L RFPOWER 20
set_level: error = kenwood_transaction: No data expected, checking ID; in ID020;
kenwood_transaction: returning RIG_OK, retval=0
kenwood.c(689):kenwood_transaction returning2(0) 
**2:kenwood.c(3551):kenwood_set_level returning(0) 
rigctl_parse.c(3422):rigctl_set_level returning2(0) 
rigctl_parse: called, interactive=1
rigctl_parse: cmd=L(4c) handle=0
rigctl_parse: vfo_opt=0
rigctl_parse.c(3317):rigctl_set_level entered
rig_parse_level called
rig_set_level called
kenwood_ts480_set_level called
**2:kenwood.c(3262):kenwood_set_level entered
rig_setting2idx called
rig_setting2idx: idx=12
**2:kenwood.c(3267):kenwood_set_level returning(-1) Invalid parameter

rigctl_parse.c(3422):rigctl_set_level returning2(-1) Invalid parameter

Invalid parameter

rz4ag, 2024-10-23:
Tested. Seems OK

closed 2024-10-23 by mdblack98