Hamlib/Hamlib #1627: WSJTX+ICOM7610+RSBA1v2 impossible to use

dm5xx, 2024-11-04:
Hi I upgraded Hamlib 4.6 2024-09-26 in WSJTX 2.7R7,

but vy strange things started to happen. Its impossible to use WSJTX with 7610 using hamlib4.6 2024-09-26. 

See screenshot. Reverted back to RC6 and old hamlib. 2024-07-13 and its working again.

Maybe its a wsjtx-problem or its hamlib. Reason unknown. Please check.
[image]

mdblack98, 2024-11-04:
Please update Hamlib again in RC7 -- should be dated 11-04

I believe that problem has been fixed.

Mike W9MDB

screenshot.jpeg (view on web)

closed 2024-11-04 by dm5xx

mdblack98, 2024-11-04:
> Hi mike,
>
> thanks for ur quick answer. and you are right, its fixed in the version u can download from sourceforge (autodownload in RC7 was not able to download the newest version, i need to add it manually) which is 2024_10_30 and also the 2024_11_04 version from github works without an issue (using RC6 for testing).
>
> Since the RC7 coming with the 2024_09_26, which actually has issues, this ticket might be important for others: Update hamlib to 2024_10_30 or later!
>
> TNX
> Mike

You need to install openssl to get the downloads of everything to work.

64-bit package is here

https://sourceforge.net/projects/wsjt-x-improved/files/Additional%20Files/OpenSSL/Win64OpenSSL_Light-1_1_1a.msi

Mike W9MDB