Hamlib/Hamlib #1632: send_cmd appears not to work with some longer strings

c-o-m-m-a-n-d-e-r-d-a-t-a, 2024-11-27:
I am finding that the send_cmd, or w option doesn't seem to work, but using the program interactively and using the same command string does work.

Example:

rigctl  -r /tmp/kappanhang-IC-705.pty -m 3085 w '\0xFE\0xFE\0xA4\0xE0\0x08\0xFD'
-- This does work, sets rig into memory mode

rigctl  -r /tmp/kappanhang-IC-705.pty -m 3085 w '\0xFE\0xFE\0xA4\0xE0\0x0E\0xB2\0x01\0xFD'
-- This doesn't work, should select SEL MEM 1, but sends radio back to VFO mode

rigctl  -r /tmp/kappanhang-IC-705.pty -m 3085 w '\0xFE\0xFE\0xA4\0xE0\0x0E\0x23\0xFD'
-- This also doesn't work, should set radio scanning on previously selected SEL MEM, but sends radio back to VFO mode

If I enter interactive mode as follows:
rigctl  -r /tmp/kappanhang-IC-705.pty -m 3085

and then do the same strings interactively:
\send_raw icom 0xFE;0xFE;0xA4;0xE0;0x08;0xFD
\send_raw icom 0xFE;0xFE;0xA4;0xE0;0x0E;0xB2;0x01;0xFD
\send_raw icom 0xFE;0xFE;0xA4;0xE0;0x0E;0x23;0xFD
-- These (almost always) work fine

I realise kappanhang could be affecting things, but not sure why it would work for the first command, but not the next two.

Thanks.

mdblack98, 2024-11-27:
Get rid of the single quotes around the hex digits

mdblack98, 2024-12-01:
You should be able to use send_raw now from the command line with the latest git version -- no escaping or quotes needed
rigctl -r /tmp/kappanhang-IC-705.pty -m 3085 send_raw icom xFE:xFE:xA4:xE0:x0E:xB2:x01:xFD
If you find this method does not have consistent results I'll need to see some debug information like this to generate a log.txt file
rigctl -r /tmp/kappanhang-IC-705.pty -m 3085 send_raw icom xFE:xFE:xA4:xE0:x0E:xB2:x01:xFD >log.txt 2>&1

c-o-m-m-a-n-d-e-r-d-a-t-a, 2024-12-16:
I finally got around to testing this out with the latest git version, here is the output:

rigctl -r /tmp/kappanhang-IC-705.pty -m 3085 -vv send_raw icom xFE:xFE:xA4:xE0:x0E:xB2:x01:xFD
async_data_handler: read_frame_direct() failed, result=-8
async_data_handler: read_frame_direct() failed, result=-8
rig_get_cache(302): unknown vfo=otherVFO
icom_set_vfo: unknown vfo 'otherVFO'
rig_set_vfo: set_vfo otherVFO failed with 'rig_get_freq: cache miss age=1000000ms, cached_vfo=otherVFO, asked_vfo=otherVFO, use_cached_freq=1
rig_get_freq(2130): vfo_opt=0, model=3085
icom_get_freq called for otherVFO, curr_vfo=None
icom_get_freq: using vfo=otherVFO
icom.c(1671) trace
set_vfo_curr: vfo=otherVFO, curr_vfo=None
set_vfo_curr: setting new vfo=otherVFO
icom.c(9441) trace
  2:rig.c(2771):rig_set_vfo entered
rig_set_vfo called vfo=otherVFO
vfo_fixup:(from rig_set_vfo:2791) vfo=otherVFO, vfo_curr=None, split=0
vfo_fixup:(from rig_set_vfo:2833) vfo=otherVFO, vfo_curr=None, split=0
rig.c(2852) trace
icom_set_vfo called vfo=otherVFO
icom_set_vfo: line#2878
icom_set_vfo: unknown vfo 'otherVFO'
icom.c(3078):icom_set_vfo returning2(-1) Invalid parameter

Invalid parameter
Invalid parameter
'
rig_get_vfo: no get_vfo
rig_get_cache(302): unknown vfo=otherVFO
rig_set_cache_freq(208): unknown vfo?, vfo=otherVFO
async_data_handler: read_frame_direct() failed, result=-8
async_data_handler: read_frame_direct() failed, result=-8
rig_set_vfo: set_vfo VFOA failed with 'async_data_handler: received frame: len=199 async=1
 1:icom.c(8973):icom_process_async_frame entered
from_bcd called
  2:event.c(594):rig_fire_freq_event entered
Event: freq changed to 0Hz on currVFO
rig_set_cache_freq(121):  vfo=currVFO, current_vfo=VFOA
  2:event.c(609):rig_fire_freq_event returning(0) 
 1:icom.c(9032):icom_process_async_frame returning(0) 
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): RX 99 characters, direct=1
async_data_handler: read_frame_direct() failed, result=-8
Protocol error

0:frame.c(455):icom_transaction returning(-8) Protocol error

icom_set_vfo: line#3084
icom.c(3088):icom_set_vfo returning2(-8) Protocol error

Protocol error
Protocol error
'
rig_get_vfo: no get_vfo
async_data_handler: read_frame_direct() failed, result=-8
async_data_handler: read_frame_direct() failed, result=-8
rig_set_vfo: set_vfo VFOB failed with 'async_data_handler: received frame: len=199 async=1
0:icom.c(8973):icom_process_async_frame entered
from_bcd called
 1:event.c(594):rig_fire_freq_event entered
Event: freq changed to 0Hz on currVFO
rig_set_cache_freq(121):  vfo=currVFO, current_vfo=VFOB
 1:event.c(609):rig_fire_freq_event returning(0) 
0:icom.c(9032):icom_process_async_frame returning(0) 
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): RX 99 characters, direct=1
async_data_handler: read_frame_direct() failed, result=-8
Protocol error

[... 55 lines cut ...]
icom_get_freq: wrong frame len=3
icom_rig_open: rig error getting frequency retry=0, err=write_block(): TX 6 bytes, method=2
read_string_generic called, rxmax=200 direct=0, expected_len=1
icom_one_transaction: frm_len=9, frm_len-1=fd, frm_len-2=00
                     -4:frame.c(403):icom_one_transaction returning(0) 
                     -5:frame.c(455):icom_transaction returning(0) 
icom.c(1682) trace
set_vfo_curr: vfo=VFOA, curr_vfo=VFOA
set_vfo_curr: curr_vfo now=VFOA
icom.c(9456):set_vfo_curr returning2(0) 
icom_get_freq: wrong frame len=3
                     -6:icom.c(1733):icom_get_freq returning(-9) Command rejected by the rig

rig_get_freq(2147): freqMainA=0, modeMainA=, widthMainA=0
rig_get_freq(2147): freqMainB=0, modeMainB=, widthMainB=0
rig_get_cache(258):  vfo=VFOA, current_vfo=VFOA
rig_get_cache(434): vfo=VFOA, freq=0, mode=, width=0
rig_set_cache_freq(121):  vfo=VFOA, current_vfo=VFOA
                     -7:rig_get_freq: elapsed=0ms
Command rejected by the rig
Command rejected by the rig

icom_get_freq: wrong frame len=2
Opened rig model 3085, 'IC-705'
rigctl_parse: rig_powerstat is not on = 0

mdblack98, 2024-12-16:
Please run that like this:
rigctl -r /tmp/kappanhang-IC-705.pty -m 3085 -vvvvv send_raw icom xFE:xFE:xA4:xE0:x0E:xB2:x01:xFD >log.txt 2>&1

And send me the log.txt file.
I ran that same command here with an IC-7300 and it works:rigctl -m 3073 -r com3 -s 115200 -vvv send_raw icom xFE:xFE:x94:xE0:x0E:xB2:x01:xFDOpened rig model 3073, 'IC-7300'0xfe 0xfe 0xe0 0x94 0xfb 0xfd

c-o-m-m-a-n-d-e-r-d-a-t-a, 2024-12-16:
Attached...
[log.txt](https://github.com/user-attachments/files/18157803/log.txt)

mdblack98, 2024-12-16:
That is not the latest Hamlib.  That is 4.5.5.If you compile the latest version than you have an old libhamlib hanging around -- most likely in /usr/libDid you remove any installed hamlib packages?
rigctl --versionrigctl Hamlib 4.6~git 2024-12-04T05:19:16Z SHA=941e69 64-bit
Which rigctl are you executing:which rigctlAnd which libraries are getting loaded -- be sure to use the path to rigctl.
ldd /usr/local/bin/rigctl

mdblack98, 2024-12-16:
Resending without the html formatting.

That is not the latest Hamlib.  That is 4.5.5.
If you compile the latest version than you have an old libhamlib hanging around -- most likely in /usr/lib
Did you remove any installed hamlib packages?

rigctl --version
rigctl Hamlib 4.6~git 2024-12-04T05:19:16Z SHA=941e69 64-bit

Which rigctl are you executing:
which rigctl
And which libraries are getting loaded -- be sure to use the path to rigctl.
ldd /usr/local/bin/rigctl

c-o-m-m-a-n-d-e-r-d-a-t-a, 2024-12-17:
Sorry my bad - I used the release rather than cloning.

I cloned, then went to build, now I get this:

hpman% ./bootstrap
Running 'autoreconf -i' to process configure.ac
and generate the configure script.
configure.ac:680: error: AM_COND_IF: no such condition "LUAJIT"
/usr/share/aclocal-1.17/cond-if.m4:23: AM_COND_IF is expanded from...
/usr/share/aclocal/ax_lua.m4:199: AX_PROG_LUA is expanded from...
lib/m4sugar/m4sh.m4:697: AS_IF is expanded from...
configure.ac:680: the top level
autom4te: error: /usr/bin/m4 failed with exit status: 1
aclocal: error: /usr/bin/autom4te failed with exit status: 1
autoreconf: error: aclocal failed with exit status: 1

I tried the git version of luajit but it does the same.

mdblack98, 2024-12-17:
Try the tar file -- it has already done bootstrap so you can just use "configure" to proceed.

https://n0nb.users.sourceforge.net/hamlib-4.6~git-20241217-be045da06.tar.gz

Otherwise you need to remove the aclocal package and probably m4sugar too -- both are non-standard.

Mike W9MDB

c-o-m-m-a-n-d-e-r-d-a-t-a, 2024-12-17:
Success! Thank you so much. This is a great step forward as it means I can now do most of what I need using just kappanhang and rigctl.

Now I just need to work out how to send the scan channel skip command. The C-IV reference shows that I should send:
1A 05 0279 03 (to skip current channel while scanning).

Can I send 0279 amidst the rest?? If I separate it out as x02 x79 it doesn't seem to do anything (unless I'm doing it wrong).

mdblack98, 2024-12-17:
The bytes have to be 1-at-a-time.

Sounds like it should skip whatever frequency it's on until you've stopped scanning.

Mike W9MDB

closed 2024-12-17 by mdblack98