Hamlib/Hamlib #1633: Unable to compile Hamlib 4.5.5 on macOS Ventura 13.7.1 and Sequoia 15.1.1 with binding for Python 3.13

echo-gravitas, 2024-11-30:
I'm unable to compile Hamlib 4.5.5 with Python binding for Python 3.18 on macOS Ventura (13.7.1) and Sequoia (15.1.1).

Executing `./bootstrap` works fine. But `./configure --with-python-binding` fails by exiting with the message

> error: Cannot find python in your system path

Executing `./configure --with-python-binding PYTHON_VERSION='3.13'` works but fails with the message:

> Could not link test program to Python. Maybe the main Python library has been installed in some non-standard library path. If so, pass it to configure, via the LIBS environment variable. Example: ./configure LIBS="-L/usr/non-standard-path/python/lib"

Executing `./configure --with-python-binding PYTHON_VERSION='3.13' LIBS="-L/Library/Frameworks/Python.framework/Versions/3.13/lib"` fails with the same message about the lib folder even it exists at the specified location.

Now I'm stuck. `config.log` attached.
Does anybody have a hint?
[config.log](https://github.com/user-attachments/files/17966306/config.log)

mdblack98, 2024-11-30:
It looks your link flags have a partial path for the Python.framework.

configure:24167: gcc -o conftest -g -O2   -I/Library/Frameworks/Python.framework/Versions/3.13/include/python3.13  -Wl,-stack_size,1000000  -framework CoreFoundation Python.framework/Versions/3.13/Python conftest.c -L/Library/Frameworks/Python.framework/Versions/3.13/lib -L/Library/Frameworks/Python.framework/Versions/3.13/lib -lpython3.13 -ldl  -framework CoreFoundation  -ldl  -framework CoreFoundation  >&5
clang: error: no such file or directory: 'Python.framework/Versions/3.13/Python'

Try this script and see what you get
import sysconfig

# Get compile flags
cflags = sysconfig.get_config_var('CFLAGS')
print(f"Compile Flags: {cflags}")

# Get extra link flags
ldflags = sysconfig.get_config_var('LDFLAGS')
print(f"Link Flags: {ldflags}")

I get this on my development Ubuntu machine.
python3 pythonflags.py
Compile Flags: -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O2 -Wall
Link Flags: -Wl,-Bsymbolic-functions  -Wl,-z,relro -g -fwrapv -O2

mdblack98, 2024-11-30:
Also please try 4.6 -- it's just about to be released.
You can grab the tar file here
https://n0nb.users.sourceforge.net/

echo-gravitas, 2024-12-01:
After uninstalling Python 3.13 (installed with installer from python.org) and install Python 3.9.6 and swig 4.3.0 with `brew`, I was able to compile Hamlib 4.5.5 and 4.6 sucessfully. The `bootstrap` file was missing in Hamlib 4.6 (intentionally I assume), but the compiler ran successfully anyway.

At this point I didn't test Hamlib with Python and my tranceivers, but I look forward to do so in a few hours.

closed 2024-12-01 by echo-gravitas