Hamlib/Hamlib #1634: kenwood_send_morse adds an unwanted space at the end of most transmissions

dl1jbe, 2024-12-01:
First reported by OK1RR on TLF mailing list for his Elecraft K4. Confirmed by myself on Kenwood TS-590.

The following commands in rigctl can show the problem:

```
b test<cr>
b text<cr>
```

is always sent as 

```
'test text'
```

Problem seems to be the left aligned positioning of the text in the 24 characters space followed by trailing spaces.

Doing some experiments showed that any number of trailing spaces gets compressed to exactly one (additional) space. As soon as the last character in morsebuffer is no space character there will be no additional space sent. Any number of spaces before the text gets ignored.

mdblack98, 2024-12-01:
And what do you expect it to do?

Does look like the Elecraft KY doesn't need the trailing spaces like the Kenwood's do (they need all 24 bytes) but the behavior on the Kenwood is not something that Hamlib can change.
Might be that the K4 would do "KY test" "KY text" as "testtext" if we get rid of the trailing spaces...that would make the behavior different than the Kenwood.

I don't think removing the space from between two invocations of KY would be the right thing to do so I need some justification to make any change.

Mike W9MDB

dl1jbe, 2024-12-01:
If I read the code correctly at the moment the K4 gets the same handling as the normal Kenwood rigs. Other variants (K3, ...) have a special handling. Maybe the K4 should go there too. That needs to be tested by someone owning one of these.

For the Kenwood rigs it seems we just need to right align the text in the 24 characters buffer - works here at least for the TS-590.

dl1jbe, 2024-12-01:
To add something for justification: Think about a CW keyboard sending the characters via hamlib. As we do not know the timing of the writer we would have to sent each character to hamlib as soon as it came in. But in the current state there would be a space between any character you type.

GeoBaltz, 2024-12-01:
Not all Kenwoods need the full 24 bytes - the 890 and 990(with current firmware) can take variable length messages.  But the PC Control manual says they do different things with multiple spaces.  Not much Hamlib can do about that.

RE: left vs right alignment - wouldn't the right alignment add a space before each message?

Synchronicity - I just wrote the code for variable CW message length last night.

dl1jbe, 2024-12-01:
I am not sure if the PC Control manual is correct or if there are different implementations.

It states that all characters after the first 'KY ' will be sent. But in my tests here any space before the first non-space character gets ignored.

GeoBaltz, 2024-12-01:
Weird.  Looking at the descriptions for the 590, 890 and 990 KY commands, they are different, but they all say something like "Characters that are left blank will be filled with spaces, but these spaces will not be converted to Morse code".  I wonder if "left blank" means NULLs (0x00)?

dl1jbe, 2024-12-02:
TS-480: " " " (space) character must be used for unused characters. These space characters will not be converted."
TS-570: " Insert " " (ASCII code 20h) for bytes that have no characters to make a 28-byte fixed length command.
              The " " byte will  not send any character"
TS-590: "Chracters that are left blank will be filled with spaces, but these spaces will not be converted to morse code"

So 'left blank' is explicitlly named as space character.

Can you please test how 890 and 990 handles spaces before the first none-space characters, e.g.

```
KY test;
KY     text;
````

Thanks.

dl1jbe, 2024-12-02:
Got a chance to test it on TS-480. Got total different results:

- Actual implementation works without creating an additional pause for trailing spaces. 
- Right aligned text in buffer 'plays' ALL spaces befoe text - first a long pause, then the text.
- Sending an single space just stops running transmission of former text.

So, it seems that Kenwoods implementation of sending morse differs quite a lot from model to model (or even between firmware releases).

mdblack98, 2024-12-02:
So is there any fix to be done?

dl1jbe, 2024-12-02:
I fear it would only be possible if we would test and separately handle it for each different variant of implementation.
Sounds unrealistic from my side.

Sorry for making so much noise about it. :-(

GeoBaltz, 2024-12-02:
Drat - typed this before and didn't send..

Tried this on my TS-890S and found that the standard command ("KY ") always adds a space.  Added code the use "KY2" if available on TS-990S and TS-890S - no spaces between multiple messages.

dl1jbe, 2024-12-03:
Added PR #1637 to fix it for TS-590S.

dl1jbe, 2025-02-07:
Already merged. Thanks

GeoBaltz, 2025-05-15:
Just to followup on the patch - should it be for only the TS-590S, or both the 590S and the 590SG?

dl1jbe, 2025-05-16:
Am Thu, 15 May 2025 11:49:33 -0700
schrieb GeoBaltz ***@***.***>:

> GeoBaltz left a comment (Hamlib/Hamlib#1634)
> 
> Just to followup on the patch - should it be for only the TS-590S, or
> both the 590S and the 590SG?
> 
> 

I assume that the handling is the same for TS590S and TS590SG, but I
could only test it for TS590S.

It would be good to have a second voice who would be able to test it on
TS590SG.

-- 
"Do what is needful!"
Ursula LeGuin: Earthsea
--