Hamlib/Hamlib #1638: bindings test fails with SWIG 4.3.0

andreasstieger, 2024-12-08:
The bindings fail to test correctly with swig 4.3.0. This affects both 4.5.5 and master. We compared the build with swig 4.3.0 vs. 4.2.1 as the only difference, with the latter it still worked. With python 3.11.11,  gcc14 14.2.1, but probably independent from that.

````
./configure --with-perl-binding --with-tcl-binding --with-lua-binding --with-python-binding PYTHON_VERSION=3.11 --with-xml-support
````

Full logs at: https://build.opensuse.org/project/show/home:AndreasStieger:branches:hardware:sdr

Relevant snippet:

````
[  117s] Making check in bindings
[  117s] /usr/bin/make  check-am
[  117s] /usr/bin/make  -f Hamlib-pl.mk all
[  117s] make[3]: Entering directory '/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings'
[  117s] "/usr/bin/perl" -MExtUtils::Command::MM -e 'cp_nonempty' -- Hamlib.bs blib/arch/auto/Hamlib/Hamlib.bs 644
[  117s] make[3]: Leaving directory '/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings'
[  117s] /usr/bin/make  check-local
[  117s] /usr/bin/make  -f Hamlib-pl.mk all
[  117s] make[3]: Entering directory '/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings'
[  117s] LUA_CPATH="/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings/.libs/?.so" /usr/bin/lua ./luatest.lua \
[  117s] 	|| echo "Lua test failed" 1>&2
[  117s] Lua 5.4 test, Hamlib 4.5.5
[  117s] 
[  117s] /usr/bin/lua: Error in Rig::Rig (arg 1), expected 'int' got 'nil'
[  117s] stack traceback:
[  117s] 	[C]: in ?
[  117s] 	[C]: in field 'Rig'
[  117s] 	./luatest.lua:26: in function 'doStartup'
[  117s] 	./luatest.lua:130: in main chunk
[  117s] 	[C]: in ?
[  117s] Lua test failed
[  117s] make[3]: Leaving directory '/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings'
[  117s] make[4]: Entering directory '/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings'
[  117s] "/usr/bin/perl" -MExtUtils::Command::MM -e 'cp_nonempty' -- Hamlib.bs blib/arch/auto/Hamlib/Hamlib.bs 644
[  117s] make[4]: Leaving directory '/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings'
[  117s] make[3]: Entering directory '/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings'
[  117s] cp ./pkgIndex.tcl ./.libs
[  117s] TCLLIBPATH=./.libs tclsh ./tcltest.tcl \
[  117s] 	|| echo "Tcl test failed" 1>&2
[  117s] couldn't load file "./.libs/hamlibtcl-4.6": ./.libs/hamlibtcl-4.6: cannot open shared object file: No such file or directory
[  117s]     while executing
[  117s] "load ./.libs/hamlibtcl-4.6 Hamlib"
[  117s]     ("package ifneeded hamlib 4.6" script)
[  117s]     invoked from within
[  117s] "package require hamlib"
[  117s]     (file "./tcltest.tcl" line 12)
[  117s] Tcl test failed
[  117s] rm -f ./.libs/pkgIndex.tcl
[  117s] make[3]: Leaving directory '/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings'
[  117s] make[3]: Entering directory '/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings'
[  117s] LD_LIBRARY_PATH=../src/.libs perl ../bindings/perltest.pl
[  117s] Can't call method "set_conf" without a package or object reference at ../bindings/perltest.pl line 13.
[  117s] Perl 5.040000 test, version: Hamlib 4.5.5
[  117s] 
[  117s] make[3]: *** [Makefile:1111: check-perl] Error 255
[  117s] make[3]: Leaving directory '/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings'
[  117s] make[3]: *** Waiting for unfinished jobs....
[  117s] make[3]: Entering directory '/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings'
[  117s] PYTHONPATH=.:./.libs \
[  117s] 	/usr/bin/python3.11 ./py3test.py \
[  117s] 	|| echo "Python test failed" 1>&2
[  117s] ./py3test.py: Python 3.11.11; Hamlib 4.5.5
[  117s] 
[  117s] Traceback (most recent call last):
[  117s]   File "/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings/./py3test.py", line 127, in <module>
[  117s]     StartUp()
[  117s]   File "/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings/./py3test.py", line 23, in StartUp
[  117s]     my_rig = Hamlib.Rig(Hamlib.RIG_MODEL_DUMMY)
[  117s]                         ^^^^^^^^^^^^^^^^^^^^^^
[  117s] AttributeError: module 'Hamlib' has no attribute 'RIG_MODEL_DUMMY'
[  117s] Python test failed
[  117s] make[3]: Leaving directory '/home/abuild/rpmbuild/BUILD/hamlib-4.5.5/bindings'
[  117s] make[2]: *** [Makefile:937: check-am] Error 2
[  117s] make[1]: *** [Makefile:939: check] Error 2
[  117s] make: *** [Makefile:639: check-recursive] Error 1
[  117s] error: Bad exit status from /var/tmp/rpm-tmp.e5Yb0w (%check)
````

mdblack98, 2024-12-08:
They broke it in swig 4.3.0 -- it's fixed in swig 4.4.0.

closed 2024-12-08 by mdblack98

andreasstieger, 2024-12-08:
For posterity, that would be https://github.com/swig/swig/issues/3058 and https://github.com/swig/swig/commit/4e315cdd7c9072fd33ac1df3d208a0990b8231c3