Hamlib/Hamlib #1643: Python binding invalid package ierarhy

rn3kk, 2024-12-26:
After `make install` I see next tree in `python/site-packages/`
[image]

But, this is wrong ierarhy, because we cant simple `import Hamlib`. Required add `sys.path.append('/usr/local/lib/python3.10/site-packages')`  for work it. In your `py3test.py` this append also exist.

Can you support classic python module?
For example we create cratch
[image]
`__init__.py` contains `from .Hamlib import *`

mdblack98, 2025-01-19:
I just made a commit that places all but Hamlib.la and Hamlib.so in the Hamlib directory.
But if I move them into there py3test.pl no longer finds Hamlib.

mdblack98, 2025-01-19:
I had to revert this stuff as it is breaking the master build on github.
Will work on this in 4.7

dforsi, 2025-04-26:
@rn3kk can you give more information about which Linux distribution you are using, what are the contents of your sys.path and how you are testing this issue?

I can `import Hamlib` without issue if `sys.path` or `$PYTHONPATH` are set properly, without needing an `__init__.py` file.
I tested this with:

- the 4.6 module from Debian
- the module compiled from sources with the default `--prefix/usr/local`
- the module compiled from sources with `--prefix=$HOME`
Uninstalling everything between the tests.

BTW `py3test.py` is error-prone because if run as-is it will load the version of Hamlib installed system-wide, in fact `make check` sets `PYTHONPATH=$(builddir):$(builddir)/.libs` before running `p3test.py` and then there is no need to change `sys.path`.

rn3kk, 2025-04-26:
Hi!
Need time for answer. This was long.