Hamlib/Hamlib #1648: IC-7300 Hamlib Error with Ubuntu 24.04 (actually Kubuntu)

oe3gwu, 2025-01-03:
I haven't used my IC-7300 in quite some time and now wanted to use it again with WSJT 2.7.0 rc6. I use Hamlib 4.5.5. Both from the Ubuntu Repos.

I get this error, when I want to activate the CAT interface:

[image]

Sometimes something similar happen with my IC-9700. But only sometimes. TheIC-7300 complains every time.

Permissions on /dev/hamIC-7300 (an fixed alias to the IC-7300 USB Interface) and /dev/ttyUSB3 is correct. I am member of the dialout group and root is the owner.

mdblack98, 2025-01-04:
Are you building wsjtx yourself?

You need a newer hamlib which you would need to build yourself.

https://n0nb.users.sourceforge.net/hamlib-4.6~git-20241219-be045da06.tar.gz

grafik.png (view on web)

oe3gwu, 2025-01-04:
As I said above, WSJT-X and Hamlib are from the Ubuntu Repos. I dont build this stuff myself, because I usually avoid dependency hells.

mdblack98, 2025-01-04:
Try the WSJT-X improved rpm package

https://sourceforge.net/projects/wsjt-x-improved/files/WSJT-X_v2.7.1/Linux/wsjtx-2.7.1-devel_improved_PLUS_241014-RC7.x86_64.rpm/download

Mike W9MDB

oe3gwu, 2025-01-04:
WSJT-X improved has the same issue as WSJT-X 2.7.0 RC7 from the Homepage with Ubuntu 24.10. It doest recognise that I got libboost-log in a newer version. I guess the Ubuntu Repo Version is customized.

[image]

mdblack98, 2025-01-04:
Hmmm -- I get this

Looks like you should be able to do
apt install libboost-filesystem1.74.0 libboost-log1.74.0 libboost-regex1.74.0 libboost-thread1.74.0

apt install ./wsjtx_2.7.1-devel_improved_PLUS_241014-RC7_amd64.deb 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'wsjtx' instead of './wsjtx_2.7.1-devel_improved_PLUS_241014-RC7_amd64.deb'
The following additional packages will be installed:
  libboost-filesystem1.74.0 libboost-log1.74.0 libboost-regex1.74.0 libboost-thread1.74.0
The following NEW packages will be installed:
  libboost-filesystem1.74.0 libboost-log1.74.0 libboost-regex1.74.0 libboost-thread1.74.0 wsjtx
0 upgraded, 5 newly installed, 0 to remove and 3 not upgraded.
Need to get 1,640 kB/22.7 MB of archives.
After this operation, 80.4 MB of additional disk space will be used.

grafik.png (view on web)

oe3gwu, 2025-01-04:
It doest work that way! :-)

[image]

mdblack98, 2025-01-04:
You must be missing some repositories

Try
apt search libboost-filesystem

oe3gwu, 2025-01-04:
I am not missing a repo. 1.74 is not part of the standard repos. And I dont use backports. Again thanks for the help, but I just want Hamlib working on my (for me perfect) WSJT-X setup.

mdblack98, 2025-01-04:
Did you search for the library -- hopefully there's some version of it out there.

oe3gwu, 2025-01-04:
Ubuntu only ships 1.83. Not 1.74. But still I dont know what this has to do with my Hamlib issue?

mdblack98, 2025-01-04:
Try this repository

http://us.archive.ubuntu.com/ubuntu

 apt policy libboost-filesystem1.74-dev
libboost-filesystem1.74-dev:
  Installed: (none)
  Candidate: 1.74.0+ds1-23.1ubuntu3
  Version table:
     1.74.0+ds1-23.1ubuntu3 500
        500 http://us.archive.ubuntu.com/ubuntu noble/universe amd64 Packages

mdblack98, 2025-01-15:
Try turning off the echo in the rig.
Set/Connectors/CI-V USB Echo Back = OFF

N0NB, 2025-08-01:
Any success with a newer version of of WSJT-X and or Hamlib?

oe3gwu, 2025-08-01:
I dont update outside of my repo when stuff in the ubuntu repo is present. But with the Ubuntu 25.04 Release, everything is working currently

N0NB, 2025-08-01:
Thanks for the update, Rainer.  I will close this.