Hamlib/Hamlib #1652: uSDX CAT error in WSJT-X

brightkill, 2025-01-06:
Greetings!

My transceiver - uSDX (uses Kenwood TS-480 protocol with hamlib)
After applying CAT settings in WSJT-X i have a hamlib error

>  read_string_generic called, rxmax=4095 direct=1, expected_len=1
> write_block(): TX 14 bytes, method=2
> 0000    46 41 30 30 30 31 34 30 37 34 30 35 35 3b           FA00014074055;  
> write_block(): TX 3 bytes, method=2
> 0000    49 44 3b                                            ID;             
> read_string_generic called, rxmax=51 direct=1, expected_len=1
> read_string_generic(): RX 30 characters, direct=1
> 0000    ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff     ................
> 0010    ff ff ff ff ff ff ff ff 49 44 30 32 30 3b           ........ID020;  
> kenwood_transaction: read_string(expected=51, len=30)='������������������������ID020;'
> kenwood_transaction: No data expected, checking ID; in ������������������������ID020;
> kenwood_transaction: wrong reply �� for command verification ID
> kenwood.c(624):kenwood_transaction returning2(-8) Protocol error
> 
> kenwood.c(1908):kenwood_set_freq returning2(-8) Protocol error
> 
> 0:rig.c(1814):rig_set_freq returning(-8) Protocol error
> 
> Protocol error
> Protocol error
>  while setting frequency
> 
> Timestamp: 2025-01-06T21:37:08.617Z

Test CAT button lights with green and Test PTT lights with red but PTT worked on transceiver.

mdblack98, 2025-01-06:
What version of WSJT-X?
What version of Hamlib?

brightkill, 2025-01-11:
> What version of WSJT-X? What version of Hamlib?

WSJT-X 2.7.0rc8
Hamlib 4.6~git

brightkill, 2025-01-11:
@mdblack98 now i have this bug while uSDX transmit at 28.074 MHz. i've built the git master last branch with this commit and seems to work fine

closed 2025-01-11 by mdblack98