Hamlib/Hamlib #1653: Hamlib handles FLRig modes incorrectly

radio-miskovice, 2025-01-07:
Hamlib handles information received from FLRig in an incorrect way.

When I connect QRP Labs QMX transceiver to js8Call (which uses Hamlib to control CAT, even to when accessing through FLRig), I can see two types of errors reported from Hamlib to stderr:

```
Hamlib: flrig_openL Unknown mode (new?) for this rig='DIGI-L'
Hamlib: flrig_openL Unknown mode (new?) for this rig='DIGI-U'
```

The connected rig, QMX, has four possible modes (as of fw 1.0.27): CW, CW-R, DIGI-U and DIGI-L. These are mapped to MD3, MD7, MD6 and MD9 in Kenwood-like command set. The reason why QMX reports DIGI-U and DIGI-L instead of the conventional FSK ans FSK-R is probably due to the fact that QMX is a SDR rig, mostly used for FT8, JS8 and similar modes (single-frequency two-tone FSK or multi-tone MFSK) and has nothing to do with the archaic method of direct keying two-tone FSK. 

So my first objection to Hamlib behavior is that it should accept mode name as-is as it is received from FLRig. It is not up to Hamlib to invent or enforce this. This is correct and true information provided by FLRig.

Later on, Hamlib reports every couple of seconds the following:
```
Hamlib: modeMapGetFLRig: FlRig does not have mode: USB
```
Of course, this is true because QMX indeed has no USB mode (nor it has LSB for that matter), it has DIGI-U instead. This is another thing that Hamlib inapproprietly meddles in - it reports to JS8Call that USB mode is available on the transceiver for JS8 communication, but this is not true, and so when JS8Call requires - via Hamlib to FLRig - that the rig switches to USB mode, it obiously must fail.

I would like to propose the following:

- when talking to FLRig, Hamlib should accept whatever modes are reported by FLRig. The main reason is that Hamlib omits a number of transceiver models while FLRig is pretty up to date, even considering many nuances (e.g. QRP Labs transceivers use enhanced Kenwood command set,  but, for instance, QDX differs from QMX, and FLRig handles them both with astonishing accuracy).
- in general, mode setting (at least), if performed through Hamlib, should be absolutely transparent between the ham software and FLrig because in the current state the Hamlib creates phantom errors that would otherwise not occur.

mdblack98, 2025-01-07:
While it's obvious that we need to fix this is really FLRig's fault for introducing a new mode.  I'll get Dave to fix FLRig.
Hamlib cannot blindly accept modes that it does not know about as they have to be mapped to internal representations and DIGI-L/U is unique to QMX.  And Dave (FLRig) didn't even use QMX's mode names of FSR/FSK and made up his own.
Everybody else uses DIGL or DATA-USB
And QMX should also accept USB mode and map itself to DIGI-U if it wants.  Then it would be TS480 compatible.
Meanwhile I will update Hamlib to show USB is not available for the QMX.  It will have to be a new rig entry.

mdblack98, 2025-01-07:
While I'm doing this I may need some questions answered as the QMX manual is lacking.
What baud rates does it support?  Makes it sound like it should support anything from 9600 to 256,000.

radio-miskovice, 2025-01-08:
> While it's obvious that we need to fix this is really FLRig's fault for introducing a new mode. I'll get Dave to fix FLRig. Hamlib cannot blindly accept modes that it does not know about as they [...]

Hello, sorry I have missed your name and at the same time I apologize for not having properly signed my issue post... Anyway

Please make sure you have the right understanding of the situation. It was not Dave who introduced the new modes, it was Hans, who designed it in the QMX firmware. IMHO Dave just respectfully implemented what has been implemented in QMX without his (Dave's) decision or prior approval. I also share Hans' (G0UPL, designer of QMX) view that the DIGI-U and DIGI-L is not the same as FSK and FSK-R, because these two modes have been traditionally based on hardware keying of frequency source, while DIGI-x takes digital input and changes transmitted frequency based on the data received. Data stream versus wire short circuit. The fact that it appears in the CAT command set as MD6 and MD9, respectively, is an understandable compromise in order not to break the Kenwood CAT command logic too much.

As for the "bitrate", as is explained in the manual, there is no physical serial port to be used in QMX. The CPU implements comm port CDC and receives data directly from the data stream on USB, therefore you can set bitrate to whatever you want, it does not matter. Bitrate would be effective only if there was a USB to serial adapter like CH340 or CP2102 and physical USART in the CPU connected to it. This is not the case.

 Finally, a friendly, well-meant remark from a fellow ham. I am afraid you show the somewhat arrogant stance I have already noted in some of the developers of ham software (Dave, W1HKJ, is a good example of one of the exceptions, respectful and considerate). You are trying to push others to accept your view of the world, at the same time you don't understand all the realities of the world. You would do better if you followed example of Dave, W1HKJ and some others.
 
 As for the matter of this issue, let me summarize. It is as simple as this:
 
 - manufacturer of ham radio transceiver defines transmission modes according to the design and identifies them by certain names
 - these mode names are printed in the manual and displayed in the rig's display in the corresponding menu
 - the owner of the rig wants to use it for a particular purpose, e.g. FT8, FT4, JS8 and knows from the manual what is the mode the rig has to be switched to, following instructions in the manual and/or settings displayed on the rig's display
 - the task of the software (e.g. JS8Call, WSJT) is to let the owner select the appropriate mode in configuration, so that they can operate the rig by means of the software
 - in order to facilitate this, the task of libraries like Hamlib or utilities like FLRig is to convey the correct set of available mode names to the user/operational software, let the operator choose whatever is appropriate and then use CAT to set the rig accordingly
 - when the manufacturer of a rig gives transmission mode certain name, it is not your job to rename it, push others to change it or otherwise meddle in the indirect communication between the rig designer and its final user. You are not the rig designer nor you are the user's daddy.

I hope it is clearer now.

73 and all the best in the new year,
Jindra
OK4RM

radio-miskovice, 2025-01-08:
Hello, Michael,

thanks for the quick reaction. I have seen the problem when using JS8Call on Raspberry Pi but I can try the DLL on my Windows PC. As for the Raspberry, I could try to build hamlib from source, I already had a successful build when compiling JS8Call.

As for the comm port, there is no baudrate involved. The CPU implements serial port CDC and reads the data directly without using physical comm port so it really does not matter at all. 

As for the differences between QDX and QMX, that's another story. I have built both (well, finished a kit :) ) and there is a difference between the two. QDX is older, also uses a subset of Kenwood CAT and the modes available are MD1 and MD2, LSB and USB, respectively.

QMX is a more ambitious project, hence it has CW, CW-R, DIGI-U (aka FSK) and DIGI-L (aka FSK-R) but the modes for LSB and USB are reserved for future use, when Hans completes SSB modulation of any analog signal (traditional voice SSB and digital modes requiring amplitude envelope).

I hope it helps,
Jindra 
OK4RM

radio-miskovice, 2025-01-08:
Oh, you are right, I was wrong. I apologize. The mode indicator "DIGI" appears only in the display (e.g. picture at page 69) and elsewhere in the menu it is indicated whether it uses lower or upper sideband. While looking at it every day I really believed this was in the manual, too.  Dave was probably driven by the same impression (but I dod not ask him yet) - what you see on the rig display and in FLrig is the same text (almost). It still makes sense to me.
I think it is similar to TS-590: to activate digi modes, you select USB (or LSB...) and then press DATA button. The display shows "USB" and "DATA" nearby. And I think in FLRig this is shown as USB-D or LSB-D, so it still looks familiar to the user, nobody should be confused. And it is distinct from "plain" USB and LSB modes which use different audio input, different filter setup, etc.

mdblack98, 2025-01-10:
FYI...Hamlib now recognizes DIGI-L and DIGI-U and Dave changed FLRig to use DIGIL and DIGIU
So both sides have changed.

closed 2025-01-10 by mdblack98

radio-miskovice, 2025-02-01:
Hi, I just built the latest hamlib from the repo and now it correctly reports and handles QMX. Many, many thanks! Hans, G0UPL, has been promising SSB mode to be finally implemented (supposedly this year) so I will let you know if there are new changes to th firmware (causing extension of the existing CAT command set, at least the valid values in MD command).