Hamlib/Hamlib #1655: rig.h : import  of  <sys/socket.h> breaks the $grig$ build

MarcFontaine, 2025-01-12:
In version Hamlib-4.6 `rig.h` imports `<sys/socket.h>` . This brings `connect` from `socket.h` into scope.
*grig* imports `rig.h` and at the same time implements some other unrelated `connect` function and
this breaks the build of *grig* with Hamlib-4.6.
Comment in `rig.h` suggest that `<sys/socket.h>` is only needed for internal $Hamlib$ use.

MarcFontaine, 2025-01-13:
Thanks for looking at the issue.
The `grig` issue showed up when updating the Hamlib version that is included in NixOS
from 4.5.5 to 4.6
https://github.com/NixOS/nixpkgs/pull/372230.
`grig` is the only package that depends on Hamlib in NixOS that breaks.
`cqrlog`,  `direwolf`, `fldigi`, `freedv`, `gpredict` , `jtdx` , `openwebrx` ,
`openwebrx.dist` ,  `wsjtx` and `wsjtz` build fine.

I tested a bit more and each of the three imports
```
#include <sys/socket.h>
#include <netinet/in.h>
#include <arpa/inet.h>
```
triggers the same error in the `grig` build
and remove all three breaks the Hamlib build itself.

mdblack98, 2025-01-13:
grig needs to be fixed.

What error do you get building grig?

NixOS/nixpkgs#372230.

mdblack98, 2025-01-13:
Check this thread for the needed grig change.  They had to rename some functions to avoid colliding with the Unix functions.

mdblack98, 2025-01-13:
Forgot the link
https://github.com/fillods/grig/issues/22
current code is here
git clone https://github.com/fillods/grig.git

closed 2025-01-31 by mdblack98