Hamlib/Hamlib #1658: IC-7610 needs ability to set D1,D2,D3

mdblack98, 2025-01-15:
Maybe something like this:
-- a/rigs/icom/icom.c
+++ b/rigs/icom/icom.c
@@ -2400,11 +2400,22 @@ static int icom_set_mode_x26(RIG *rig, vfo_t vfo, rmode_t mode,
     if (priv->filter_usbd > 0 && (mode == RIG_MODE_PKTUSB
                                   || mode == RIG_MODE_PKTLSB))
     {
+        int twidth, tmode;
         rig_debug(RIG_DEBUG_TRACE, "%s: filter usbd=%d, width=%d\n", __func__,
                   priv->filter_usbd, (int)width);
         buf[2] = priv->filter_usbd;
 
-        if (width >= 1 && width <= 3) { buf[2] = width; }
+        // IC-7610 for example had D1,D2,D3
+        // So we allow a width of two nibbles AB where A=1,2,3 for D1,D2,D3 and B = 1,2,3 for filter numbers
+        // Filter number is used by many rigs
+        if (width <= 33)
+        {
+        twidth = width & 0x03;
+        tmode = width / 10;
+        if (twidth >= 1 && twidth <= 3) { buf[2] = twidth; }
+        if (tmode >= 1 && tmode <= 3) { buf[1] = tmode; }
+        width = RIG_PASSBAND_NOCHANGE;
+        }

guitarpicva, 2025-02-17:
Probably same idea as is already implemented for IC-F8101.

mdblack98, 2025-02-17:
Should work for all Icom radios.

For width you use 1,2,3 instead of the normal width

If you want WSJT-X to do it that' s another story.  Might be able to figure out a new config file setting that would do it.

Mike W9MDB