Hamlib/Hamlib #1661: Broken links in documentation

mhassell, 2025-01-20:
Hello, I am looking through the documentation to see what the standard commands are for rigctl but when I follow the urls I get a 404 error.  Specifically, the 404 occurs with the links here:

> For complete documentation of the command line options and control commands for the daemons, see the Hamlib Utilities section of the [Documentation] page, or jump right to the [rigctld section](http://hamlib.sourceforge.net/manuals/hamlib.html#rigctld) followed by [rigctld commands](http://hamlib.sourceforge.net/manuals/hamlib.html#rigctld-commands) of the [rigctld protocol](http://hamlib.sourceforge.net/manuals/hamlib.html#rigctld-protocol). There are PDF files of each utility available for ease of viewing and printing.

mdblack98, 2025-01-20:
What is the parent URL where you are getting those references?

mhassell, 2025-01-20:
https://github.com/Hamlib/Hamlib/wiki/Network-Device-Control

mdblack98, 2025-01-20:
Fixed.   Let me know if you find any others.

closed 2025-01-20 by mdblack98