Hamlib/Hamlib #1669: BUG: Python bindings are missing RIG_MODEL_*

petrkr, 2025-01-29:
Do not know which version caused it, but in rigctl Hamlib 4.5.5 with Hamlib.py Version 4.3.0 are missing every RIG_MODEL_* That make error due initialize Rig() class

```
  File "/home/petrkr/git/ham/pyvara/test.py", line 75, in main
    rig = initialize_rig("localhost")
  File "/home/petrkr/git/ham/pyvara/test.py", line 12, in initialize_rig
    rig = Hamlib.Rig(Hamlib.RIG_MODEL_NETRIGCTL)
```

Version 4.1.0  of Hamlib.py those models have

[image]

mdblack98, 2025-01-29:
Try Hamlib 4.6.1
Mike W9MDB

 
 

image.png (view on web)

petrkr, 2025-01-29:
I can not, 4.5.5 is latest available version in arch linux repository. I do
not want break system packages by installation fron sources as it is part
if system python site-packages.

So I have to downgrade and wait for mainteners to update packages.

Will try to remember and will write.

petrkr, 2025-01-30:
I tested build package from 4.6.1 source and version of Hamlib.py is still 4.3.0 without RIG_MODEL_*

Because Arch linux use 4.5.5 since April 2023 (https://gitlab.archlinux.org/archlinux/packaging/packages/hamlib/-/commits/main?ref_type=HEADS)

I think it would be connected with update to Python 3.13.  Because I am sure I updated system since 2023 many times, so I must had 4.5.5 already, but those days there was python 3.11 and 3.12.. So only change was python 3.13 at December 2024

But I do not know if I can compile older python in arch and somehow have more versions beside

petrkr, 2025-01-30:
Ok... So I already found where issue is.

problem is with updated swig from 4.2.1 to 4.3.0

there are many unsupported constants values during swig python code binding creating

I tried to downgrade to 4.2.1 and that does not complain about this one and Hamlib.py is generated fine.

So probably there is need to update those header files.

```
../include/hamlib/riglist.h:80: Warning 304: Unsupported constant value (ignored)
../include/hamlib/riglist.h:82: Warning 304: Unsupported constant value (ignored)
../include/hamlib/riglist.h:83: Warning 304: Unsupported constant value (ignored)
...
```

mdblack98, 2025-01-30:
That is a bug in swig 4.3.0 -- it's been fixed in 4.4.0 -- you can grab the git code and it should work fine.
Mike W9MDB

mdblack98, 2025-01-30:
> Because 4.4.0 is not released yet, you will not FIX macros (as described in swig issue tracker), arch linux and I will not install git-based packages to stable system.
>
> So it means until swig is released and passed to stable arch linux packages, util that I am not able to use python based projects..
>
> Good,.. so now I have to somehow downgrade half of dependencies and remember to not update, or scratch that plannef SSTV transmission...
>
> Damn... if I will knew simple system update will do such mess, I will not do it....
>
> Now lot of work with rollbacks.
>
> Anyway thanks for letting me know where issue is.

I can't fix the macros -- there's nothing to fix.  Macros have been like that since Hamlib was invented.
Sorry that swig broke things.
Mike

closed 2025-01-30 by mdblack98

petrkr, 2025-01-31:
Why you closed it? It is NOT fixed yet. Production system still not works. 

https://github.com/swig/swig/issues/3058#issuecomment-2439696279

mdblack98, 2025-01-31:
It is fixed in swig 4.4.0Has nothing to do with Hamlib.

 
 

swig/swig#3058 (comment)

petrkr, 2025-01-31:
No it is definitvely not fixed, 4.4.0 does not exists. In swig article is you have to change macro. It's like hot-potato between two projects and only harm is end-user who can not use it, because this is critical bug where is no workaround possible.

So what I have to do to be able transmit this weekend? And also satisfaction all lelal stuff like ISO certificates, which I need, so really there is no way to use source code and break production system.

mdblack98, 2025-01-31:
It cannot be fixed in Hamlib -- it's a swig bug.   You need to push them to publish it.
Mike

petrkr, 2025-03-22:
Just got few time for external projects and also I really need to use this library in python now. So  I tried that mentioned patch https://github.com/swig/swig/issues/3058#issuecomment-2439696279

which fixes your header files and now it works even with swig 4.3.0.

I already filled bug report on package in Arch lilnux, so I hope they will use that patch until you both with swig decide who will fix it earlier.

If I will be developer of Hamlib, I will do that header change, regardless if that is or that is not my mistake, but just to satisfy user, who can not use this library for 3 months with python already.

Regards
Petr