Hamlib/Hamlib #1674: What is the correct syntax for the send_raw command to icom ?

eicket, 2025-02-23:
Is there anyone who can provide me with the correct syntax for the send_raw command to an ICOM rig please ?

I tried the commands below on Hamlib 4.6.1 and 4.6.2 

send_raw icom \0xFE\0xFE\0x88\0xE1\0x06\0x03\0xFD
send_raw icom xFE:xFE:x88:xE1:x06:x03:xFD
send_raw 0xFE;0xFE;0x88;0xE1;0x03;0xFD
w \0xFE\0xFE\0x88\0xE1\0x06\0x03\0xFD

All of these commands seem to have a problem in recognizing the send_cmd_term character and go into a timeout after about 1.4 secs, without returning any response ..

Thanks for any help !
73's Erik
ON4PB

N0NB, 2025-04-03:
Erik, did Mike's suggestion work?

N0NB, 2025-06-22:
@eicket I suppose that no response means all is well.

Feel free to reopen this if needed.

closed 2025-06-22 by N0NB