Hamlib/Hamlib #1684: rigctld reports improper exit code if TCP port is unavailable and does not report error message

radio-miskovice, 2025-03-18:
### Situation: 

When testing some software I could not start rigctld. Instead of running it exited after a few milliseconds and exit code was 1, no error message was printed on console.

According to documentation exit code 1 means incorrect command line parameter. However, this was not the case. However, the command line was the same as in previous cases when it worked. There was no change in the system or in rigctld version.

Problem was caused by "forgotten" instance of rigctld, i.e. one instance was already running and therefore port 4532 could not be bound and the second instance of rigctld exited with exit code 1.

### Problem(s): 

1. There is no error message in case of this kind of critical error except when verbose command line parameter is selected. 
2. The exit code is 1 but this exit code has a different meaning on documentation. 

### Proposed solution:

1. This kind of condition should be printed to console or log regardless of the selected level of verbosity because it prevents the program to exit immediately and hence it does not provide the expected functionality at all. The user should be informed what os the cause of the critical error.
2. The exit code should not be 1 but a different number and it should be properly noted in the documentation.

### Program version and OS:

rigctld Hamlib 4.7~git 2025-02-05T14:01:14Z SHA=c939f1 64-bit

Linux tecra-compute 6.12.9+bpo-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.12.9-1~bpo12+1 (2025-01-19) x86_64 GNU/Linux

mdblack98, 2025-03-19:
Add a couple of -v's to see error messages.

rigctld -vv
main: bind: Address already in use
listening: Network error 9: Bad file descriptor

closed 2025-03-19 by mdblack98

radio-miskovice, 2025-03-19:
Michael, I know all of this. Question for you and for the development team: Who are you developing it for? Hackers and geeks or ordinary users?

Thank you,
Jindra