Hamlib/Hamlib #1685: rigctl power2mW parsing failure

kyleboyle, 2025-03-23:
Possibly incorrect recognition of the power2mW command string. Using the shortcode (2) works.

Probably related: mW2power will crash the rigctl process, but the short code 4 works.

```
rigctl.exe -m 1049 -r COM4 -s 115200

Rig command: m f l RFPOWER
Mode: LSB
Passband: 3000

Rig command:
Rig command: Frequency: 7093300

Rig command:
Rig command: Level Value: 0.8

Rig command: 2 0.8 7093300 LSB
Mode: Power mW: 80000

Rig command: power2mW 0.8 7093300 LSB
error = rig_parse_mode called
**2:rig.c(6655):rig_power2mW entered
***3:newcat.c(3517):newcat_power2mW entered
****4:newcat.c(11034):newcat_get_rigid entered
rig_id = 800
****4:newcat.c(11057):newcat_get_rigid returning(800)
default - rig_id = 800, *mwpower = 80000
***3:newcat.c(3595):newcat_power2mW returning(0)
**2:rig.c(6659):rig_power2mW returning(0)
rigctl_parse.c(3284):rigctl_power2mW returning2(0)
rigctl_parse: called, interactive=1
rigctl_parse: cmd=p(70) handle=0
rigctl_parse: vfo_opt=0
rigctl_parse.c(3822):rigctl_get_parm entered
rig_parse_parm called
rig_has_get_parm called
rig_ext_lookup called
rigctl_parse.c(3842):rigctl_get_parm returning2(-1) Invalid parameter

Invalid parameter

```

verbose failure:
```
rigctl.exe -m 1049 -r COM4 -s 115200 -vvvv
rigctl.c(532) Startup: C:\hamlib-w64-4.6.2\bin\rigctl.exe -m 1049 -r COM4 -s 115200 -vvvv
rigctl Hamlib 4.6.2 2025-02-09T21:03:50Z SHA=870364 64-bit
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_test_2038: MSVCRT 2038 test = 0xf0000000:Mon Aug 05 06:04:00 2097
rig_init: 2038 time test passed
yaesu: initrigs4_yaesu called
rig_init: rig_model=Yaesu FT-710 20241118.7
rig_init: rig has VFO_A
rig_init: rig has VFO_B
rig_init: rig has VFO_MEM
**2:newcat.c(477):newcat_init entered
**2:newcat.c(522):newcat_init returning(0)
rig_set_conf called
rig_confparam_lookup called for rig_pathname
rig_set_conf: rig_pathname='COM4'
rig.c(947):rig_open entered
serial_open: serial port COM4 is OK
serial_open: COM4
**2:rig.c(8402):async_data_handler_start entered
**2:rig.c(8409):async_data_handler_start returning(0)
rig_open: 000000000275744c rs->comm_state==1?=1
**2:rig.c(6882):rig_get_powerstat entered
[... 171 lines cut ...]
rigctl_parse: vfo_opt=0
rigctl_parse.c(3822):rigctl_get_parm entered
rig_parse_parm called
rig_has_get_parm called
rig_ext_lookup called
rigctl_parse.c(3842):rigctl_get_parm returning2(-1) Invalid parameter

Invalid parameter
```

verbose success:
```
Rig command: 2 0.8 7093300 LSB
Mode: rigctl_parse.c(3264):rigctl_power2mW entered
rig_parse_mode called
**2:rig.c(6655):rig_power2mW entered
***3:newcat.c(3517):newcat_power2mW entered
****4:newcat.c(11034):newcat_get_rigid entered
****4:newcat.c(11057):newcat_get_rigid returning(800)
***3:newcat.c(3595):newcat_power2mW returning(0)
**2:rig.c(6659):rig_power2mW returning(0)
Power mW: 80000
rigctl_parse.c(3284):rigctl_power2mW returning2(0)

```

mdblack98, 2025-03-24:
When you use the long names you have to start with a backslash.

\power2mW 0.8 7093300 LSB

Mike W9MDB

kyleboyle, 2025-03-24:
Ah, thank you.

closed 2025-03-24 by kyleboyle