Hamlib/Hamlib #1686: Build instructions for Windows 64-bit (not 32-bit exe)

radio-miskovice, 2025-03-29:
Hello, I cannot find build instructions for building a native 64-bit applications.

N0NB, 2025-04-03:
Does the info beginning at line 163 of https://github.com/Hamlib/Hamlib/blob/master/scripts/build-w64.sh help?

N0NB, 2025-04-04:
Please tell me which line worked for you.

I have been looking for feedback/corrections on that section of the file.  I'm even setting up a Windows 10 VM to try and figure out if the `/machine:i386` option is correct.  Should it be something else for a 64 bit build?

I want to get these instructions correct as that section is included in the Windows binaries that are generated for each release.

radio-miskovice, 2025-04-04:
Nate, unfortunately we have emcomm drills this wekend so I am not certain how much I will be able to test. Probably the easiest is to start with Visual Studio, I have VS2022. I just need to understand how to "capture" the source code in VS. With software originally written for build on Linux I have never tried this. At the same PC I have MSYS2, the source code is in another path, of course. In my latest attempt to build it in MSYS2 I got some errors (duplicate definitions) but I did not go further because at this moment we have everything for the forthcoming drills in Linux.  
I think on Sunday afternoon I should probably have at least one variant checked.

N0NB, 2025-04-04:
Thanks.

Good job on participating in an emcomm drill.  We're probably woefully unprepared here.

I wanted to find out what the exact machine type would be and dug into MS' Web site last night to find out and when I drilled down it said to run `lib.exe /?` for the complete list.  Gee, thanks, MS.  🙄

I'll keep this issue open until I'm reasonably certain that the command string is correct.

N0NB, 2025-04-04:
I managed to get the VM setup and going (what a pain).

From the directory the `.def` file is installed:

```
C:\Program Files\hamlib-w64-4.7~git\lib\msvc>dir
 Volume in drive C has no label.
 Volume Serial Number is 98C4-78B5

 Directory of C:\Program Files\hamlib-w64-4.7~git\lib\msvc

04/04/2025  11:04 PM    <DIR>          .
04/04/2025  11:04 PM    <DIR>          ..
04/04/2025  04:39 AM            14,896 libhamlib-4.def
04/04/2025  04:39 AM           449,012 libhamlib-4.lib
               2 File(s)        463,908 bytes
               2 Dir(s)  20,796,633,088 bytes free

C:\Program Files\hamlib-w64-4.7~git\lib\msvc>
```

I think I'm close on the command syntax but must be missing something:

```
C:\Program Files\hamlib-w64-4.7~git\lib\msvc>"c:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.43.34808\bin\Hostx64\x64\lib.exe" /machine:X64 /def:"C:\Program Files\hamlib-w64-4.7~git\lib\msvc\libhamlib-4.def" /out:"C:\Program Files\hamlib-w64-4.7~git\lib\msvc\libhamlib-4.exp"
Microsoft (R) Library Manager Version 14.43.34809.0
Copyright (C) Microsoft Corporation.  All rights reserved.

LINK : fatal error LNK1154: specified import library filename matches exports file 'C:\Program Files\hamlib-w64-4.7~git\lib\msvc\libhamlib-4.exp'
```

I can't find documentation from MS for "LNK1154".  Adding `/libpath:"C:\Program Files\hamlib-w64-4.7~git\lib\msvc"` to the command line made no difference on the error message.  The error seems nonsensical to me.  I even tried changing the output name by adding a '7' and still the same error.

I'll keep doing some searches.

N0NB, 2025-04-04:
BTW, here is the help output for `lib.exe`.

```
C:\Program Files\hamlib-w64-4.7~git\lib\msvc>"c:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.43.34808\bin\Hostx64\x64\lib.exe" /?
Microsoft (R) Library Manager Version 14.43.34809.0
Copyright (C) Microsoft Corporation.  All rights reserved.

usage: LIB [options] [files]

   options:

      /DEF[:filename]
      /ERRORREPORT:{NONE|PROMPT|QUEUE|SEND}
      /EXPORT:symbol
      /EXTRACT:membername
      /INCLUDE:symbol
      /LIBPATH:dir
      /LINKREPRO:dir
      /LINKREPROFULLPATHRSP:filename
      /LINKREPROTARGET:filename
      /LIST[:filename]
      /LTCG
      /MACHINE:{ARM|ARM64|ARM64X|EBC|X64|X86}
      /NAME:filename
      /NODEFAULTLIB[:library]
      /NOLOGO
      /OUT:filename
      /REMOVE:membername
      /SUBSYSTEM:{BOOT_APPLICATION|CONSOLE|EFI_APPLICATION|
                  EFI_BOOT_SERVICE_DRIVER|EFI_ROM|EFI_RUNTIME_DRIVER|
                  NATIVE|POSIX|WINDOWS|WINDOWSCE}[,#[.##]]
      /VERBOSE
      /WX[:NO]
      /WX[:nnnn[,nnnn...]]
```

I miss man pages.

K7MDL2, 2025-04-20:
This page may also help, it identifies a long list of target machine types.

[https://learn.microsoft.com/en-us/dotnet/api/microsoft.visualstudio.vcprojectengine.machinetypeoption?view=visualstudiosdk-2022](url)

In the lib.exe docs (which I think you found)
[https://learn.microsoft.com/en-us/cpp/build/reference/running-lib?view=msvc-170](url)
this suggests you normally do not need to specify /MACHINE

I do not think Windows is case sensitive but these are all in upper case.

/MACHINE

Specifies the target platform for the program. Usually, you don't need to specify /MACHINE. LIB infers the machine type from the .obj files. However, in some circumstances, LIB can't determine the machine type and issues an error message. If such an error occurs, specify /MACHINE. In /EXTRACT mode, this option is for verification only. Use lib /? at the command line to see available machine types.

I could not find LNK 1154 either but searching with the error text "specified import library filename matches exports file"  gave some hints.
[https://learn.microsoft.com/en-us/cpp/build/reference/using-an-import-library-and-export-file?view=msvc-170](url)

You likely have a circular reference, in this case maybe the order of which file is linked first is the key.  Or maybe the filename extensions are not considered, or changed, so only the filename counts and must be different?  Don't know.

- Mike K7MDL

N0NB, 2025-04-20:
Thanks, Mike.

Almost three weeks ago I opened a discussion thread on the MinGW64 project forum:  https://sourceforge.net/p/mingw-w64/discussion/723798/thread/e23dceba20/?limit=25#b31f

I received one reply a little over a week ago but I'm still no further in my understanding.

N0NB, 2025-04-26:
The `libusb` project https://github.com/libusb/libusb may be a model for what we want to achieve.

gvanem, 2025-10-26:
> Does the info beginning at line 163 of https://github.com/Hamlib/Hamlib/blob/master/scripts/build-w64.sh help?

Why not make Hamlib compile with MSVC instead of this terrible MinGW-w64 hack?

N0NB, 2025-10-27:
I've no objection so long as GNU/POSIX support is not broken in the process.  I don't use or even know how to use MSVC nor do I ever intend to, so I'm not the guy to make those changes.  Someone will need to do that work and test it extensively.  Patches/pull requests welcome.