Hamlib/Hamlib #1698: Kenwood TM-D710 Error

promo776, 2025-04-19:
Homebrew hamlib 4.6.2

Console output:

rig_init: backend for TM-D710(G) does not contain hamlib_check_rig_caps
kenwood_transaction: Command is not correctly terminated 'BC 0,0'
kenwood_transaction: Command is not correctly terminated 'BC 0,0'
kenwood_transaction: Command is not correctly terminated 'BC 0,0'
rig_get_vfo: returning -8(
STATE(rig)->tx_vfo: VFOA
**2:rig.c(3452):rig_get_vfo entered
rig_get_vfo: cache miss age=1000000ms
**rig.c(3488) trace
tmd710_get_vfo: called
tmd710_get_vfo_and_mode: called
tmd710_get_vfo_num: called
kenwood_transaction called cmd=BC
kenwood_transaction: cmdstr = BC
read_string_generic called, rxmax=4095 direct=1, expected_len=1
write_block(): TX 3 bytes
read_string_generic called, rxmax=11 direct=1, expected_len=1
read_string_generic(): RX 7 characters, direct=1
'enwood_transaction: read_string len=7 'BC 0,0
kenwood_transaction: Command is not correctly terminated 'BC 0,0'
kenwood.c(711):kenwood_transaction returning2(-8) Protocol error

Protocol error
Protocol error
)
kenwood_transaction: Command is not correctly terminated 'FO 0,0441500000,0,0,0,0,0,0,13,41,051,05000000,1'
Opened rig model 2034, 'TM-D710(G)'

promo776, 2025-04-19:
It seems like the issue is related to _transaction_read_ in **kenwood.c**:

```
transaction_read:
    /* allow room for most any response */
    // this len/expected stuff is confusing -- logic in some places includes the semicolon
    // so we add 1 to our read_string length to cover these cases
    // eventually we should be able to get rid of this but requires testing all Kenwood rigs
```

When I look at the output from a serial console, the D710 is not sending a semicolon:

BC 0,0

larsks, 2025-06-17:
@promo776 please try out the change in #1768; that should resolve the error.

N0NB, 2025-06-18:
Fixed by PR #1768.

closed 2025-06-18 by N0NB