Hamlib/Hamlib #1700: TT Jupiter 538

rexhunt, 2025-04-20:
Trying to get a Tentec 538 Jupiter to work with Hamlib, Ive started using rigctl for debugging.

The radio does not accept the first command, but correctly acts on a second one. ie
`Rig command: m
get_mode: error = rig.c(1644):rig_open returning2(0) 
main: rig_open retcode=0
rigctl_parse: called, interactive=1
../../src/event.c(86): Starting rig poll routine thread
rig_set_cache_timeout_ms: called selection=0, ms=1000
rigctl_parse: cmd=m(6d) handle=0
rigctl_parse: vfo_opt=0
rigctl_parse: 0x635d470dfe2c rig not open...trying to reopen
rig.c(947):rig_open entered
rig_settings_get_path: path=/home/rex/.config/hamlib_settings
rig_open: async_data_enable=0, async_data_supported=0
rig_open(1121): 0x635d470e3e34 rs->comm_state==1?=1
ser_close: restoring options
rig.c(1125):rig_open returning2(-1) Invalid parameter

2:rigctl_parse.c(2240):rigctl_get_mode entered
rig_get_mode: rig or rig->caps is null
2:rigctl_parse.c(2246):rigctl_get_mode returning(-1) Invalid parameter

Invalid parameter

Rig command: m
Mode: LSB
Passband: 2400
`

The second command reliably works, as long as the data direction is the same (ie. requesting mode/frequency or setting either, but not changing)

I have started rigctl with the following:
`rigctl --model=16002 --set-conf=rig_pathname=/dev/ttyUSB0`

N0NB, 2025-04-20:
Hi Rex.

You should be able to get more debugging output by adding `-vvvvv` to the `rigctl` command line.

rexhunt, 2025-04-21:
Thank you for the quick reply.

New start command
`rigctl --model=16002 --set-conf=rig_pathname=/dev/ttyUSB0 -vvvvv`

Startup:

> rigctl.c(445) Startup: rigctl --model=16002 --set-conf=rig_pathname=/dev/ttyUSB0 -vvvvv
> rigctl Hamlib 4.6.2 2025-02-24T14:35:06Z SHA=91d110 64-bit
> Report bugs to <hamlib-developer@lists.sourceforge.net>
> 
> rig_test_2038: enter _TIME_BITS=64, __TIMESIZE=64 testing enabled for GLIBC 2.39
> rig_init: 2038 time test passed
> rig_check_rig_caps: p1=0x71aeae1a4940, p2=0x71aeae1aaec8, rig_model=0x71aeae1a4940, macro_name=0x71aeae1aaec8
> initrigs4_tentec: _init called
> rig_init: rig_model=Ten-Tec TT-538 Jupiter 20240913.0
> rig_init: rig has VFO_A
> rig_init: rig has VFO_B
> rig_token_lookup called for rig_pathname
> rig_confparam_lookup called for rig_pathname
> rig_set_conf called
> rig_confparam_lookup called for rig_pathname
> rig_set_conf: rig_pathname='/dev/ttyUSB0'
> rig.c(947):rig_open entered
> rig_settings_get_path: path=/home/rex/.config/hamlib_settings
> rig_open: async_data_enable=0, async_data_supported=0
> serial_open: /dev/ttyUSB0
> serial_setup: tcgetattr
> serial_setup: cfmakeraw
> serial_setup: cfsetispeed=57600,0x1001
> serial_setup: cfsetospeed=57600,0x1001
> serial_setup: data_bits=8
> serial_setup: stopbits=1
> serial_setup: parity=0
> serial_setup: Handshake=Hardware
> serial_setup: tcsetattr TCSANOW
> 0000    0d 0d 20 20 52 41 44 49 4f 20 53 54 41 52 54 0d     ..  RADIO START.
> **2:rig.c(8402):async_data_handler_start entered
> async_data_handler_start: async data support disabled since async_data_enabled=0
> **2:rig.c(8409):async_data_handler_start returning(0) 
> rig_open: 0x63d07655fe34 rs->comm_state==1?=1
> **2:rig.c(3452):rig_get_vfo entered
> rig_get_vfo: cache miss age=1000000ms
> **../../src/rig.c(3488) trace
> **2:rig_get_vfo: elapsed=0ms
> **2:rig.c(3522):rig_get_vfo returning(0) 
> **2:rig.c(2436):rig_get_freq entered
> rig_get_freq called vfo=VFOA
> rig_get_freq(2454) called vfo=VFOA
> vfo_fixup:(from rig_get_freq:2461) vfo=VFOA, vfo_curr=VFOA, split=0
> vfo_fixup: final vfo=VFOA
> ../../src/rig.c(2463) vfo=VFOA, curr_vfo=VFOA
> rig_get_freq: cache miss age=1000000ms, cached_vfo=VFOA, asked_vfo=VFOA, use_cached_freq=0
> rig_get_freq(2577): vfo_opt=0, model=16002
> write_block(): TX 3 bytes
> 0000    3f 41 0d                                            ?A.             
> read_string_generic(): RX 6 characters, direct=1
> 0000    41 00 3c ce 6c 0d                                   A.<.l.          
> rig_get_band called
> rig_get_freq: band changing to BAND80M
> **2:rig_get_freq: elapsed=2ms
> **2:rig.c(2714):rig_get_freq returning(0) 
> **2:rig.c(2436):rig_get_freq entered
> rig_get_freq called vfo=VFOB
> rig_get_freq(2454) called vfo=VFOB
> vfo_fixup:(from rig_get_freq:2461) vfo=VFOB, vfo_curr=VFOA, split=0
> vfo_fixup: final vfo=VFOB
> ../../src/rig.c(2463) vfo=VFOB, curr_vfo=VFOA
> rig_get_freq: cache miss age=1000000ms, cached_vfo=VFOB, asked_vfo=VFOB, use_cached_freq=0
> rig_get_freq(2577): vfo_opt=0, model=16002
[... 534 lines cut ...]
> 0000    57 0f 0d                                            W..             
> rig_get_mode: retcode after get_mode=0
> rig_get_mode(3094): debug
> ***3:cache.c(40):rig_set_cache_mode entered
> ***3:cache.c(156):rig_set_cache_mode returning(0) 
> **2:rig_get_mode: elapsed=4ms
> **2:rig.c(3111):rig_get_mode returning(0) 
> **2:network.c(1589):network_multicast_publisher_start entered
> ../../src/network.c(1598): multicast publisher address=0.0.0.0, port=4532
> ../../src/network.c(1616): not starting multicast publisher
> **2:network.c(1618):network_multicast_publisher_start returning(0) 
> **2:network.c(1800):network_multicast_receiver_start entered
> ../../src/network.c(1810): multicast receiver address=0.0.0.0, port=4532
> ../../src/network.c(1817): not starting multicast receiver
> **2:network.c(1819):network_multicast_receiver_start returning(0) 
> **2:event.c(272):rig_poll_routine_start entered
> **2:event.c(312):rig_poll_routine_start returning(0) 
> ../../src/event.c(85): Starting rig poll routine thread
> rig.c(290):add_opened_rig entered
> rig_set_cache_timeout_ms: called selection=0, ms=1000
> rig.c(302):add_opened_rig returning2(0) 
> rig.c(1644):rig_open returning2(0) 
> main: rig_open retcode=0
> rigctl_parse: called, interactive=1
>

db5fp, 2025-10-20:
I don't know if it's the same error or in the same direction. But using hamlib/rigctld with my Jupiter gives me errors after TX. Hamlib error: Protocol error while setting frequency.

I've tried rigctld -m 16002 -r /dev/ttyS1 -vvvvvv --set-conf=serial_handshake=Hardware w/O or w serial_handshake. 

I've tried to analyze /proc

1: uart:16550A port:000002F8 irq:3 tx:8172 rx:10911 fe:5 RTS|CTS|DTR|DSR|CD (okay state)

during/when error occurs frequency is changed from 28.074.00 to 28.074.55 (example on 10m)

[rig_ctld_log.txt](https://github.com/user-attachments/files/23002767/rig_ctld_log.txt)
[rig_ctld_log_hardware_handshake.txt](https://github.com/user-attachments/files/23002768/rig_ctld_log_hardware_handshake.txt)
[rig_ctld_log_hardware_handshake_2.txt](https://github.com/user-attachments/files/23002769/rig_ctld_log_hardware_handshake_2.txt)

73
Frank
DB5FP

db5fp, 2025-11-12:
Hi, any chance to get this bug to be solved? What else than logs is needed to proceed?