Hamlib/Hamlib #1701: RIGCAPS_NOT_CONST #define needs to be removed

tmiw, 2025-04-20:
I recently upgraded my Linux test environment to Fedora 42 and it looks like they changed some compiler stuff as well as upgraded Hamlib to 4.6.2. However, that seems to have caused the following build failure in FreeDV:

```
/home/mooneer/freedv-gui/src/rig_control/HamlibRigController.cpp: In static member function ‘static void HamlibRigController::InitializeHamlibLibrary()’:
/home/mooneer/freedv-gui/src/rig_control/HamlibRigController.cpp:180:26: error: invalid conversion from ‘int (*)(rig_caps*, void*)’ to ‘int (*)(const rig_caps*, void*)’ [-fpermissive]
  180 |         rig_list_foreach(&HamlibRigController::BuildRigList_, &RigList_);
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                          |
      |                          int (*)(rig_caps*, void*)
In file included from /home/mooneer/freedv-gui/src/rig_control/HamlibRigController.h:36,
                 from /home/mooneer/freedv-gui/src/rig_control/HamlibRigController.cpp:32:
/usr/include/hamlib/rig.h:3794:18: note: initializing argument 1 of ‘int rig_list_foreach(int (*)(const rig_caps*, void*), void*)’
 3794 | rig_list_foreach HAMLIB_PARAMS((int (*cfunc)(const struct rig_caps *, rig_ptr_t),
      |                  ^~~~~~~~~~~~~
```

Investigating further, it appears that `RIGCAPS_NOT_CONST` wasn't removed when `const` was readded to `rig_list_foreach` in 4.6.1. It'd be a good idea to do so for 4.6.3, but in the meantime I updated FreeDV to no longer look for that #define and always use the const version (see https://github.com/drowe67/freedv-gui/pull/861/commits/83e080f44aa5d682b619e928e21f5d33efc3990e).

N0NB, 2025-04-20:
Hi Mooneer.

Please submit a pull request with your changes if you can.

tmiw, 2025-04-21:
I went to fork the repo and found out that it was already commented out in rig.h:

```c
// As of 2025-01-03 removeing this -- fldigi was the only one that got it right
// riglist_foreach is now constant again but others are not
// #define RIGCAPS_NOT_CONST 1
```

Per @Tyrbiter in the freedv-gui PR mentioned above, it sounds like Fedora 42 was running an earlier version of 4.6 and *not* 4.6.2, hence the compiler issues I was having. I'll go ahead and close this, sorry for the confusion!

closed 2025-04-21 by tmiw