Hamlib/Hamlib #1704: N3FJP AC Log frequency error for bands 1GHz and higher

K7MDL2, 2025-04-20:
AC log rig uses the Dummy rig and aclog.c translates the AC Log commands to hamlib commands for this virtual rig interface.

When this hamlib rig interface receives frequencies with commas in them, 1,296.100Mhz for example, the parsing stops at the comma and thus only the GHz placed digit is extracted. There is a comma for all frequencies 1GHz and higher.  

The aclog_get_freq() function does not account for that comma and ends up with just the GHz digit * 1e6 converting MHz to Hz to get 1.000MHz.  Same goes for 2304 getting 2MHz and 5760 getting 5Mhz.  Stripping the comma and other whitespace before multiplying should resolve the problem.

In aclog_get_freq() function, the frequency is read from the http string.  
<FREQ>1,296.171100</FREQ>

I will need to set up a build environment for hamlib and likely wsjt-x to try it myself, will take some time.

Relevant code excerpt:  See the sscanf line.

char *p = strstr(value, "<FREQ>");
    *freq = 0;

    if (p) { sscanf(p, "<FREQ>%lf", freq); }

    *freq *= 1e6; // convert from MHz to Hz

    if (*freq == 0)
    {
        rig_debug(RIG_DEBUG_ERR, "%s: freq==0??\nvalue=%s\n", __func__,
                  value);
        RETURNFUNC(-RIG_EPROTO);
    }
    else
    {
        rig_debug(RIG_DEBUG_TRACE, "%s: freq=%.0f\n", __func__, *freq);
    }

    if (vfo == RIG_VFO_A)
    {
        priv->curr_freqA = *freq;
    }
    else // future support in ACLOG maybe?
    {
        priv->curr_freqB = *freq;
    }

-----Original Message-----
From: Mike Lewis via wsjt-devel <[wsjt-devel@lists.sourceforge.net](mailto:wsjt-devel@lists.sourceforge.net)> 
Sent: Thursday, April 17, 2025 10:23 AM
To: [wsjt-devel@lists.sourceforge.net](mailto:wsjt-devel@lists.sourceforge.net)
Cc: Mike Lewis <[k7mdl@hotmail.com](mailto:k7mdl@hotmail.com)>
Subject: [wsjt-devel] AC Log Hamlib error on frequencies 1GHz and higher

I am using N3FJP AC Log which has rig control.  I set up various versions (2.7.0 and 2.8.0_improved)  of WSJT-X to use the N3FJP AC Log rig definition.   Radios tested are IC-905, IC-705 (with a transverter aware band decoder) and Elecraft K3. The later 2 have transverters on 903 and 1296 and they report the transverter's frequency over CAT.

Focusing on just the K3 or IC-905, when any band below 1GHz is selected, all works as it should, with or without WSJT-X running.

Change bands by any means (radio or WSJTX) to any band higher than 903, WSJT-X reads the frequency as only the 1GHz digit times 1,000,000.   For example.  
1296 results in 1.000,000Hz.  
2304.100.000 results in 2.000MHz
5760.1 results in 5.000MHz

After WSJTZX now think the frequency is 1.000Mhz (for 1296) it thinks a band change occurredon the WSJT-X side and sends 1.0MHz to AC Log where it is updated and AC Log in turn changes the radio to 1.000Mhz.

I turned on WSJT_X Rig Control logging and got a narrowed log that I think shows the problem is in the WSJT-X hamlib code.   I tried WSJT-X V2.8.0_improved today, same issue with frequencies above >999Mhz.

In this attached log, I have the K3 set to 1296.171.100.  AC Log is controlling the rig.  All is good so far.

I then start up WSJT-X.  It queries AC Log for freq and gets 1296.   Has correct band (23) and mode (SSB).
string='<CMD><READBMFRESPONSE><BAND>23</BAND><MODE>SSB</MODE><MODETEST>PH</MODETEST><FREQ>1,296.171100</FREQ></CMD>

The next thing is it extracts the frequency from the message wrong
[2025-04-17 16:32:23.578614][00:00:02.941014][RIGCTRL:trace] aclog_get_freq: freq=1000000

Now WSJTX thinks it is 1.000MHz.  It polls AC Log a bit more, it still gets 1296 as proper.
Then it decided the frequency has changed and sends 1.000MHz to AC Log.  Now AC log says 1.000055MHz.
[... 177 lines cut ...]
[2025-04-17 16:32:23.585489][00:00:02.951063][RIGCTRL:debug] skip_freq: not skipping set_freq on vfo VFOA
[2025-04-17 16:32:23.585489][00:00:02.951066][RIGCTRL:trace] rig_set_freq: TARGETABLE_FREQ vfo=VFOA
[2025-04-17 16:32:23.585489][00:00:02.951069][RIGCTRL:trace] aclog_set_freq
[2025-04-17 16:32:23.585489][00:00:02.951073][RIGCTRL:trace] aclog_set_freq: vfo=VFOA freq=1000055
[2025-04-17 16:32:23.585489][00:00:02.951079][RIGCTRL:debug] ***3:aclog.c(256):aclog_transaction entered
[2025-04-17 16:32:23.585489][00:00:02.951084][RIGCTRL:debug] ****4:aclog.c(223):write_transaction entered
[2025-04-17 16:32:23.585489][00:00:02.951087][RIGCTRL:debug] network_flush called
[2025-04-17 16:32:23.585489][00:00:02.951142][RIGCTRL:debug] event.c(85): Starting rig poll routine thread
[2025-04-17 16:32:23.585489][00:00:02.951182][RIGCTRL:trace] write_block(): TX 95 bytes
[2025-04-17 16:32:23.585489][00:00:02.951193][RIGCTRL:trace] 0000    3c 43 4d 44 3e 3c 43 48 41 4e 47 45 46 52 45 51     <CMD><CHANGEFREQ
[2025-04-17 16:32:23.585489][00:00:02.951197][RIGCTRL:trace] 0010    3e 3c 56 41 4c 55 45 3e 31 2e 30 30 30 30 35 35     ><VALUE>1.000055
[2025-04-17 16:32:23.585489][00:00:02.951212][RIGCTRL:trace] rig_set_cache_timeout_ms: called selection=0, ms=1000
[2025-04-17 16:32:23.585489][00:00:02.951230][RIGCTRL:trace] 0020    3c 2f 56 41 4c 55 45 3e 3c 53 55 50 50 52 45 53     </VALUE><SUPPRES
[2025-04-17 16:32:23.585489][00:00:02.951238][RIGCTRL:trace] 0030    53 4d 4f 44 45 44 45 46 41 55 4c 54 3e 54 52 55     SMODEDEFAULT>TRU
[2025-04-17 16:32:23.585489][00:00:02.951242][RIGCTRL:trace] 0040    45 3c 2f 53 55 50 50 52 45 53 53 4d 4f 44 45 44     E</SUPPRESSMODED
[2025-04-17 16:32:23.585489][00:00:02.951246][RIGCTRL:trace] 0050    45 46 41 55 4c 54 3e 3c 2f 43 4d 44 3e 0d 0a        EFAULT></CMD>..
[2025-04-17 16:32:23.585489][00:00:02.951252][RIGCTRL:debug] ****4:aclog.c(247):write_transaction returning(0)
[2025-04-17 16:32:23.585489][00:00:02.951256][RIGCTRL:debug] ****4:aclog.c(141):read_transaction entered
[2025-04-17 16:32:23.585489][00:00:02.951652][RIGCTRL:trace] read_string_generic(): RX 56 characters, direct=1
[2025-04-17 16:32:23.585489][00:00:02.951668][RIGCTRL:trace] 0000    3c 43 4d 44 3e 3c 43 48 41 4e 47 45 46 52 45 51     <CMD><CHANGEFREQ
[2025-04-17 16:32:23.585489][00:00:02.951673][RIGCTRL:trace] 0010    52 45 53 50 4f 4e 53 45 3e 3c 56 41 4c 55 45 3e     RESPONSE><VALUE>
[2025-04-17 16:32:23.585489][00:00:02.951677][RIGCTRL:trace] 0020    31 2e 30 30 30 30 35 35 3c 2f 56 41 4c 55 45 3e     1.000055</VALUE>
[2025-04-17 16:32:23.585489][00:00:02.951681][RIGCTRL:trace] 0030    3c 2f 43 4d 44 3e 0d 0a                             </CMD>..
[2025-04-17 16:32:23.585489][00:00:02.951686][RIGCTRL:trace] read_transaction: string='<CMD><CHANGEFREQRESPONSE><VALUE>1.000055</VALUE></CMD>
'

N0NB, 2025-04-20:
Hi Mike.

Looking at the `sscanf` manual page I see this section:

```
       •      For  decimal conversions, an optional quote character (').  This
              specifies that the input number may include  thousands'  separa‐
              tors  as  defined  by the LC_NUMERIC category of the current lo‐
              cale.  (See setlocale(3).)  The quote character may  precede  or
              follow the '*' assignment‐suppression character.
```

if line 464 in `rigs/dummy/aclog.c` is changed to:

``` c
    if (p) { sscanf(p, "<FREQ>%'lf", freq); }
```

Does that resolve the bug?

N0NB, 2025-04-20:
I do wonder if making that change would be an issue in locales where the thousands separator is `.` rather than `,`.

K7MDL2, 2025-04-20:
A 2nd decimal makes things complicated.   I would strip comma and decimals except for the right most decimal.  The sender is assured to supply the value in MHz so that guarantees at least 1 decimal.  Any others to the left become expendable.

Another approach is to strip all commas and all decimals and skip the multiply by 1e6.  The value is 6 digits after the decimal so converts to Hz by itself.

N0NB, 2025-04-21:
> A 2nd decimal makes things complicated. I would strip comma and decimals except for the right most decimal. The sender is assured to supply the value in MHz so that guarantees at least 1 decimal. Any others to the left become expendable.

As N3FJP is only going to be run on Windows, does the `LC_NUMERIC` environment variable even exist?

> Another approach is to strip all commas and all decimals and skip the multiply by 1e6. The value is 6 digits after the decimal so converts to Hz by itself.

A regex would probably make this reasonably simple, but I'm averse to adding build dependencies￼, especially for what appears to be one line of code.  I will continue to research possible solutions.

K7MDL2, 2025-04-21:
I have asked N3FJP if the 1Hz resolution is guaranteed.   

Seems like sscanf in hamlib running under WSJT-X in my example, can be run on Windows, and Linux, others and thus could be affected regardless if AC Log is Windows only.

N0NB, 2025-04-21:
Only `RIG_MODEL_ACLOG` would be affected by this fix.

K7MDL2, 2025-04-21:
Correct, and only AC Log sends to it (legitimately).

N0NB, 2025-04-21:
I'd like to try the apostrophe fix and then if there is still an issue in the future, try a more aggressive fix.

K7MDL2, 2025-04-21:
Sounds like a good plan.  I am not yet setup to build hamlib but have multiple radios that can do microwave bands to test with via AC Log.

N0NB, 2025-04-21:
Do you test with the Windows binary version of the daily snapshots?  If so, I add it tomorrow and it would be in the Tuesday morning snapshot.

K7MDL2, 2025-04-21:
I am not familiar with the daily snapshots.  Also need to figure out how to load a particular version of hamlib into WSJTX, have not done that in years.

N0NB, 2025-04-21:
The daily snapshots are generated on a virtual machine here on my desktop around 4 to 4:30 AM Central time and posted to https://n0nb.users.sourceforge.net/  It generates the source archive and also builds the Windows 32 and 64 bit binaries.  Aside from having to download the entire `.ZIP` archive, the 32 and 64 bit `.DLL` files are available separately.  My understanding is that these can be "dropped in" and used for testing the master branch code.  Mike, W9MDB (SK) asked for this capability a couple years back.

K7MDL2, 2025-04-21:
I downloaded the daily .dll and dropped it into the wsjt-x bin folder after renaming the old dll.

Previous is 4.6.1 from 1/21
***@***.***

New is 4.7 from 4/20
***@***.***

I confirmed that 4/20 dll has the same problem so it will work well to test your 4/21 and later builds.

N0NB, 2025-04-21:
Okay.  I'll do a patch adding the apostrophe and it should be available
tomorrow morning for you to test.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

K7MDL2, 2025-04-23:
I tried the 4/22 DLL just now and same problem.  I suspect the apostrophe simply changes between comma vs decimal according to locale but does not strip it out.

N0NB, 2025-04-23:
Well, shucks.

I'm not sure why that is failing as the man page text infers to me that it should be treated as part of the decimal conversion.  It's possible that it checks for the presence of `LC_NUMERIC` in the environment and it's apparently not a simple value.  I think we can safely scrap that idea!

I think that locale can be ignored as I suspect AClog will always pass the string in the format you showed with the comma as the thousands separator so a simple parsing routine should suffice.

K7MDL2, 2025-04-23:
64bit.

N0NB, 2025-04-24:
Sorry for the delay, I missed your reply last night.

My test `.DLL` is at: https://n0nb.users.sourceforge.net/test/libhamlib-4.dll

K7MDL2, 2025-04-24:
Not quite.  Everything left of the decimal are now stripped, on the right 6 digits survive.

[2025-04-24 21:32:54.316187][00:00:41.432087][RIGCTRL:trace] read_transaction: string='<CMD><READBMFRESPONSE><BAND>6</BAND><MODE>FT8</MODE><MODETEST>DIG</MODETEST><FREQ>50.313000</FREQ></CMD>

50.313000 becomes 0.313000.

[2025-04-24 21:32:54.817366][00:00:41.933607][RIGCTRL:debug] *1:rig.c(2577):rig_get_freq returning(0)
[2025-04-24 21:32:54.817366][00:00:41.933620][RIGCTRL:trace] rig_get_freq frequency=313000
[2025-04-24 21:32:54.817366][00:00:41.933631][RIGCTRL:trace] frequency: 313000

N0NB, 2025-04-25:
* On 2025 24 Apr 16:38 -0500, K7MDL wrote:
> K7MDL2 left a comment (Hamlib/Hamlib#1704)
> 
> Not quite.  Everything left of the decimal are now stripped, on the right 6 digits survive.
> 
> [2025-04-24 21:32:54.316187][00:00:41.432087][RIGCTRL:trace] read_transaction: string='<CMD><READBMFRESPONSE><BAND>6</BAND><MODE>FT8</MODE><MODETEST>DIG</MODETEST><FREQ>50.313000</FREQ></CMD>
> 
> 50.313000 becomes 0.313000.
> 
> [2025-04-24 21:32:54.817366][00:00:41.933607][RIGCTRL:debug] *1:rig.c(2577):rig_get_freq returning(0)
> [2025-04-24 21:32:54.817366][00:00:41.933620][RIGCTRL:trace] rig_get_freq frequency=313000
> [2025-04-24 21:32:54.817366][00:00:41.933631][RIGCTRL:trace] frequency: 313000

Can you tell my C skills might be a bit rusty?  🤣

I'll have to take a look at my code again.  I'll let you know.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

N0NB, 2025-04-25:
Here is my actual diff:

``` diff
diff --git a/rigs/dummy/aclog.c b/rigs/dummy/aclog.c
index 8baeef075..39954debc 100644
--- a/rigs/dummy/aclog.c
+++ b/rigs/dummy/aclog.c
@@ -19,6 +19,7 @@
 *
 */
 
+#include <ctype.h>
 #include <stdio.h>
 #include <stdlib.h>
 #include <string.h>
@@ -421,9 +422,14 @@ static rmode_t modeMapGetHamlib(const char *modeACLog)
 /*
 * aclog_get_freq
 * Assumes rig!=NULL, STATE(rig)->priv!=NULL, freq!=NULL
+*
+* string='<CMD><READBMFRESPONSE><BAND>23</BAND><MODE>SSB</MODE><MODETEST>PH</MODETEST><FREQ>1,296.171100</FREQ></CMD> '
 */
 static int aclog_get_freq(RIG *rig, vfo_t vfo, freq_t *freq)
 {
+    int i, j = 0;
+    char f_string[32];
+
     char value[MAXARGLEN];
     struct aclog_priv_data *priv = (struct aclog_priv_data *) STATE(rig)->priv;
 
@@ -461,7 +467,30 @@ static int aclog_get_freq(RIG *rig, vfo_t vfo, freq_t *freq)
     char *p = strstr(value, "<FREQ>");
     *freq = 0;
 
-    if (p) { sscanf(p, "<FREQ>%'lf", freq); }
+    if (p)
+    {
+        // Count an extra character to move the pointer to the first digit.
+        p += strlen("<FREQ>n");
+
+        // Parse "1,296.171100" ignoring the comma.
+        for (i = 0; p[i] != '<'; i++)
+        {
+            if (isdigit(p[i]))
+            {
+                f_string[j++] = p[i];
+            }
+            else if (ispunct(p[i]) && p[i] == '.')
+            {
+                f_string[j++] = p[i];
+            }
+        }
+
+        f_string[j] = '\0';
+        rig_debug(RIG_DEBUG_TRACE, "%s: f_string=%s\n", __func__, f_string);
+
+        *freq = strtold(f_string, NULL);
+        rig_debug(RIG_DEBUG_TRACE, "%s: freq=%.0f\n", __func__, *freq);
+    }
 
     *freq *= 1e6; // convert from MHz to Hz
```

K7MDL2, 2025-04-25:
It is stripping off the leading digit 1 at a time until there are no more digits to the right.  
For example 1296.174 is set.  It becomes
296.174 then
96.174 then
6.174 then
0.174 finally

It got rid of comma bit also dropped the leading digit.  
I am thinking 
p += strlen("<FREQ>n");
should be 
p += strlen("<FREQ>");

_______________

[2025-04-25 22:21:11.007222][00:00:20.885470][RIGCTRL:trace] read_transaction: string='<CMD><READBMFRESPONSE><BAND>2</BAND><MODE>SSB</MODE><MODETEST>PH</MODETEST><FREQ>1,296.174000</FREQ></CMD>
'
[2025-04-25 22:21:11.007222][00:00:20.885478][RIGCTRL:trace] read_transaction: got </CMD>
[2025-04-25 22:21:11.007222][00:00:20.885488][RIGCTRL:debug] *****5:aclog.c(209):read_transaction returning(0)
[2025-04-25 22:21:11.007222][00:00:20.885502][RIGCTRL:debug] ****4:aclog_transaction: elapsed=147ms
[2025-04-25 22:21:11.007222][00:00:20.885512][RIGCTRL:debug] ****4:aclog.c(312):aclog_transaction returning(0)
[2025-04-25 22:21:11.007222][00:00:20.885520][RIGCTRL:trace] aclog_get_freq: f_string=296.174000
[2025-04-25 22:21:11.007222][00:00:20.885532][RIGCTRL:trace] aclog_get_freq: freq=296
[2025-04-25 22:21:11.007222][00:00:20.885542][RIGCTRL:trace] aclog_get_freq: freq=296174000
[2025-04-25 22:21:11.007222][00:00:20.885555][RIGCTRL:debug] ***3:aclog.c(517):aclog_get_freq returning(0)
[2025-04-25 22:21:11.007222][00:00:20.885588][RIGCTRL:debug] rig_get_band called
[2025-04-25 22:21:11.007222][00:00:20.885600][RIGCTRL:debug] **2:rig_get_freq: elapsed=147ms
[2025-04-25 22:21:11.007222][00:00:20.885607][RIGCTRL:debug] rig_lock: client lock disengaged
[2025-04-25 22:21:11.007222][00:00:20.885621][RIGCTRL:debug] **2:rig.c(2734):rig_get_freq returning(0)
[2025-04-25 22:21:11.007222][00:00:20.885634][RIGCTRL:trace] rig_set_freq: Asked freq=1296174000, got freq=296174000
[2025-04-25 22:21:11.007222][00:00:20.885653][RIGCTRL:debug] rig_set_freq: vfo=VFOA, save=VFOA
[2025-04-25 22:21:11.007222][00:00:20.885666][RIGCTRL:debug] *1:rig_set_freq: elapsed=149ms
[2025-04-25 22:21:11.007222][00:00:20.885676][RIGCTRL:debug] rig_lock: client lock disengaged
[2025-04-25 22:21:11.007222][00:00:20.885687][RIGCTRL:debug] *1:rig.c(2416):rig_set_freq returning(0)
[2025-04-25 22:21:11.007222][00:00:20.885713][RIGCTRL:trace] frequency: 1296174000
[2025-04-25 22:21:11.007222][00:00:20.885727][RIGCTRL:trace] force signal: true
[2025-04-25 22:21:11.007222][00:00:20.885746][RIGCTRL:trace] #: Transceiver::TransceiverState(online: yes Frequency {1296174000Hz, 0Hz} Mode: UNK; SPLIT: off; PTT: off; POWER: 0mWatts; SWR: 0; TUNE: false)

K7MDL2, 2025-04-25:
I tried to cross compile the DLL on a Rpi5 and was able to get a DLL built.  

It does not work yet, I get an error on startup about missing msys_usb_1.0.dll.   I have not been able to track down yet it comes from.

N0NB, 2025-04-26:
Are you using the `build-w64.sh` script?  If so, I have the following installed in `$HOME/builds` on the virtual machine:

``` console
$ tree -dL 1 builds
builds
├── libusb-1.0.20
├── libusb-1.0.22
├── libusb-1.0.24
├── libusb-win32-bin-1.2.4.0
└── nsis

5 directories
```

The Hamlib source archive gets upacked into `$HOME/builds` and the script uses the `LIBUSB_VER` variable to set which version to copy the `libusb-1.0.dll` file from.

It may be that at the time I developed this script--several years ago--MinGW didn't include the `msys_usb_1.0.dll` file.  My present VM--Debian 11 Bullseye--doesn't have it.  I've started building up a Debian 12 Bookworm amd64 VM and don't find `msys_usb_1.0.dll` on it either.

I get the `libusb` packaged archive through: https://libusb.info/

Which leads to: https://github.com/libusb/libusb/releases

Anyway, https://github.com/Hamlib/Hamlib/blob/master/scripts/build-w64.sh should do the trick for you.  I've not tried it on anything but `i686` or `amd64` (x86_64) architectures.

K7MDL2, 2025-04-26:
I used https://hamlib.sourceforge.net/manuals/4.3/Rdmewin.html as the guide and used the build-w64.sh.  I also used libusb-1.0.28.  Today I downloaded the build-w64.sh you linked above and also downloaded libusb-1.0.20 and .24 which is what your script has set by default.

There are some warnings/errors but goes on to build the hamlib dll anyway.

I see this cp error
    
    cp: cannot stat '/home/k7mdl/builds/libusb-1.0.24/MinGW64/dll/libusb-1.0.dll': No such file or directory

and uses one from someplace else.  

And this one:

    cp: cannot stat '/home/k7mdl/builds/libusb-1.0.24/MinGW64/dll/libusb-1.0.dll': No such file or directory

In both cases above, there is no MinGWxx anything.  There is a ~/builds/MinGW64/libusb-1.0.dll, likely from the MinGW install.   libusb-1.0.28 has the MinGW64 folder where libusb-1.0.24 does not.

With libusb-1.0.28 I get this error at the end

    x86_64-w64-mingw32-dlltool: Syntax error in def file /home/k7mdl/builds/hamlib-4.6.2/hamlib-w64-4.6.2/lib/msvc/libhamlib-4.def:562

I also see these, not sure they matter

    troff: ./doc/man1/rigctlsync.1:262: environment stack underflow
    troff: ./doc/man1/rigctlsync.1:262: cannot chop empty macro
 
    configure: WARNING: using cross tools not prefixed with host triplet
    configure: WARNING: libusb_init was not found in libusb-1.0--USB backends will be disabled

Using lib 20 and 24 I was able to build a working DLL using the hamlib 4.6.2 branch.  I was able to repro the initial error where 1,296 changes to 1.0Mhz.  

I then applied the code changes in aclog.c you listed in the diff and was able to repro the stripping of the elading digit until there were no more leading digits to strip.  SO I know I am building a good DLL.

Then I removed the 'n' in p += strlen("<FREQ>n"); and now it works proper.   So make this change and you are golden.

    p += strlen("<FREQ>");

I appreciate your help on fixing this!

Now that I can build hamlib and a DLL I can look into the missing PKTUSB mode issue and suggest changes for you rather than making you guess :-).   I have to set WSJT_X to MODE =None since there is no PKTUSB mode conversion made for hamlib.

K7MDL2, 2025-04-26:
FYI,  here is the working results:

[2025-04-26 23:36:41.108189][00:00:03.180322][RIGCTRL:trace] read_transaction: string='<CMD><READBMFRESPONSE><BAND>23</BAND><MODE>FT8</MODE><MODETEST>DIG</MODETEST><FREQ>1,296.174055</FREQ></CMD>
'
[2025-04-26 23:36:41.108189][00:00:03.180330][RIGCTRL:trace] read_transaction: got </CMD>
[2025-04-26 23:36:41.108189][00:00:03.180336][RIGCTRL:debug] ******6:aclog.c(209):read_transaction returning(0)
[2025-04-26 23:36:41.108189][00:00:03.180347][RIGCTRL:debug] *****5:aclog_transaction: elapsed=142ms
[2025-04-26 23:36:41.108189][00:00:03.180351][RIGCTRL:debug] *****5:aclog.c(312):aclog_transaction returning(0)
[2025-04-26 23:36:41.108189][00:00:03.180355][RIGCTRL:trace] aclog_get_freq: f_string=1296.174055
[2025-04-26 23:36:41.108189][00:00:03.180363][RIGCTRL:trace] aclog_get_freq: freq=1296
[2025-04-26 23:36:41.108189][00:00:03.180367][RIGCTRL:trace] aclog_get_freq: freq=1296174055
[2025-04-26 23:36:41.108189][00:00:03.180371][RIGCTRL:debug] ****4:aclog.c(515):aclog_get_freq returning(0)

N0NB, 2025-04-27:
An off by one error.  Imagine that.  🙄

I'll make a new DLL from my setup later that you can test just to ensure that it works.

N0NB, 2025-04-27:
Updated the DLL.  Give it a try, please, Mike.

K7MDL2, 2025-04-29:
I verified your test .DLL is now working correctly on 5.7G, 2.3G, 1.2G and lower bands.  Tnx!

N0NB, 2025-05-02:
I've pushed the changes to the master branch and tomorrow morning, at least by 1000z a new DLL built from master should be at:

https://n0nb.users.sourceforge.net/dll64/libhamlib-4.dll

K7MDL2, 2025-05-02:
Loaded it up and it is working.  Thanks Nate!

[image]

There are some other issues related to missing mode assignments.   Now that I can build a DLL I can work on solutions for that.  We can close this issue.

closed 2025-05-02 by K7MDL2

N0NB, 2025-05-03:
👍🏻