Hamlib/Hamlib #1705: Yaesu FTDX-101

aa5sh, 2025-04-20:
I think this happens on more Yaesu rigs than just the 101.  But if I change the Frequency then Mode it adjusts the Frequency.  For Example if I issue the following commands the frequency changes as indicated with ****
michaelmorgan@AA5SH-MPB ~ % rigctl -m 1040 -r /dev/cu.SLAB_USBtoUART5
Rig command: F 14300000
**** 14300000
Rig command: M USB 0
**** 14299300
Rig command: F 14030000
**** 14030000
Rig command: M CW 500
**** 14030700

K7MDL2, 2025-04-20:
Several radios offset their dial frequency when changing modes such as SSB to CW by the pitch frequency, Icom and Elecraft included.  

I suspect your pitch frequency is 700Hz.  I would also guess that in your example you started in CW before you changed to 14300000.  When you changed to USB the radio will shift down -700.   You reset the frequency back to 14300000, then changed to CW the radio would add +700Hz to keep the tone at the same spot to the remote listener.  This may be configurable in the radio.

aa5sh, 2025-04-20:
It does look like my cw pitch was set to 700hz (didn't make that connection
I typically run it at 580 not sure why it was changed).  I will see if I
can find any settings that may affect this thank you.

aa5sh, 2025-04-20:
Ok there is a setting in the radio that is affecting this.  On my 101 it is:
FUNC -> CW SETTING -> MODE CW -> CW FREQ DISPLAY = PITCH OFFSET (or) DIRECT FREQ

I changed it from PITCH OFFSET to DIRECT FREQ and the frequency seems to be working properly when clicking on spots.  I’m not sure what impact this has on the radio/operation if any.

closed 2025-04-20 by aa5sh