Hamlib/Hamlib #1712: LUA test issue is back in Fedora 42

markjfine, 2025-04-22:
Just a heads-up:
Recently updated to Fedora 42 and the perennial `AS_IF([test x"${cf_with_lua_binding}" = "xyes"]` failure in configure.ac is back. You can get past it by temporarily dnl'ing everything between line 680 and 693 if you don't need the lua binding, but I'm sure that's not the proper way.

N0NB, 2025-04-28:
Hi Mark.

The serial number of [the autoconf archive version of `ax_lua.m4`](https://git.savannah.gnu.org/gitweb/?p=autoconf-archive.git;a=blob_plain;f=m4/ax_lua.m4) is at 47 whereas the version in our repository is 39.  Can you update and do a new bootstrap?  If not I should be able to do that here and give you a source archive to test.

markjfine, 2025-04-28:
Renamed macros/ax_lua.m4 to force it to use serial 47 in /usr/share/aclocal/ax_lua.m4.  Bootstrap no longer gives the error, but running aclocal manually does:
```
configure.ac:680: error: AM_COND_IF: no such condition "LUAJIT"
/usr/share/aclocal-1.17/cond-if.m4:23: AM_COND_IF is expanded from...
/usr/share/aclocal/ax_lua.m4:199: AX_PROG_LUA is expanded from...
lib/m4sugar/m4sh.m4:697: AS_IF is expanded from...
configure.ac:680: the top level
autom4te: error: /usr/bin/m4 failed with exit status: 1
aclocal: error: autom4te failed with exit status: 1
```
Could this be an aclocal 1.17 vs. 1.16 thing?

N0NB, 2025-04-28:
I'm not at all that well versed to answer that question.  It may require asking upstream.

I see that Debian also packages the archive.  Of course, this was snapshotted in September 2022 for Debian Stable.  The serial for `ax_lua.m4` in that package is 42.  Would it be helpful to try an archive with it?

markjfine, 2025-04-29:
It couldn't hurt, but it won't solve the problem. I'm thinking it's either something to do with aclocal, or with lua / luajit itself.

N0NB, 2025-04-29:
You can try this:  https://n0nb.users.sourceforge.net/test/hamlib-4.7~git.tar.gz

markjfine, 2025-04-29:
Is this a branch within the repo? I'd rather keep the git source tree clean.

markjfine, 2025-04-29:
Yeah, so serial 42 doesn't solve the problem. Bootstrap continues to show the same LUAJIT error as aclocal above. It does work with serial 47 though.

So I looked at the differences between 42 and 47. 42 doesn't address anything to do with LUAJIT, whereas 47 does. Something in the new autoconf system (v1.17) and/or the lua (v5.4.7) and luajit (v2.1.1744318430) packages that are installed in Fedora 42 is causing it to look for a LUAJIT option that's not addressed in ax_lua.m4 serial 42.

Bottom line, serial 42 isn't the solution, but 47 (or something in-between) might be. It's almost better to not include the macro and let whatever the packages install to /usr/share/aclocal take precedence.

N0NB, 2025-04-29:
As I understand it, to generate source archive tarballs I need that macro on my build VM.  The same goes for anyone else boostrapping the build system.  I am much more inclined to update to serial 47 in our repository for 4.6.3 than rely on the various versions that would exist for others doing the bootstrap.

I'll update to 47 and push that into master.

This might be worth notifying the `ax_lua.m4` macro maintainers about.

markjfine, 2025-04-29:
Pretty sure that'll cure the symptom wrt hamlib. Thanks.

May want to search the issues for "LUAJIT". Think there were some other issues that looked similar to this. Unsure if they were closed or not, but this should close them.

N0NB, 2025-04-30:
I've a laptop with Debian Testing (forthcoming Debian 13, Trixie) and it has aclocal 1.17 and I don't see any error regarding Lua.  At the end of the default `configure` run, it shows `With Lua binding no`.

Regardless I plan to update the macro to 47.

markjfine, 2025-04-30:
It's certainly weird. Others have seen this months ago and this just hit me after the F42 update.

N0NB, 2025-04-30:
Just updating the `ax_lua.m4` file with no other changes results in this when bootstrapping:

``` console
$ ../hamlib/bootstrap 
Running 'autoreconf -i' to process configure.ac
and generate the configure script.
configure.ac:680: error: AM_COND_IF: no such condition "LUAJIT"
/usr/share/aclocal-1.16/cond-if.m4:23: AM_COND_IF is expanded from...
macros/ax_lua.m4:199: AX_PROG_LUA is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
configure.ac:680: the top level
autom4te: error: /usr/bin/m4 failed with exit status: 1
aclocal: error: /usr/bin/autom4te failed with exit status: 1
autoreconf: error: aclocal failed with exit status: 1
```

I see it on both Debian 12 and 13.  Even though the comments in the new `ax_lua.m4` state that setting `LUAJIT` is *optional*, the lack of it in `configure.ac` is a hard failure here.  Adding the following to `configure.ac` appears to resolve the error:

``` m4
dnl Newer ax_lua.m4 macro, at least Serial 47, requires LUAJIT be set.
AC_ARG_WITH([luajit],
    [AS_HELP_STRING([--with-luajit],
        [prefer LuaJIT over PUC Lua, even if the latter is newer. @<:@default=no@:>@])])
AM_CONDITIONAL([LUAJIT], [test "x$with_luajit" != 'xno'])
```

Although it does result in the additional help text:

``` console
  --with-luajit           prefer LuaJIT over PUC Lua, even if the latter is
                          newer. [default=no]
```

Which I don't know a thing about, but I doubt is harmful.  If setting it to `yes` causes an issue I'll ask for a fix then!

N0NB, 2025-04-30:
Just for fun, setting `--with-luajit` on this machine results in:

``` console
checking for inline... inline
../hamlib/configure: line 7411: AX_COMPILER_VENDOR: command not found
../hamlib/configure: line 7419: syntax error near unexpected token `AX_PREPEND_FLAG'
../hamlib/configure: line 7419: `    AX_REQUIRE_DEFINED(AX_PREPEND_FLAG)'
```

It may be better to omit the option and just have the conditional set to satisfy the macro.

N0NB, 2025-04-30:
I've reduced the change to `configure.ac` to:

``` m4
dnl Newer ax_lua.m4 macro, at least Serial 47, requires LUAJIT be present.
AM_CONDITIONAL([LUAJIT], [false])
```

This appears to satisfy `ax_lua.m4` without introducing some possible confusing help text and a `configure` option we don't need.

markjfine, 2025-04-30:
It's possible what you ran into was something clashing in the `configure` and `automake` caches, though I thought `bootstrap` was to initialize all that. Just in case, I was clearing them before running `bootstrap`, but the missing line in `configure.ac` is likely why I was still seeing the LUAJIT error when running `aclocal`. This was clearly the other piece of the puzzle.

If you'd like me to test the updated `configure.ac` on Fedora let me know.

markjfine, 2025-05-01:
Cleaned my tree down to match the bare minimum (no configuration, makefiles, libtool, etc.), but included the updated macro and added the two lines above to configure.ac at line 679.

Ran `./bootstrap; ./configure; make -j15; sudo make install; sudo ldconfig -v` ... all with no issues.
Ran `aclocal`... no issues.

Think we're good!

N0NB, 2025-05-01:
One thing to be careful of is that when I had the macro archive package installed that `aclocal` updated all of the macro files that it found.  The clue was that I was getting some strange CFLAGS warnings when bootstrapping that I'd not seen before.  Running `git status` showed all of the files that had been updated which included many more than the two expected!

I have a shell alias called `hl-clean` that calls `git clean -dxf` that removes all of the generated files from the working tree but doesn't touch any tracked files that have been changed.  This is good because stashing the changes isn't necessary to start from scratch and bootstrap.  This also left the updated macro files that `aclocal` had updated.  A simple `git restore <file list>` brought things back into their expected order.

I'm having to relearn a few tricks.

N0NB, 2025-05-02:
I've pushed the changes to master and they will appear in tomorrow's daily snapshot at:

https://n0nb.users.sourceforge.net/

I guess there really isn't anything to test from the tarball as this is a bootstrap issue.

markjfine, 2025-05-02:
Builds with no issues on F42 as well as macOS 15.4.1. 👍🏼

closed 2025-05-02 by markjfine

N0NB, 2025-05-02:
I keep both sites in sync for redundancy.

I'm not even sure if SF offers personal Web space any more.  Way back when I needed a place to host daily snapshots, as I didn't want to pollute the download area with them, I used the personal Web space.  It has worked quite well.

These days most of the work is done through GitHub, for better or for worse...

N0NB, 2025-05-02:
It's a shell of its former self, but it hosts our mailing list, so I continue to keep the project active over there.

markjfine, 2025-05-02:
I have one personal project over there that's so old that it's insignificant... Smart Gnome Control is from 2004 and uses Gnome v1! lol

markjfine, 2025-05-02:
One of these days (he says, tongue firmly in cheek), I should try to update it... hopefully before Gnome goes to v5.

N0NB, 2025-05-02:
2004, I was probably still using IceWM.  I was warming up to KDE via KNOPPIX but left GNOME alone for the most part.  Now I use GNOME and just can't go back to the simplistic workflow of Windows/KDE/etc.

I don't think GNOME will ever be at "4" or "5", rather it's just evolving as it goes.  From what I see, version 50 will  continue to be based on GTK4/libadwaita.  There really isn't much talk of a GTK5 from what I've seen of late.