Hamlib/Hamlib #1713: Green Heron RT-21 Issue

aa5sh, 2025-04-24:
I am getting 'Invalid Parameter' errors periodically with my RT-21.  I am connecting over the network connection.  I'm not sure what may be causing this.  Here is a debug output, notice some times it returns the right az 48.8 but a couple times it looks like it is getting something else.  Not sure if this is something with the interface or can be accounted for some how:
michaelmorgan@AA5SH ~ % rotctld -m 405 -r 192.168.2.133:6555 -vvvvv
rotctld, Hamlib 4.6.2 2025-02-09T21:03:50Z SHA=870364 64-bit
Report bugs to <hamlib-developer@lists.sourceforge.net>

rot_init called
initrots4_rotorez called
rot_register (401)
rot_register (402)
rot_register (403)
rot_register (404)
rot_register (405)
rot_register (406)
rotorez_rot_init called
rot_open called
rot_open: using network address 192.168.2.133:6555:TCP
network_open: TCP connect
network_open: hoststr=192.168.2.133, portstr=6555
network_open: client port=11468
Opened rot model 405, 'RT-21'
Backend version: 20230507.0, Status: Stable
Connection opened from ::ffff:127.0.0.1:52285
rotctl(d): ? '' '' '' ''
rotctl(d): p '' '' '' ''
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes
0000    42 49 31 3b                                         BI1;            
read_string_generic(): RX 5 characters, direct=1
0000    34 38 2e 38 3b                                      48.8;           
rt21_rot_get_position: "48.8;" after conversion = 48.8
rt21_rot_get_position: azimuth = 48.8 deg; elevation = 0.0 deg
rot_get_position: got az=48.80, el=0.00
rotctl(d): p '' '' '' ''
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes
0000    42 49 31 3b                                         BI1;            
read_string_generic(): RX 5 characters, direct=1
0000    34 38 2e 38 3b                                      48.8;           
rt21_rot_get_position: "48.8;" after conversion = 48.8
rt21_rot_get_position: azimuth = 48.8 deg; elevation = 0.0 deg
rot_get_position: got az=48.80, el=0.00
rotctl(d): p '' '' '' ''
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes
0000    42 49 31 3b                                         BI1;            
read_string_generic(): RX 5 characters, direct=1
0000    34 38 2e 38 3b                                      48.8;           
rt21_rot_get_position: "48.8;" after conversion = 48.8
rt21_rot_get_position: azimuth = 48.8 deg; elevation = 0.0 deg
rot_get_position: got az=48.80, el=0.00
rotctl(d): p '' '' '' ''
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes
0000    42 49 31 3b                                         BI1;            
read_string_generic(): RX 5 characters, direct=1
0000    34 38 2e 38 3b                                      48.8;           
rt21_rot_get_position: "48.8;" after conversion = 48.8
rt21_rot_get_position: azimuth = 48.8 deg; elevation = 0.0 deg
rot_get_position: got az=48.80, el=0.00
rotctl(d): p '' '' '' ''
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes
0000    42 49 31 3b                                         BI1;            
read_string_generic(): RX 5 characters, direct=1
0000    34 38 2e 38 3b                                      48.8;           
rt21_rot_get_position: "48.8;" after conversion = 48.8
rt21_rot_get_position: azimuth = 48.8 deg; elevation = 0.0 deg
rot_get_position: got az=48.80, el=0.00
rotctl(d): p '' '' '' ''
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes
0000    42 49 31 3b                                         BI1;            
read_string_generic(): RX 5 characters, direct=1
0000    34 38 2e 38 3b                                      48.8;           
rt21_rot_get_position: "48.8;" after conversion = 48.8
rt21_rot_get_position: azimuth = 48.8 deg; elevation = 0.0 deg
rot_get_position: got az=48.80, el=0.00
rotctl(d): p '' '' '' ''
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes
0000    42 49 31 3b                                         BI1;            
read_string_generic(): RX 5 characters, direct=1
0000    34 38 2e 38 3b                                      48.8;           
rt21_rot_get_position: "48.8;" after conversion = 48.8
rt21_rot_get_position: azimuth = 48.8 deg; elevation = 0.0 deg
rot_get_position: got az=48.80, el=0.00
rotctl(d): p '' '' '' ''
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes
0000    42 49 31 3b                                         BI1;            
read_string_generic(): RX 6 characters, direct=1
0000    01 30 00 3e 34 38                                   .0.>48          
rotctl_parse: NETROTCTL_RET -1
rotctl(d): p '' '' '' ''
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes
0000    42 49 31 3b                                         BI1;            
read_string_generic(): RX 3 characters, direct=1
0000    2e 38 3b                                            .8;             
rotctl_parse: NETROTCTL_RET -1
rotctl(d): p '' '' '' ''
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes
0000    42 49 31 3b                                         BI1;            
read_string_generic(): RX 5 characters, direct=1
0000    34 38 2e 38 3b                                      48.8;           
rt21_rot_get_position: "48.8;" after conversion = 48.8
rt21_rot_get_position: azimuth = 48.8 deg; elevation = 0.0 deg
rot_get_position: got az=48.80, el=0.00
rotctl(d): p '' '' '' ''
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes
0000    42 49 31 3b                                         BI1;            
read_string_generic(): RX 5 characters, direct=1
0000    34 38 2e 38 3b                                      48.8;           
rt21_rot_get_position: "48.8;" after conversion = 48.8
rt21_rot_get_position: azimuth = 48.8 deg; elevation = 0.0 deg
rot_get_position: got az=48.80, el=0.00
rotctl(d): p '' '' '' ''
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes
0000    42 49 31 3b                                         BI1;            
read_string_generic(): RX 5 characters, direct=1
0000    34 38 2e 38 3b                                      48.8;           
rt21_rot_get_position: "48.8;" after conversion = 48.8
rt21_rot_get_position: azimuth = 48.8 deg; elevation = 0.0 deg
rot_get_position: got az=48.80, el=0.00
rotctl(d): p '' '' '' ''
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes
0000    42 49 31 3b                                         BI1;            
read_string_generic(): RX 5 characters, direct=1
0000    34 38 2e 38 3b                                      48.8;           
rt21_rot_get_position: "48.8;" after conversion = 48.8
rt21_rot_get_position: azimuth = 48.8 deg; elevation = 0.0 deg
rot_get_position: got az=48.80, el=0.00
rotctl(d): p '' '' '' ''
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes
0000    42 49 31 3b                                         BI1;            
read_string_generic(): RX 5 characters, direct=1
0000    34 38 2e 38 3b                                      48.8;           
rt21_rot_get_position: "48.8;" after conversion = 48.8
rt21_rot_get_position: azimuth = 48.8 deg; elevation = 0.0 deg
rot_get_position: got az=48.80, el=0.00
Connection closed from ::ffff:127.0.0.1:52285

`
`

closed 2025-04-24 by aa5sh

N0NB, 2025-04-25:
Hi Michael.

I'm curious as to the resolution of this issue.  You can email me privately if you'd like, n0nb@n0nb.us.

aa5sh, 2025-04-25:
I emailed Green Heron about this and Jeff with GH said it looked like two devices were trying to communicate with my rotor at the same time over two different methods (Ethernet and Serial/USB).  Hamlib is sending BI1; and it looked like something else sent a R21; and the responses are different so HamLib didn't get what it expected.  I'm not sure exactly what else was talking with it but I do have a handful of computers so not sure if something was running on another one.  He had a beta firmware on the website which isolates the channels so the communication responses don't get confused.  I loaded the beta firmware and so far I am getting much better results.  I loaded version 4.14g1 of the firmware.