Hamlib/Hamlib #1732: Why does Hamlib work one week and then never again?

tomb18, 2025-05-12:
Author of Win4K4, Win4Icom and Win4Yaesu here:

For years I have always recommended against hamlib. Always seems to be broken.  We spend a good amount of time on the phone a while ago and got everything working only to find that two weeks later it was broken.
Same thing just happened.  I made a release of Win4K4Suite for the Elecraft K4 and all was working with WSJT.  Dozens of people using it.
Now all of a sudden with no changes HamLib does not work. First of all transmit is erratic. Using the Test PTT in WSJT results in the transmit staring but then it never turns off.  If I look  at the CAT stream, all of a sudden there are numerous ID statements coming from HamLib.  They time out.

The proper ID for the Elecraft K4 is ID0;, not ID017.  Using ID017 works but id0; does not. 

Like always Omni-0Rig works perfectly fine.

hamlibdk, 2025-05-18:
**Hamlib** is a dynamic tool. It is worked on by Amateurs - for Amateurs. It is a community product.

Because one is an "Amateur" that does not mean that the software is not fit for purpose or is it not World-Class. There is nothing like this product - and its scope - anywhere on the planet.

This is an EXTREMELY LARGE and WIDE RANGING TOOL. **Sometimes even the smallest change can unexpectedly affect something else**.

Yes bugs are frustrating. 

**The key is patience rather than throwing stones**. 

Constructively and accurately report issues. You will be surprised at the zeal with which developers and maintainers of this product work on constructive reports.

N0NB, 2025-05-23:
Hi Tom.

As we don't have releases with a granularity of weekly, except for daily source snapshots which are for testing and development purposes, not to be considered "releases", I'm puzzled by your report that Hamlib breaks from one week to the next.  Some things that will help us are "trace" logs for the users reporting problems.  These should include the version and date of Hamlib.

As for the second paragraph, is your software linking to Hamlib as a C library or are you starting `rigctld` and using its network protocol?  If `rigctld` is started it will in all likelihood remain running even if your software shuts down unless your software takes the step of terminating the `rigctld` process.  It should be safe to check for a running `rigctld` when your software starts and connect to it, assuming it's controlling the radio you want.

Looking through the struct list in `rigs/kenwood/kenwood.c` I don't see an ID of 0:

``` C
  98 static const struct kenwood_id_string kenwood_id_string_list[] =
  99 {
 100     { RIG_MODEL_TS940,  "001" },
 101     { RIG_MODEL_TS811,  "002" },
 102     { RIG_MODEL_TS711,  "003" },
 103     { RIG_MODEL_TS440,  "004" },
 104     { RIG_MODEL_R5000,  "005" },
 105     { RIG_MODEL_TS140S, "006" },
 106     { RIG_MODEL_TS790,  "007" },
 107     { RIG_MODEL_TS950S, "008" },
 108     { RIG_MODEL_TS850,  "009" },
 109     { RIG_MODEL_TS450S, "010" },
 110     { RIG_MODEL_TS690S, "011" },
 111     { RIG_MODEL_TS950SDX, "012" },
 112     { RIG_MODEL_TS50,   "013" },
 113     { RIG_MODEL_TS870S, "015" },
 114     { RIG_MODEL_TS570D, "017" },  /* Elecraft K2|K3|KX3 also returns 17 */
 115     { RIG_MODEL_TS570S, "018" },
 116     { RIG_MODEL_TS2000, "019" },
 117     { RIG_MODEL_TS480,  "020" },
 118     { RIG_MODEL_PT8000A, "020" }, // TS480 ID but behaves differently
 119     { RIG_MODEL_SDRUNO, "020" }, // TS480 ID but behaves differently
 120     { RIG_MODEL_TS590S, "021" },
 121     { RIG_MODEL_TS990S, "022" },
 122     { RIG_MODEL_TS590SG,  "023" },
 123     { RIG_MODEL_TS890S,  "024" },
 124     { RIG_MODEL_THD7A,  "TH-D7" },
 125     { RIG_MODEL_THD7AG, "TH-D7G" },
 126     { RIG_MODEL_TMD700, "TM-D700" },
 127     { RIG_MODEL_TMD710, "TM-D710" },
 128     { RIG_MODEL_THD72A, "TH-D72" },
 129     { RIG_MODEL_THD74, "TH-D74" },
 130     { RIG_MODEL_TMV7, "TM-V7" },
 131     { RIG_MODEL_TMV71,  "TM-V71" },
 132     { RIG_MODEL_THF6A,  "TH-F6" },
 133     { RIG_MODEL_THF7E,  "TH-F7" },
 134     { RIG_MODEL_THG71,  "TH-G71" },
 135     { RIG_MODEL_MALACHITE,  "020" },
 136     { RIG_MODEL_NONE, NULL }, /* end marker */
 137 };
```

I'll admit to not examining the K4 Programmer's manual, but I recall asking Wayne, N6KR, way back when about the K4 and it stated then that it would be the same as the K3.  Perhaps it has been expanded since then as I see a bit of reference to the K4 capabilities over and above the K3.

With the passing of Mike, W9MDB, the project is regaining its footing.  Several of us are striving to squash a lot of bugs before we make a 4.6.3 release.  As I noted earlier, detailed reports are much more useful than more general ones and we would appreciate more detail.  Please.

N0NB, 2025-05-23:
I hunted down the K4 Prgrammers Reference:  https://ftp.elecraft.com/K4/Manuals%20Downloads/K4%20Programmer's%20Reference,%20rev.%20D4.pdf

Two ID modes are defined:

> ID (Radio Identification)
SET/RESP format in K41 mode: IDs; where string s is "0" (default) or the user's ID text
from the status display (typically a call sign). If the string is changed via an ID command, it
will appear in the radio's status/ID area after a reboot of the radio.
RESP format in K40 mode: ID017; (GET only; for K3 compatibility).

However, as of Hamlib 4.6.0, code was added to specifically set the K4 to K40 when requesting the ID.  The fact that in K41 mode the radio can return `0` or an arbitrary string may have been deemed problematic and so the decision was to only verify that a radio of likely a K4 is at the other end of the line, seems to have been the choice.  I suspect only real reason to query the ID string in K41 mode would be to get or set the user ID string that is shown at power up (assuming it is similar to the K3 in that regard).  If so, it may be possible to do this through another Hamlib set of calls instead of the ID.

tomb18, 2025-05-23:
"However, as of Hamlib 4.6.0, code was added to specifically set the K4 to K40 when requesting the ID. ":

One should not arbitrarily set the K4 meta command. What should be done is to first query the radio to see what mode it is in.  Then get the ID.

Just a little bit of background here. I have tried to support HamLib in my products for 8+ years. As I mentioned previously it seemingly breaks very often. About a year ago I worked with Mike on the phone going through issues with HamLib and my products and we did resolve them.  However, once again, after a short period of time, they broke.
I think one of the issues is that the software that uses HamLib may use different versions of HamLib.  For example, in the past WSJT gave no indication what version of HamLib it was using.  Therefore there could have been a mismatch of a new version of WSJT and an old version of Hamlib in some clients.  The latest releases of WSJT has added a button to upgrade HamLib which is a good idea.  But other software may or may not have this.  Not only that, this leaves it the user choice to upgrade or not.
I have over 10,000 users, so when these things happen I tell them to use Omni-rig.  That has never failed. I cannot keep recommending HamLib when there is a stable problem free alternative.
Anyways, no matter what I say, people will use HamLib anyways. So hopefully, issues can be resolved.  To me, working with Mike, HamLib seemed awfully complex which makes things prone to bugs.
Tom

N0NB, 2025-06-15:
Tom,

Reading your link I see no mention of Hamlib and it's all about Icom emulation, which is fine, I just don't see the connection with the issue at hand.  If WSJT-X can use Omnirig then tell your users to use it and let it go at that.