Hamlib/Hamlib #1738: make distcheck fails when configured --with-perl-binding=yes or --with-indi=yes

dforsi, 2025-05-17:
After `./configure --with-binding=yes` the command `make distcheck` fails with:
```
ERROR: files left after uninstall:
./lib/x86_64-linux-gnu/perl/5.40.1/Hamlib.pm
./lib/x86_64-linux-gnu/perl/5.40.1/auto/Hamlib/Hamlib.so
./lib/x86_64-linux-gnu/perl/5.40.1/auto/Hamlib/.packlist
./lib/x86_64-linux-gnu/perl/5.40.1/perllocal.pod
make[1]: *** [Makefile:914: distuninstallcheck] Error 1
```

After `./configure --with-indi=yes` the command `make distcheck` fails with:
```
make[2]: *** No rule to make target 'distclean'.  Stop.
```

N0NB, 2025-05-23:
Did the recent merge correct this to your satisfaction?

dforsi, 2025-05-24:
No, the recent merge fixed the yes/no options passed to `./configure`.

This issue is caused by the facts
1) that those 4 Perl files aren't deleted; we can `rm` them explicitly because they are just 4 and are unlikely to change and we will see if more should be deleted, or we can look into a different way, but I'm not a Perl expert
This is the target in the current `binding/Makefile.am`
```
## 'distclean' target is a NOOP in Hamlib-pl.mk
distclean-perl: ## Hamlib-pl.mk
##	$(MAKE) $(AM_MAKEFLAGS) -f Hamlib-pl.mk distclean
```

2) (I think) `rotators/indi` is included twice so `make distclean` is executed twice in that directory and the second time the Makefile has already been deleted.
Basically, for `rotators/indi` must be added once to `SUBDIRS` if it gets compiled or once to `DIST_SUBDIRS` if it doesn't.

To fix this issue and  #1730 I think that changes are needed in `configure.ac` and some `Makefile.am` and possibly `config.h.in` to define some symbols that can be used to conditionally include stuff.

Help needed to find the less disruptive way or this must be postponed.