Hamlib/Hamlib #1740: Issue with latest pull and building WSJT-X 2.7.0 [ clock_gettime64 could not be located in the DLL ]

hamlibdk, 2025-05-18:
Hi Folks,

An issue has been encountered with Hamlib pulled from master as of UTC 2025-05-18 @ 08:00:00 .

The development environment is a fully clean and deployed JTSDK 4.0.1u1 (see **https://sourceforge.net/projects/hamlib-sdk/** ) and has worked perfectly up until the time that Mike W9MDB became SK.

[ Fully updates means that the deployed MSYS2 environment is FULLY updated with pacman -Syuu, LibUSB is the supplied 1.0.28 version that is known to work, and a known-to-work Boost 1.88.0 environment has also been deployed with the kit. The Qt environ is QT 5.15.2 ]

- Hamlib pulls the latest code from GIT, updates and compiles cleanly.
- WSJT-X pulls the latest code from GIT and compiles cleanly.

But on execution the following message is displayed:

_The procedure entry point clock_gettime64 could not be located in the dynamic link library c:\wsjt\wsjtx\bin\libhamlib4-dll_

The environment here is:

- A fully updated JTSDK 4.0.0.1u1 (including pacman updates of the embedded MSYS2 environment)
- LibUSB 1.0.28
- Boost 1.88.0
- Windows 11 25H4 [ Fully updates ]

This issue has been replicated on a number of machines.

What further information do you require?

73

Hamlib SDK Developers

................................................................................................................

$ build-hamlib

---------------------------------------------------------------
       JTSDK64 Tools MSYS2 Hamlib Build Script v4.0.0.1
---------------------------------------------------------------

* This script compiles Hamlib Libraries for the JTSDK build system.

---------------------------------------------------------------
 COMPILE INFORMATION [ Hamlib ]
---------------------------------------------------------------

 Script Option(s) ...: None (Default = Dynamic with LibUSB)

 Date ...............: 18-05-2025
 Package ............: Hamlib
 User ...............: stepheni
 CPU/Job Count ......: 1
 Compiler ...........: /mingw64/bin/gcc
 Platform ...........: MINGW64
 Source Dir .........: /home/stepheni/
 Build Dir ..........: /home/stepheni/src/hamlib/build
 LibUSB Include .....: /c/JTSDK64-Tools/tools/libusb/1.0.28/include
 LibUSB DLL .........: /c/JTSDK64-Tools/tools/libusb/1.0.28/bin
 Package Config......: /c/JTSDK64-Tools/tools/hamlib/qt/5.15.2/lib/pkgconfig/hamlib.pc

 Install Prefix .....: /c/JTSDK64-Tools/tools/hamlib/qt/5.15.2

---------------------------------------------------------------
 CHECKING TOOL-CHAIN [ QT 5.15.2 ]
---------------------------------------------------------------

 ar .................: OK
 nm .................: OK
 ld .................: OK
 gcc ................: OK
 g++ ................: OK
 ranlib .............: OK
[... 375 lines cut ...]
    With Readline support           yes
    With INDI support               no

    Enable HTML rig feature matrix  no
    Enable WinRadio                 yes
    Enable Parallel                 yes
    Enable USRP                     no
    Enable USB backends             yes
    Enable shared libs              yes
    Enable static libs              no

-----------------------------------------------------------------------

---------------------------------------------------------------
 RUNNING MAKE CLEAN [ Hamlib ]
---------------------------------------------------------------

* /c/JTSDK64-Tools/config/hlclean flag not set

---------------------------------------------------------------
 INSTALLING [ Hamlib ]
---------------------------------------------------------------

* Clearing out old /c/JTSDK64-Tools/tools/hamlib/qt/5.15.2: Complete
,,,, (etc)

hamlibdk, 2025-05-18:
Ps: 

If one drops in the DLL's from https://n0nb.users.sourceforge.net/ (i.e. https://n0nb.users.sourceforge.net/hamlib-w64-4.7~git-20250518-ad824fa85.zip ) then all works ! 

The issue is building from the LATEST GIT Pull with the VERY LATEST MSYS2 packages. All should just work !

N0NB, 2025-05-23:
As I replied on the mailing list earlier this week:

> I'm not familiar with the SDK or building WXJT-X, etc.  I can state that                                                            
"clock_gettime64" is not found in the Hamlib repository as 'git grep                                                                
clock_gettime64' returns nothing.                                                                                                   
                                                                                                                                    
> I presume the issue exists outside of Hamlib.  Sorry.                                                                               

And:

> * On 2025 19 May 10:17 -0500, George Baltz wrote:                                                                                   
>> I noticed 'Build_OS is mingw32' yet something is looking for a 64 bit                                                             
>> routine. Is that OK?                                                                                                              
                                                                                                                                    
> When I do a DDG for clock_gettime64 the first hit is:                                                                               
                                                                                                                                    
> https://www.gnu.org/software/libc/manual/html_node/64_002dbit-time-symbol-handling.html                                             
                                                                                                                                    
> I wonder if this is some sort of macro that MSYS2 is looking for?  Or,                                                              
that MinGW supplies from the C library.                                                                                             
                                                                                                                                    
>Using objdump on an older 64 bit DLL I get:                                                                                         
                                                                                                                                    
> $ objdump -p libhamlib-4.dll | grep clock_gettime                                                                                   
        bae924     13  clock_gettime                                                                                                
                                                                                                                                    
> And on today's 32 bit DLL:                                                                                                          
                                                                                                                                    
> $ objdump -p libhamlib-4.dll | grep clock_gettime                                                                                   
        a6823c     13  clock_gettime                                                                                                
                                                                                                                                    
> Searching for clock_gettime64 in either file returned zero results each.                                                            
                                                                                                                                    
> Undoubtedly this has to do with Y2038 mitigation.                                                                                   

I am at a loss of what this might be.  The build scripts used by the VM that creates the daily snapshot and release DLL is nothing special in regards to `clock_gettime`.  For reference, my scripts run the `configure` script as follows:

``` shell
# Configure and build hamlib for x86_64-w64-mingw32, with libusb-1.0

./configure --host=${HOST_ARCH} \
 --prefix=${INST_DIR} \
 --without-cxx-binding \
 --disable-static \
 CPPFLAGS="-I${LIBUSB_1_0_BIN_PATH}/include" \
 LDFLAGS="-L${LIBUSB_1_0_BIN_PATH}/MinGW64/dll"
```

What happens after that to correctly present `clock_gettime` is above my pay grade!