Hamlib/Hamlib #1761: IC-705: "unknown mode=WFM" when using rigctl to switch to WFM

tmiw, 2025-06-08:
When using `rigctl` to control my IC-705, I noticed that it doesn't recognize WFM mode (despite the radio actually supporting it; ref page 16 from [the CI-V manual for the IC-705](https://www.icomeurope.com/wp-content/uploads/2020/08/IC-705_ENG_CI-V_1_20200721.pdf)).

Version info:

```
MacBookPro:build_osx mooneer$ ../hamlib/bin/rigctl -m 3085 -r /dev/cu.usbmodem142201 -vvvv
rigctl.c(532) Startup: ../hamlib/bin/rigctl -m 3085 -r /dev/cu.usbmodem142201 -vvvv
rigctl Hamlib 4.6.2 2025-02-09T21:03:50Z SHA=870364 64-bit
Report bugs to <hamlib-developer@lists.sourceforge.net>
```

Debugging output when attempting to switch modes:

```
Rig command: m
rigctl_parse.c(2338):rigctl_get_mode entered
**2:rig.c(2979):rig_get_mode entered
icom.c(2898):icom_get_mode entered
icom_get_mode called vfo=currVFO
icom.c(2733):icom_get_mode_without_data entered
icom_get_mode_without_data called vfo=currVFO
icom_get_vfo_number_x25x26(10016): current_vfo=currVFO, vfo=currVFO -> vfo_number=0
***3:frame.c(525):icom_transaction entered
icom_transaction: cmd=0x26, subcmd=0x00, payload_len=0
****4:frame.c(145):icom_one_transaction entered
****4:frame.c(503):icom_one_transaction returning(0) 
***3:frame.c(556):icom_transaction returning(0) 
icom_get_dsp_flt called, mode=USB
***3:frame.c(525):icom_transaction entered
icom_transaction: cmd=0x1a, subcmd=0x03, payload_len=0
****4:frame.c(145):icom_one_transaction entered
****4:frame.c(503):icom_one_transaction returning(0) 
***3:frame.c(556):icom_transaction returning(0) 
from_bcd called
icom.c(2116):icom_get_dsp_flt returning2(3000) 
icom.c(2881):icom_get_mode_without_data returning2(0) 
icom_get_mode mode=4
icom_get_mode databuf[2]=0, mode=4
icom.c(3038):icom_get_mode returning2(0) 
***3:cache.c(40):rig_set_cache_mode entered
***3:cache.c(156):rig_set_cache_mode returning(0) 
**2:rig_get_mode: elapsed=3ms
**2:rig.c(3111):rig_get_mode returning(0) 
Mode: USB
Passband: 3000
rigctl_parse.c(2361):rigctl_get_mode returning2(0) 

Rig command: M WFM 20000
rigctl_parse.c(2310):rigctl_set_mode entered
rig_parse_mode called
**2:rig.c(2778):rig_set_mode entered
rig_set_mode called, vfo=currVFO, mode=WFM, width=20000, curr_vfo=currVFO
***3:icom.c(2496):icom_set_mode entered
****4:rig.c(2979):rig_get_mode entered
icom.c(2898):icom_get_mode entered
icom_get_mode called vfo=currVFO
icom.c(2733):icom_get_mode_without_data entered
icom_get_mode_without_data called vfo=currVFO
icom_get_vfo_number_x25x26(10016): current_vfo=currVFO, vfo=currVFO -> vfo_number=0
*****5:frame.c(525):icom_transaction entered
icom_transaction: cmd=0x26, subcmd=0x00, payload_len=0
******6:frame.c(145):icom_one_transaction entered
******6:frame.c(503):icom_one_transaction returning(0) 
*****5:frame.c(556):icom_transaction returning(0) 
icom_get_dsp_flt called, mode=USB
*****5:frame.c(525):icom_transaction entered
icom_transaction: cmd=0x1a, subcmd=0x03, payload_len=0
******6:frame.c(145):icom_one_transaction entered
******6:frame.c(503):icom_one_transaction returning(0) 
*****5:frame.c(556):icom_transaction returning(0) 
[... 143 lines cut ...]
****4:frame.c(145):icom_one_transaction entered
****4:frame.c(503):icom_one_transaction returning(0) 
***3:frame.c(556):icom_transaction returning(0) 
icom_get_dsp_flt called, mode=LSB
***3:frame.c(525):icom_transaction entered
icom_transaction: cmd=0x1a, subcmd=0x03, payload_len=0
****4:frame.c(145):icom_one_transaction entered
****4:frame.c(503):icom_one_transaction returning(0) 
***3:frame.c(556):icom_transaction returning(0) 
from_bcd called
icom.c(2116):icom_get_dsp_flt returning2(3000) 
icom.c(2881):icom_get_mode_without_data returning2(0) 
icom_get_mode mode=8
icom_get_mode databuf[2]=0, mode=8
icom.c(3038):icom_get_mode returning2(0) 
***3:cache.c(40):rig_set_cache_mode entered
***3:cache.c(156):rig_set_cache_mode returning(0) 
**2:rig_get_mode: elapsed=3ms
**2:rig.c(3111):rig_get_mode returning(0) 
Mode: LSB
Passband: 3000
rigctl_parse.c(2361):rigctl_get_mode returning2(0) 
```

Let me know if you need me to do anything else to help debug this.

GeoBaltz, 2025-06-09:
This seems kind of strange to me, but Page 4-5 of the manual says the WFM IF filter width is fixed at 200kHz, not 20kHz.  try 'M WFM 200000' and see if that works.

73
n3gb

GeoBaltz, 2025-06-11:
Quick question - what frequency was the rig set to?  Although I can't find any specific restrictions, WFM sensitivity ratings are only given for 76-108MHz.

tmiw, 2025-06-12:
> Quick question - what frequency was the rig set to? Although I can't find any specific restrictions, WFM sensitivity ratings are only given for 76-108MHz.

Not sure about the person who originally reported this to me but in my testing I was using 94.9 MHz.

GeoBaltz, 2025-06-12:
I finally got simic705.c into good enough shape to get this sequence to run, and AFAICT hamlib is sending the right commands to do the change mode.  But since I made simic705 do what I *think* is right, it has no bearing on the Real World :-(

Can you run this test again with another 'v' added to the CLI so we can see the actual bytes going over the wire?  After that, try setting WFM from the IC-705 front panel, and then do 'm' in rigctl to see if it can read and interpret the rig status.

tmiw, 2025-06-13:
See below:

[hamlib-wfm.txt](https://github.com/user-attachments/files/20718883/hamlib-wfm.txt)

I first set the frequency to 94.9 Mhz (originally on 20 meters), then try to set the mode to WFM (which fails), then I set WFM on the radio's display and request current mode (which properly returns WFM).

GeoBaltz, 2025-06-13:
Aha!  Yes, 4.6.2 was sending an invalid mode change to the IC-705, which rightfully rejected it.  I had been wasting my time (and yours) looking for a bug in the current state of hamlib, when this had already been fixed.

Should be OK in 4.6.3, thanks to commit 12cc40f4f77c  

Sorry for not finding this sooner.

tmiw, 2025-06-14:
Yep, confirmed fixed on 4.6.3. Thanks for looking into this!

closed 2025-06-14 by tmiw