Hamlib/Hamlib #1767: kenwood_transaction() is erroneously stripping eol characters, breaking hamlib for tm-d710/tm-v71a devices.

larsks, 2025-06-17:
I was seeing the symptoms reported in #1698. I spent some time debugging the code, and the problem is that by the time we reach this block of code:

https://github.com/Hamlib/Hamlib/blob/aca0b2d1c63eb27b23ce376eaf630da169c17a0a/rigs/kenwood/kenwood.c#L483-L495

We have already called `remove_nonprint(buffer)` at line 448:

https://github.com/Hamlib/Hamlib/blob/aca0b2d1c63eb27b23ce376eaf630da169c17a0a/rigs/kenwood/kenwood.c#L446-L448

This strips the command termination character (`\r`), causing the "Check that command termination is correct" to always fail, effectively breaking hamlib for all TM-D710/TM-V71 type devices.

---

We can confirm this behavior by inspecting the value of `buffer` before and after the call to `remove_nonprint`:

```
Breakpoint 3, kenwood_transaction (rig=rig@entry=0x435d60, cmdstr=cmdstr@entry=0x7ffff740176e "ID", data=data@entry=0x7fffffffc300 "",
    datasize=datasize@entry=128) at kenwood.c:443
443         rig_debug(RIG_DEBUG_TRACE, "%s: read_string len=%d '%s'\n", __func__,
(gdb) p buffer
$1 = "ID TM-V71\r", '\000' <repeats 117 times>
```

This is the `rig_debug` call just before the call to `remove_nonprint`; we can see that the buffer has the `\r` command termination character. After calling `remove_nonprint`...

```
(gdb) until 450
kenwood_transaction (rig=rig@entry=0x435d60, cmdstr=cmdstr@entry=0x7ffff740176e "ID", data=data@entry=0x7fffffffc300 "", datasize=datasize@entry=128)
    at kenwood.c:450
450         if (retval < 0)
(gdb) p buffer
$2 = "ID TM-V71", '\000' <repeats 118 times>
```

...we see that the termination character has been dropped.

N0NB, 2025-06-17:
Hi Lars.

I think that commit d1e0e3f should be fixed rather than reverted entirely otherwise #1652 is active, although it may not be fixed in its entirety owing to the stripping of the command terminator.  It seems that detecting and ending processing of the string when `\r` is encountered should be correct.  Do you agree?

N0NB, 2025-06-17:
Perhaps we can get help from @brightkill who opened #1652.

I agree, the transceiver should be fixed, but that may or may not be possible and even if it is, the likelihood is high that not all users will apply the fix so Hamlib is going to have to deal with the broken emulation.

Reading through #1652 I see that the terminator is the semicolon (`0x3b`).  Mike chose the standard `isprint()` function which will pass `;` but not `\r`.  The semicolon is the classic Kenwood command terminator.

Kenwood evidently opted to use `\r` in these radios.  I wonder if the reason why is because these radios pass `;` as data and not the terminator?  I wonder if radios that use `;` as the terminator ever pass `\r` as data?

Can you try this as a solution:

``` C
static int remove_nonprint(char *s)
{
    int i, j = 0; 
    if (s == NULL) return 0;

    for (i = 0; s[i] != '\0'; ++i) 
    {    
        if (isprint((unsigned char)s[i]) || (unsigned char)s[i] == '\r' )
        {    
            s[j++] = s[i]; // Copy printable character
        }    
    }    
    s[j] = '\0'; // Null-terminate the string

    return j; // Return the new length of the string
}
```

N0NB, 2025-06-17:
Adding the diff:

``` diff
diff --git a/rigs/kenwood/kenwood.c b/rigs/kenwood/kenwood.c
index 1596c2dfe..2b27073c1 100644
--- a/rigs/kenwood/kenwood.c
+++ b/rigs/kenwood/kenwood.c
@@ -238,7 +238,7 @@ static int remove_nonprint(char *s)
 
     for (i = 0; s[i] != '\0'; ++i)
     {
-        if (isprint((unsigned char)s[i]))
+        if (isprint((unsigned char)s[i]) || (unsigned char)s[i] == '\r')
         {
             s[j++] = s[i]; // Copy printable character
         }
```

N0NB, 2025-06-17:
If the `cmdtrm` appears at some random place in the read date string, then a protocol error should be raised.

GeoBaltz, 2025-06-17:
OK, I'll wait for his response.

There are just too many different scenarios to try to parse the message at this low a level; I think it's just all or nothing.  If there is a stray nonprintable in a message ending in '\r', we can let the upper levels handle it.

commit 85c9e15e (2025-06-18): Un-break hamlib on TM-D710/TM-V71/etc

Commit d1e0e3f introduced a `remove_nonprint` method that breaks hamlib on
all TM-D710/TM-V71A devices by erroneously removing the command termination
character (`\r`).

This commit adopts a solution proposed by @GeoBaltz that only runs
`remove_nonprint` if the command termination character itself is printable.

Resolves: #1767 #1698
PR #1768 (2025-06-18): Un-break hamlib on TM-D710/TM-V71/etc
This reverts commit d1e0e3f204d3b0f5b3a17e83a50d360184136e0d.

That commit introduced a `remove_nonprint` method that breaks hamlib on all
TM-D710/TM-V71A devices by erroneously removing the command termination
character.

Resolves: #1767 #1698
closed 2025-06-18 by N0NB