Hamlib/Hamlib #1804: Icom 910 not working under Hamlib 4.7~2025-07-102:21:24Z with WSJT-x

kyle470, 2025-07-14:
Hello. I am using WSJT-x and while the program works fine with my Icom 706mk2g, it does not seem to directly support my Icom 910. I can, however, get WSJT-x to control my IC-910 by using Ham Radio Deluxe and setting WSJT-x to connect to Ham Radio Deluxe. I am using the default ci-v address 60.

All of the on-screen errors I see in WSJT-X refer to the Icom 9700. See attached. Maybe helpful?

[image]

Thanks 73.

Kyle
K0KN

kyle470, 2025-07-18:
Hello.  I attached a screen shot to the issue,  here it is again.  The errors all mention the IC-9700 I noticed. 

[image]

N0NB, 2025-07-18:
It also mentions ic7300.c which is completely wrong.  I don't have any Icom gear to test but just running `rigctl` gives me this output:

``` console
$ rigctl -m 3044 -vvvvv
../../hamlib/tests/rigctl.c(472) Startup: rigctl -m 3044 -vvvvv
rigctl Hamlib 4.7~git 2025-06-26T17:38:23Z SHA=ef0b7e 64-bit
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_test_2038: enter _TIME_BITS=64, __TIMESIZE=64 testing enabled for GLIBC 2.36
rig_init: 2038 time test passed
rig_check_rig_caps: p1=0x7ff11bef03c0, p2=0x7ff11bef6948, rig_model=0x7ff11bef03c0, macro_name=0x7ff11bef6948
initrigs4_icom: _init called
rig_init: rig_model=Icom IC-910 20250517.2
Requesting 47928 bytes for rig_struct
Requesting 728 bytes for rig_cache
rig_init: rig has VFO_A
rig_init: rig has VFO_B
rig_init: rig has VFO_SUB_A
rig_init: rig has VFO_SUB_B
rig_init: rig has VFO_MAIN_A
rig_init: rig has VFO_MAIN_B
rig_init: rig has VFO_SUB
rig_init: rig has VFO_MAIN
rig_init: rig has VFO_MEM
../../hamlib/src/rig.c(947) trace
*1:icom.c(706):icom_init entered
icom_set_x25x26_ability: Hamlib thinks rig does not support x25/x26 command
*1:icom.c(776):icom_init returning(0) 
rig.c(1003):rig_open entered
rig_settings_get_path: path=/home/nate/.config/hamlib_settings
rig_open: async_data_enable=0, async_data_supported=0
serial_open: /dev/ttyS0
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=19200,0x000e
serial_setup: cfsetospeed=19200,0x000e
serial_setup: data_bits=8
serial_setup: stopbits=1
serial_setup: parity=0
serial_setup: Handshake=None
serial_setup: tcsetattr TCSANOW
rig_open: 0x5595aaf011a4 rs->comm_state==1?=1
*1:icom.c(1157):icom_rig_open entered
icom_rig_open: IC-910 v20250517.2
**2:icom.c(824):icom_get_usb_echo_off entered
***3:frame.c(524):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
****4:frame.c(144):icom_one_transaction entered
write_block(): TX 6 bytes
0000    fe fe 60 e0 03 fd                                   ..`...          
****4:frame.c(329):icom_one_transaction returning(-5) Communication timed out

icom_transaction: retry=0: Communication timed out

icom_transaction: failed: Communication timed out
Communication timed out

***3:frame.c(555):icom_transaction returning(-5) Communication timed out

**2:icom.c(832):icom_get_usb_echo_off returning(-5) Communication timed out

icom_rig_open: echo status result=-5
icom_rig_open: Unable to determine Icom echo status -- is rig on and connected?
*1:icom.c(1192):icom_rig_open returning(-5) Communication timed out

ser_close: restoring options
rig.c(1512):rig_open returning2(-5) Communication timed out
```

This strikes me as WSJT-X not calling the correct model number.

kyle470, 2025-07-18:
Thanks I opened a ticket with WSJT-x. I figured they'd refer me here first, oh well. Thanks.  I'll report back if they respond.  73