Hamlib/Hamlib #1813: SDR#/gpredict

markjfine, 2025-07-24:
Do we know if the current SDR#/gpredict backend is working with SDR# 1921?

Was just testing a mingw64 build of Smart Gnome Control on Win11 and was unable to connect to SDR# using the their gpredict plugin sitting on 127.0.0.1:4532. Attempting to connect gives this error:
[image]

However, if I keep that connection set and switch to one of the two Dummy backends the gpredict module in SDR# shows as being connected (although none of the commands do anything). So it appears as if the port isn't firewalled, just a possible issue with the SDR#/gpredict backend itself.

N0NB, 2025-07-30:
Is there a way to communicate without the gpredict plugin?

markjfine, 2025-07-30:
Don't think so, since the plugin is what was sitting on the port to read command traffic for SDR#.

That said, the plugin hasn't been updated since 2018, and I know that Youssef's completely redone SDR# in the past year or so. It might be easier for me to just rebuild the plugin using an updated SDK (if it's available).