Hamlib/Hamlib #1829: I/O error cant aktivate Hamlib on wsjt-x 2.8.0 improved,

sm5mek, 2025-08-03:
When i try to open wsjt-x improved i got error to contact rig, no connection to ham lib, audio works but no reading of frekvencies.         
 #Here an screenshot of problem. Link to [sm5mek](https://www.sm5mek.se/2025/08/03/error-hamlib-wsjt-x/)

[image]

[image]

[image]

N0NB, 2025-08-03:
Also, about ten days ago I switched to building the snapshot `.DLL`s on a 64 bit Debian Trixie (soon to be Debian 13 in a week) virtual machine.  This requires updating some other `.DLL`s, so the WSJT-X team will probably catch up to this in a short time.

An update to the VM was going to happen anyway as the prior VM was going to lose security support once Debian 13 is released.

N0NB, 2025-08-04:
As this [thread shows](https://sourceforge.net/p/hamlib/mailman/message/59215755/) a renamed directory caused some issues for building WSJT-X.  This has apparently been resolved.  Work will be done in WSJT-X to work with the replacement `rig_get_conf2()` function.