Hamlib/Hamlib #1868: indi_wrapper.cpp warnings from clang

N0NB, 2025-08-24:
Compiling with clang 14 and 19 (Debian 12 and 13) the following warning is produced (same in both cases):

``` console
  CXX      indi_wrapper.lo
../../../hamlib/rotators/indi/indi_wrapper.cpp:482:18: warning: implicit conversion from 'int' to 'azimuth_t' (aka 'float') changes value from 2147483647 to 2147483648 [-Wimplicit-const-int-float-conversion]
        mDstAz = INT_MAX;
               ~ ^~~~~~~
/usr/lib/llvm-14/lib/clang/14.0.6/include/limits.h:46:19: note: expanded from macro 'INT_MAX'
#define INT_MAX   __INT_MAX__
                  ^~~~~~~~~~~
<built-in>:76:21: note: expanded from here
#define __INT_MAX__ 2147483647
                    ^~~~~~~~~~
../../../hamlib/rotators/indi/indi_wrapper.cpp:483:18: warning: implicit conversion from 'int' to 'elevation_t' (aka 'float') changes value from 2147483647 to 2147483648 [-Wimplicit-const-int-float-conversion]
        mDstEl = INT_MAX;
               ~ ^~~~~~~
/usr/lib/llvm-14/lib/clang/14.0.6/include/limits.h:46:19: note: expanded from macro 'INT_MAX'
#define INT_MAX   __INT_MAX__
                  ^~~~~~~~~~~
<built-in>:76:21: note: expanded from here
#define __INT_MAX__ 2147483647
                    ^~~~~~~~~~
2 warnings generated.
```

Meanwhile GCC is silent.

N0NB, 2025-08-25:
@nonoo Norgert, do you have thoughts on this?

nonoo, 2025-08-25:
Try using __INT_MAX__ instead INT_MAX indi_wrapper.cpp, or use gcc instead clang.

N0NB, 2025-08-25:
I don't understand C++ well enough to figure out how `mDstAz` and `mDstEl` being set to `INT_MAX` are referenced elsewhere in the code.  I don't see the function referenced directly in the functions (methods?) where these variables are used, so I presume inheritance is involved?  Why was `INT_MAX` chosen as the apparent initial value?  Couldn't `0` be used instead?  Please help me understand this.

Regarding Clang, as Hamlib is supported and compiled on MacOS which uses Clang, I do test with it to catch things.  While the Continuous Integration is also run on MacOS, `libnova` is not available in "brew" per @dforsi so INDI is not built on it and this warning is never shown.

Also, do you know if there is any way to support INDI on MS Windows?  It seems that `libindi` and `libnova` are not available in MinGW and so support for the INDI driver is not compiled for the W binaries.

dforsi, 2025-08-30:
@nonoo since it's just initialization, can we set those to zero?
```
        mDstAz = 0.0;
        mDstEl = 0.0;
```

This driver should work with the INDI simulators?
I tried with `indiserver indi_simulator_telescope indi_simulator_rotator` together and one at a time, but I always get  "telescope not connected"

On Debian the version is
INDI Library: 1.9.9
Code 1.9.9-tgz. Protocol 1.7.

In one terminal I run
`indiserver indi_simulator_telescope indi_simulator_rotator`
in another terminal I run:
`tests/rotctl -m 2001--vvvv P 10 20`

and I get:
```
rotctl Hamlib 4.7~git 2025-08-30T19:36:45Z SHA=415bcf 64-bit
Report bugs to <hamlib-developer@lists.sourceforge.net>

rot_init called
initrots4_indi called
rot_register (2001)
rot_open called
ser_set_dtr: DTR=0
ser_set_rts: RTS=0
INDI::BaseClient::connectServer: creating new connection...
indi: server connected
Opened rot model 2001, 'INDI'
Backend version: 0.2, Status: Stable
rot_set_position called az=10.00 el=20.00
rot_set_position: south_zero=0 
indi: telescope not connected
rot_close called
rot_cleanup called
```

nonoo, 2025-08-31:
I don't remember if it should work with the simulator and why I used INT_MAX, probably because that is a truly invalid value which indicates that the destination coordinates are invalid.

N0NB, 2025-09-07:
I'll test that idea.

closed 2025-09-09 by N0NB