Hamlib/Hamlib #1870: Split GitHub workflow script into multiple scripts

N0NB, 2025-08-26:
Currently `.github/workflows/c-cpp.yml` run Continuous Integration (CI) on Ubuntu "latest" and MacOS "latest".  While this works and most configuration settings are shared, there are exceptions, particularly for MacOS which makes its stanzas a bit less straightforward than they could be.

I floated the idea to @dforsi and @markjfine in PR #1862 and both thought there is some merit to the idea.  [Daniel commented as follows](https://github.com/Hamlib/Hamlib/pull/1862#issuecomment-3222094525):

> I think that's a good idea because we can see if a failure is only in one OS. It will also make it easier to add another workflow for Windows or Raspberry or something else. In the current workflow I like the sharing of the `--with` and `--enable` options, while overriding values for some steps, I don't know how to share them between different scripts. 

> I see that workflows can be executed manually with the trigger `on workflow_dispatch`, this can be useful to test the split scripts before having them run automatically, there isn't much documentation because it seems to work similar to `on push`/`on pull_request`: 

> https://github.blog/changelog/2020-07-06-github-actions-manual-triggers-with-workflow_dispatch/
> https://docs.github.com/en/actions/reference/workflows-and-actions/workflow-syntax?s=09#onworkflow_dispatch

With this approach I presume that each build will be self-contained with some apparent duplication between scripts.  This really isn't an issue as these files just sit in the repository and are not a part of the source distribution.

N0NB, 2025-09-13:
I have pushed a pair each of Ubuntu and MacOS workflow files.  At the moment each pair consists of a build with features and no features with the configure options from `c-cpp.yml`.

To cut down on used resources and maybe speed up execution time they only do `make dist`.

To improve things each pair should be merged with both features and no-features for each OS.  I ran into a shortage time to figure out how to give the resulting `tar` files unique names.

It seems GitHub only runs a new workflow file if it exists in the master branch (once there it can be changed in a feature branch and it will be run, at least it did for me).  Constantly pushing changes that didn't work was getting tedious so I found a program called [act](https://github.com/nektos/act) that allows for mimicking the GitHub workflow system locally using a Docker image.  Unfortunately, only a rather old version of Ubuntu is available, so MacOS can't be tested locally.

Anyway, they are there now and maybe @dforsi can tweak them up.

dforsi, 2026-08-30:
Now the workflows are run for any branch not only for `master` or `feature` (PR #2179 and #2180)