Hamlib/Hamlib #1881: rigerror2() should be documented - and perhaps given a better name

peterbmarks, 2025-09-02:
We were looking for a less intimidating error message to show users than the error plus stack trace that rigerror() returns and was informed of rigerror2().

This is exactly what we want but is not documented in the rig API https://hamlib.sourceforge.net/manuals/4.3/group__rig.html

It might be good to give these better names, perhaps rigerror_stack_trace() and rigerror_user_message().

Peter
VK3TPM

N0NB, 2025-09-06:
I've uploaded an updated 4.7 manual generated by Doxygen 1.14 with the changes in PR #1889.

dforsi, 2025-09-07:
Since the same error numbers and strings are used also for amplifiers and rotators, if the name is changed, I suggest using `hamlib_` as prefix instead of `rig`.

N0NB, 2025-09-07:
That's likely a sensible idea, Daniele.  Perhaps a link to the old name should remain at least for one major release, something like:

``` c
typedef hamlib_error_full rigerror;
typedef hamlib_error_simple rigerror2;
```