Hamlib/Hamlib #1895: build system is trying to include hamlibdatetime.h but it doesn't exist

techn0mad, 2025-09-07:
**Describe the bug**
During the build, make fails when trying to generate `hamlibdatetime.h`

**To Reproduce**
Steps to reproduce the behavior (please use `rigctl` with the `-vvvvv` option if at all possible and post the output under number 4 below):
1. `./bootstrap`
2. `./configure . . .`
3. `make`
4. Make output:
```
Making all in amplifiers/expert
  CC       expert.lo
  CCLD     libhamlib-expert.la
Making all in src
make[1]: don't know how to make ./hamlibdatetime.h. Stop

make[1]: stopped in /home/larry/Hamlib/src
*** Error code 1

Stop.
make: stopped in /home/larry/Hamlib
```

**Expected behavior**
The generation and subsequent inclusion of `hamlibdatetime.h` is successful and make completes without error.

**Desktop (please complete the following information):**
 - OS: FreeBSD 14.3-RELEASE-p2
 - Client application: custom (Python bindings)
 - Client application version: n/a

**Additional context**
Checking to see what's going on in the Makefile rule, which uses `diff -qN` to determine the status of the `hamlibdatetime.h` file:
```
$ diff -qN ./hamlibdatetime.h.tmp ./hamlibdatetime.h
Files ./hamlibdatetime.h.tmp and ./hamlibdatetime.h differ
$ echo "Exit code: $?"
Exit code: 1
$ rm -f hamlibdatetime.h hamlibdatetime.h.tmp
$ diff -qN ./hamlibdatetime.h.tmp ./hamlibdatetime.h
diff: ./hamlibdatetime.h.tmp: No such file or directory
diff: ./hamlibdatetime.h: No such file or directory
$ echo "Exit code: $?"
Exit code: 2
```
Apparently the Makefile does not differentiate between return codes (i.e. files differ vs. files do not exist)

techn0mad, 2025-09-07:
I did not explicitly set `CFLAGS`. I think the issue here is that the Makefile rule just checks the result from `diff -qN` to see if it is non-zero, whereas the actual return code from diff has different meanings (i.e. 1 means the files are different, 2 means one or more of the files does not exist).

The FreeBSD man page for diff states:

> EXIT STATUS
>      The diff utility exits with one of the following values:
> 
>            0       No differences were found.
>            1       Differences were found.
>            >1      An error occurred.

It seems that the Makefile rule should be checking for all possible cases rather than only the "no differences" case.

dforsi, 2025-09-07:
I was wrong about the double dollar sign: it is needed to pass `$?` tho the shell.
The rule is testing for "not zero" and it executes the part after the `||`:
> test $$? -eq 0 ||

techn0mad, 2025-09-08:
Nate, although there's no "--version" option to display it explicitly, I'm pretty sure FreeBSD uses BSD make rather than Gnu make, as do MacOS, OpenBSD, and NetBSD.

N0NB, 2025-09-25:
I've asked about this on the GNU Automake mailing list.  One suggestion was to remove the `$(builddir)/` string from the `BUILT_SOURCES` variable in `src/Makefile.am` and doing so allows `make` to proceed to build the target which now fails with:

```
Making all in src
Files ./hamlibdatetime.h.tmp and ./hamlibdatetime.h differ
*** Error code 1

Stop.
make[1]: stopped in /home/nate/git/hlb/src
*** Error code 1

Stop.
make: stopped in /home/nate/git/hlb
```

Next is to proceed to resolve that problem.

closed 2025-09-27 by N0NB