Hamlib/Hamlib #1897: On FreeBSD, the configure script finds the libedit library but fails to detect the headers that come with it

techn0mad, 2025-09-07:
**Describe the bug**
Both editline and readline headers exist but configure didn't detect them, when it should default to using (editline).

**To Reproduce**
Steps to reproduce the behavior (please use `rigctl` with the `-vvvvv` option if at all possible and post the output under number 4 below):
1. `./bootstrap`
2. `./configure . . .`
3. `make`
4. Make output:
```
  CCLD     _Hamlib.la
Making all in tests
  CC       rigctl-rigctl.o
rigctl.c:668:9: error: use of undeclared identifier 'rl_readline_name'
  668 |         rl_readline_name = "rigctl";
      |         ^
```

**Expected behavior**
Make build runs to completion successfully.

**Desktop (please complete the following information):**
 - OS: FreeBSD 14.3-RELEASE-p2
 - Client application: custom (Python bindings)
 - Client application version: n/a

**Additional context**
The issue appears to be that configure's readline detection is incomplete - it finds the library but doesn't test for the headers. Since FreeBSD installs editline with readline-compatible headers in `/usr/local/include/readline/`, enabling `HAVE_READLINE_READLINE_H` should work.

Workaround:
```
# Force the configure result to recognize the editline readline.h
sed -i 's|/\* #undef HAVE_READLINE_READLINE_H \*/|#define HAVE_READLINE_READLINE_H 1|' include/hamlib/config.h

# Verify the change
grep "HAVE_READLINE_READLINE_H" include/hamlib/config.h
```

N0NB, 2025-09-27:
I tried running your workaround from the commandline and get the following:

``` console
nate@freebsd:~/git/hamlib $ sed -i 's|/\* #undef HAVE_READLINE_READLINE_H \*/|#define HAVE_READLINE_READLINE_H 1|' include/hamlib/config.h
sed: 1: "include/hamlib/config.h
": command i expects \ followed by text
```

techn0mad, 2025-09-28:
Hi Nate:

As a quick take, I'd guess it is because there should be an argument
following the `-i` argument to sed, so the command would need to look like
`sed -i `` 's|/\* #undef ...`

N0NB, 2025-09-28:
Please just post a diff of how the file is modified.  I'm just wasting time on a command that doesn't seem to work.

closed 2025-09-28 by N0NB