Hamlib/Hamlib #1925: Setting and Getting frequency on K3 sometimes gives incorrect result

N7DR, 2025-09-26:
**Describe the bug**
This has been tested only with a K3 (and a K3S; both of them running the latest firmware). The version of hamlib is the version that is included in debian 13 (debian stable, trixie), which appears to be 4.6.2.

Occasionally, rig_set_freq() returns with RIG_OK, but the frequency has in fact not been changed (as one can see on the display). Checking the frequency with rig_get_freq() returns the incorrect frequency, as if the frequency had indeed been set. If one then calls rig_get_freq() several times, with no intervening activity, the incorrect frequency continues to be returned, for one second, after that, the returned frequency changes to the original actual rig frequency (as reflected by the display).

In other words, consider a K3 on frequency A.

1. Send rig_set_freq(B), and receive a RIG_OK.
2. But in fact the display does not change, nor does the audio, because th e rig did not in fact change frequency.
3. Send rig_get_freq(). Receive the response: B.
4. Continue to send a series of rig_get_freq(). These continue to return B, but then suddenly, one second after the original rig_set_freq(B) was sent, rig_get_freq() returns A (and continues to do so thereafter).

**To Reproduce**
This happens only rarely. Perhaps once in every few hundred times that rig_set_freq() is sent.

This C++ program demonstrates the problem (I will also try to attach the same program to this report):

----

// program to demonstrate problem with setting frequency with hamlib on the K3

/*  basic symptom is that *occasionally* the following happens:

    1. call rig_set_freq() to set the frequency;
    2. the call returns RIG_OK [but the rig does not actually change frequency, as can be seen on the physical display]
    3. calling rig_get_freq() responds as if the frequency HAD changed
    4. explicitly requesting the frequency, bypassing hamlib, returns the correct (unchanged) frequency, regardless of whether
         one uses the raw frequency command or the longer "IF" status command
    5. if one keeps reading the frequency using hamlib, without sending any further frequency-change commands,
         then, after one second, hamlib will start to return the correct (unchanged) frequency (the "one second" in this
         item appears to be consistent.

    So the following loops through a sequence of frequency sets (using hamlib), followed by reads using
    both hamlib and the fast and slow explicit commands, and halts on the first occurrence of the behaviour described
    above.

    For simplicity, there is little error checking, and no robust error-handling mechanism in this code.
    However, the code does check the return values from the hamlib frequency-setting and -getting functions.

    The basic flaw seems to be that the frequency-setting function is returning RIG_OK even when it fails to
    set the frequency.

    (There is a question as to why the K3 ever fails to set the frequency correctly... maybe hamlib isn't even sending the command? When
    one uses explicit commands to set the frequency, bypassing hamlib, I don't think I have ever seen a failure.)
*/

#include <array>
#include <iostream>
#include <ranges>
#include <string>
#include <thread>
#include <vector>

#include <cstring>

#include <fcntl.h>
#include <termios.h>
#include <unistd.h>

#include <hamlib/rig.h>

using namespace std;
using namespace std::this_thread;

namespace SR  = std::ranges;
namespace SRV = std::ranges::views;

constexpr size_t FREQ_REPLY_LENGTH   { 14 };          // FA command returns 14 characters
constexpr char   SEMICOLON           { ';' };
constexpr size_t STATUS_REPLY_LENGTH { 38 };          // IF command returns 38 characters
[... 300 lines cut ...]
 hamlib retry after 600ms : hamlib read frequency = 28200000
 hamlib retry after 700ms : hamlib read frequency = 28200000
 hamlib retry after 800ms : hamlib read frequency = 28200000
 hamlib retry after 900ms : hamlib read frequency = 28200000
 hamlib retry after 1000ms : hamlib read frequency = 28290000
 hamlib retry after 1100ms : hamlib read frequency = 28290000
 hamlib retry after 1200ms : hamlib read frequency = 28290000
 hamlib retry after 1300ms : hamlib read frequency = 28290000
 hamlib retry after 1400ms : hamlib read frequency = 28290000
 hamlib retry after 1500ms : hamlib read frequency = 28290000
 hamlib retry after 1600ms : hamlib read frequency = 28290000
 hamlib retry after 1700ms : hamlib read frequency = 28290000
 hamlib retry after 1800ms : hamlib read frequency = 28290000
 hamlib retry after 1900ms : hamlib read frequency = 28290000
n7dr@shack20:~/projects/k3test/build$ 

**Desktop (please complete the following information):**
 - OS: debian trixie (13.0)
 - Client application: the above copy-and-pasted program
 - Client application version: exactly as above

**Additional context**
I Think that this is a pretty long-term bug. I discovered it because my logging program, drlog, would occasionally misbehave when setting.getting frequency on my K3 or K3S (a misbehaviour perhaps once per hour in a contest seemed to be typical). For a long time (at least a couple of years) I thought it was due to a race condition the drlog code. But I finally had time to try to debug the code, and eventually convinced myself that nope, there is no relevant race condition, so I started looking carefully into the interaction between hamlib and the rig, and discovered the above behaviour.

[k3test.cpp](https://github.com/user-attachments/files/22568256/k3test.cpp)

N0NB, 2025-09-28:
Please provide the command to compile and link your test program.

N7DR, 2025-09-28:
Nate Bargmann wrote on 9/28/25 14:40:
> N0NB left a comment (Hamlib/Hamlib#1925)
> 
> Please provide the command to compile and link your test program.
> 

Sorry; I didn't even think of that. cmake produces something quite 
complicated, but stripped down it is (I just checked that this works to 
generate a correct a.out; and I expect that it's obvious that you don't really 
need the -g and -O flags):

  g++ -g3 -O2 -std=c++23 k3test.cpp -lhamlib

You do need the c++23 (or c++26) value for the -std flag.

   73 -- Doc  N7DR

-- 
Web:  http://enginehousebooks.com/drevans

N0NB, 2025-09-29:
BTW, for serial port passthrough I added the following lines to my VM startup script:

``` bash
    -chardev serial,path=/dev/ttyS0,id=k3serial \
    -device pci-serial,chardev=k3serial
```

This results in the guest serial port `/dev/ttyS1` being attached to the host serial port.  I suppose this is due to the guest `/dev/ttyS0` being attached to the serial port monitor.

N7DR, 2025-09-29:
I increased the levels of hamlib DEBUG information, and ran the program until it hit a failure. The output showed no change for levels and BUG, ERR, WARN. But when set to VERBOSE, I captured the attached file (just the part with the failure included).

A not-very-thorough look at it suggests that hamlib thinks that it correctly set the frequency to 28200000 (which is what it was told to do), but in fact the frequency did not change from its prior value of 28290000.

Then, for one second following the command to change frequency, hamlib doesn't actually communicate with the rig at all, but simply returns the (cached?) incorrect value of 28200000 [which seems pretty weird; I mean: someone might have turned the dial, so it's not sensible to cache the value, if that is indeed what's happening]. After one second, the next call actually issues another real frequency request to the rig, which returns the correct old frequency of 28290000.

I note that I haven't looked at any hamlib code at all. I believe that it's C rather than C++, and I don't regard myself as competent to attempt to follow and/or debug C code; the last time I wrote C was, I believe, in 1989.

N7DR, 2025-10-02:
Incidentally I find this a puzzle near the top of the file 100-end.txt:

write_block(): TX 14 bytes
0000    46 41 30 30 30 32 38 32 30 30 30 30 30 3b           FA00028200000;  
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string_generic(): RX 6 characters, direct=1
0000    49 44 30 31 37 3b                                   ID017;          
kenwood_transaction: read_string len=6 'ID017;'
kenwood_transaction: No data expected, checking ID; in ID017;
kenwood_transaction: returning RIG_OK, retval=0

I ask myself why on Earth would hamlib feel the need to check that the rig ID is 017? It's done that multiple times already; it's not at all obvious to my simple brain why that needs to be done more than once, at start-up. FWIW, a combination of grep and wc tells me that hamlib checks the rig ID 100 times on that particular run of k3test... it never seems to learn and make use of the fact that it already knows what the ID is [and, indeed, why does it care, once it's established the fact at start-up?].

Should I file a separate bug report saying something like "Extraneous calls for rig ID?". Or am I missing something blindingly obvious?

N7DR, 2025-10-02:
Oh, I should mention that I have a P3 sitting between the computer and the rig. It occurred to me that there's a (small but) non-zero chance that the reason that the rig is occasionally failing to execute the change in frequency is because the P3 firmware isn't properly forwarding the command to the rig. I think that quite a lot of communication goes on between the rig and the P3 on the serial line that connects them (Elecraft doesn't seem to provide any details), and there's probably at least some scope for Elecraft to have failed to make sure that commands received by the P3 are properly forwarded.

This is all out of scope for the bug of course, but it might be one explanation for why the assumption that hamlib seems to make that the command it sends is always correctly executed by the rig might turn out to be invalid.  Or the failure could be a silly race condition in the K3 firmware, of course; in which case the presence of the P3 is irrelevant.

N7DR, 2025-10-03:
Nate Bargmann wrote on 10/2/25 20:29:

> 
> I was going to note the fact that I have a P3 in line as well, thinking that you didn't.
> 
> I've run your test program several times (meaning around 6 times of 500 loops) and have not had a single failure.  Each run has completed apparently normally.
> 
> I will say this, as intense as WSJT-X and Fldigi exercise Hamlib and with the thousands of hams using those programs every day, if there were an issue I'm sure the issues would abound!  As a result I'm wondering if the issue is local to your station (always a delicate thing to bring up in an issue!).

Well, yes, I understand. The only thing I can think of is some sort of 
intermittent failure in the RS-232 board on the computer... but that seems so 
unlikely as to be instantly capable of being dismissed. Or, I suppose, a cable 
issue... but again, it's hard to see how the problem could be so intermittent 
in that case (but I will root around and see if I have a spare RS-232 cable 
with the correct connectors that I could swap into the circuit). This kind of 
intermittency does smell like a race condition (in either the P3 or the K3) to 
me... but I've been known to be wrong :-)

I will say that the paranoid defensive programmer in me says that one should 
never transmit a command that's supposed to change the state on the receiving 
device without checking that it's actually been executed, regardless of 
whether that failure ever actually occurs. But probably that's just me.

A quick question because I've never used serial ports in a VM, and I perhaps 
don't understand the nuances of doing so: your setup does end up with the 
commands and responses physically travelling at 38400 baud over a real serial 
link all the way between the computer and the rig, right? (i.e., the RS232 
configuration item on the K3 is set to "38400b"), with a stopover at the P3. 
That is: the computer<-->P3 and P3<-->K3 paths are both real RS232 at 38400 baud?

   73 -- Doc  N7DR

-- 
Web:  http://enginehousebooks.com/drevans

N0NB, 2025-10-04:
> A quick question because I've never used serial ports in a VM, and I perhaps
> don't understand the nuances of doing so: your setup does end up with the
> commands and responses physically travelling at [...]

Yes.

For the VM to see the port, I have added QEMU command options to my startup script to map the serial port into the VM.  Because QEMU maps `/dev/ttyS0` in the VM to the serial console (needed for some emulation where a video board wouldn't normally be present), my hardware `/dev/ttyS0` is mapped to `/dev/ttyS1` in the VM.  I simply edited the port in your program and recompiled to make that work.  QEMU handles the pass through from the VM to the hardware.

N0NB, 2025-10-04:
The K3S line starts with a much higher serial number--8k or something like that.  I put my K3 together right at 15 years ago.  Time flies with a good radio on the desk.

N7DR, 2025-10-04:
Nate Bargmann wrote on 10/4/25 09:31:
> N0NB left a comment (Hamlib/Hamlib#1925)
> 
> The K3S line starts with a much higher serial number--8k or something like that.  I put my K3 together right at 15 years ago.

Very close to the same here :-) K3 #4430; first QSO 18 July, 2010.

The K3S was added when I realised through experience that if one has an 
Elecraft rig, it's a good idea to have two, because they aren't the most 
reliable of rigs; even though I also built my K3, I'm not particularly happy 
or adept at working on it myself. (The K3S, which was factory-built, just came 
back from Watsonville. Its last trip was last year.)

>  Time flies with a good radio on the desk.

I suspect that it flies anyway :-)

   73 -- Doc  N7DR

-- 
Web:  http://enginehousebooks.com/drevans

N0NB, 2025-10-04:
I must be blessed as I've never had an issue with mine.  It was even out on Field Day one year after I added the P3.  It survived the move out here to the farm 11 years ago, but maybe its day is coming.  Dunno.

N7DR, 2025-10-04:
Nate Bargmann wrote on 10/4/25 10:03:
> N0NB left a comment (Hamlib/Hamlib#1925)
> 
> I must be blessed as I've never had an issue with mine.  It was even out
> on Field Day one year after I added the P3.  It survived the move out here
> to the farm 11 years ago, but maybe its day is coming.  Dunno.
> 

I haven't added them up, but I suspect that on average each rig has gone back
after about two years of heavy use as my main rig (for some reason that I
can't properly quantify, I prefer the K3: it just seems to be more sensitive
at very low bandwidths; every time I tell myself it must be my imagination and
try them both, I always end up back on the K3).

Some time ago I stopped sending them back for minor faults that I could live
with -- the K3 has refused to put out full power on 20m for three or four
years now: it gives a HI CUR warning over about 80W. I'm waiting until that
fault spreads to more bands or the rig develops some more serious fault before
sending it back. If the turnaround time was less, I'd probably have already
sent it back, but my experience is that I end up being without the rig for
between two and three months, which is more than I'm comfortable with unless
the rig really is unusable here. Way off topic now :-)

   73 -- Doc  N7DR

-- 
Web:  http://enginehousebooks.com/drevans

N7DR, 2025-11-01:
So here's some new info: if I leave the P3 in line, but power it down, then, so far, I have been unable to reproduce the problem. I wish I had an RS232 inline analyzer I could put between the P3 and the K3, but I don't. Absent such a device, all I can do is indulge in quasi-informed speculation.

Just to be clear, this is definitely reproducible here. P3 powered on => the reported problem exists; P3 powered off => either the problem is completely eliminated, or the probability of its occurrence is reduced to the point that it has so far been undetectable.

What I speculate, on the basis of these observations, is that there are problems in hamlib, and also in the firmware in the K3 (and perhaps the P3).

The problem in hamlib is, basically, that when it throws a command at a rig, it just seems to assume that everything is hunkydory and the rig will do what it's been told. This seems to me to be far too optimistic... who's to say that the rig: (a) receives the command exactly as it was sent; and (b) correctly acts on the command? Given that the rig control protocol contains no error detection or checking, nor any kind of delivery guarantee, nor, at least in the case of Elecraft, that a received command is absolutely guaranteed to execute, it seems to me that any command sent to the rig and intended to change the rig's state should be checked that it was actually executed. (Even in the absence of the problem we're tailing about here, a lot of shacks have enough RFI to make the assumption that serial lines work 100% correctly pretty iffy.)  And it is even more over-optimistic to cache what it expects to have have happened and return the cached state for a period of time (i.e., one second in this case) before actually checking the state.

The problems in the firmware ... well, those are harder to guess at, but the difference in behaviour between P3-powered-on and P3-powered-off suggest that whatever protection exists in the K3 to ensure that received commands aren't garbled or intermingled is not robust: it looks likely that whatever communication occurs between the P3 and the K3 (as opposed to the computer and the K3, running on the same serial line) are not always properly protected from commingling.  The protocol used by Elecraft was (obviously) designed for non-multiplexed serial lines, in a period when baud rates were low. Why they continued to use it post-(say)2000, when much more robust systems were available, I have no idea. Anyway, this is all rather irrelevant (and might be wrong anyway).

Elecraft doesn't seem to have documented the serial messaging taking place between the P3 and the K3 (at least, I can't find any documentation) but I imagine that the details depend on precisely how the P3 is set up. It may well be that you and I have our P3s configured differently, and that is why you did not see the reported issue (obviously, I'm guessing again; it's the only explanation that came to my mind -- there has to be *something* different about our setups).